microsoft/AirSim #2419: Question: Energy level of the multirotor

charbelmansour, 2020-02-14:
Hello,
I find that there exist these sensors in Airsim:
-Camera
-IMU
-Magnetometer
-GPS
-Barometer
-Distance sensor
-Lidar
Is there a sensor related to the energy of a multirotor? I mean if there exists a sensor indicating how much energy remains in the multirotor.
If yes, can you tell me please how it works, I mean by what factor does the energy level of the multirotor decrease based on the traveled distance.
Best Regards,
Thank you for your help.

rajat2004, 2020-02-14:
Currently, there's no battery on energy consumption calculation in AirSim. https://github.com/microsoft/AirSim/issues/672 could be helpful for implementing such a feature

aiaarif, 2020-02-14:
You can take a look at this repo. They claim to have implemented the energy model. https://github.com/srivatsankrishnan/AirSim-1

charbelmansour, 2020-02-16:
okay thank you I will check the site.

charbelmansour, 2020-02-16:
Can you tell me please where I can find rpc-base-reset-plus-energy branch in this repository
https://github.com/srivatsankrishnan/AirSim-1 @aiaarif

aiaarif, 2020-02-17:
You can switch to different branch by clicking branch icon and selecting the appropriate one. Github should have a nice tutorial on that.

charbelmansour, 2020-02-18:
Hello @aiaarif 

I have cloned the project and run it on unreal engine. The good news that it works fine.

However, we are running this simulation on Window and the drone is not flying.

Can you please advise us how we can make the drone fly. We have imported the Hello Drone folder but still the drone is not flying.

Thank you for your help,

Charbel

aiaarif, 2020-02-19:
Please refer to the documentation of Airsim. They have a step by step by guide for that.

charbelmansour, 2020-02-24:
Hello, I cloned this project: https://github.com/srivatsankrishnan/AirSim-1
I want to get information about the energy of the drone. In this repository, they have implemented a method getTripStats that returns information about the trip including the energy consumed.
I am able to control the drone using python commands.
I am trying to run this code:

import airsim

import numpy as np
import os
import tempfile
import pprint

client = airsim.MultirotorClient(ip="127.0.0.1")
client.confirmConnection()
client.enableApiControl(True)
client.armDisarm(True)

res = client.getTripStats()
print (res.flight_time)

I'm getting the following error:
[image]

I also tried this code to get the state of the multirotor:
import airsim

import numpy as np
import os
import tempfile
import pprint

client = airsim.MultirotorClient(ip="127.0.0.1")
client.confirmConnection()
client.enableApiControl(True)
client.armDisarm(True)

state = client.getMultirotorState()
s = pprint.pformat(state)
print("state: %s" % s)

I get the following error:
[image]

Any suggestions about the python command that allows to retrieve the trip stats including the energy?

Thank you for your help, 

Best Regards,
Charbel

charbelmansour, 2020-02-24:
Please note that the codes that I wrote above were saved in a file xyz.py
To run each code, I execute python xyz.py in the command prompt.

aiaarif, 2020-02-26:
Please report issue about 'that' repository in 'that' repository and close this question here.

closed 2020-03-10 by ironclownfish