microsoft/AirSim #2422: Can't connect to Unreal running on Windows from Linux machine

ivannson, 2020-02-18:
I am trying to connect `CarClient` running on Ubuntu to a simulation running on Windows (Windows firewall is disabled), using #1551. 

In the script that is running on Ubuntu, I am trying to connect using: 
```python
self.client = airsim.CarClient(ip="windows.server.ip.address", port=41451)
self.client.confirmConnection()
```

However, it is not connecting and exits with `WARNING:tornado.general:Connect error on fd 9: ECONNREFUSED`.

Am I doing this correctly?

rajat2004, 2020-02-18:
Not sure about the correct way to check the IP address, maybe check with `ipconfig` also
Check the network adapter if the correct IP is being used, such as using Ethernet to connect to the Ubuntu machine while WiFi is also on
Just some suggestions from the top of my head

ivannson, 2020-02-25:
@rajat2004 Thanks for the suggestions, I think the IP addresses are fine. 

Just to check, in theory, to connect between the two, I do the following:
- On Ubuntu, running Python API, set the IP address of the rendering machine (Windows)
- On Windows, running Unreal, specify the IP address of the rendering machine (so itself) in `settings.json` in the `LocalHostIP` field

Is there anything else that needs to be done?

jonyMarino, 2020-08-17:
Closed as in #2386

closed 2020-08-17 by jonyMarino