microsoft/AirSim #2423: What real drone configurations would work with AirSim in a proper way?

Supaple-x, 2020-02-19:
Hi guys,
I would like to build my own drone for use with AirSim.
So do you have some recomendation specs for drone building to work good with AirSim?

rajat2004, 2020-03-11:
Do you mean like an actual physical drone? Or building a model in UE or Unity?

Supaple-x, 2020-03-11:
> Do you mean like an actual physical drone? Or building a model in UE or Unity?

Hi,
I mean actual physical drone. In ue4 I have constructed several models.
But how I will put this into practice?
May be here exists the best practice or tutorial to construct real drone to use with airsim?

rajat2004, 2020-03-11:
There is a small Readme for using [Real Drones](https://github.com/microsoft/AirSim/blob/master/docs/custom_drone.md), doesn't have any info regarding the actual physical parameters of the drones, from the image looks like a normal F450 frame
Though am not sure how does the physical characteristics of the drone matter between AirSim and the real drone, all the low-level control would be taken care of by the firmware. Unless it's something like an RL-based approach to fly the drone with it directly driving the motors.

clev0910, 2020-05-06:
I actually wanted to use airsim to control a model of a drone that I made with Autodesk inventor, but UE4 crashes each time I try to run AirSim with it. Are there any particular settings required?