microsoft/AirSim #2424: Coaxial Octorotor HIL

M-J-Murray, 2020-02-20:
I'm attempting to setup a simulation in HIL mode with a octorotor coaxial airframe. I have followed the steps in https://github.com/microsoft/AirSim/wiki/hexacopter. However, the drone doesn't seem to be responding to inputs. 

Firstly, I setup a new init file for the airframe configuration:
1004_rc_octo_cox.hil
```bash
#!/bin/sh
#
# @name HIL Octocoptor Coaxial
#
# @type Simulation
# @class Copter
#
# @maintainer Lorenz Meier <lorenz@px4.io>
#

sh /etc/init.d/rc.mc_defaults

set MIXER octo_cox
set PWM_OUT 12345678

param set SYS_HITL 1
```

Then I changed the relevant qgroundcontrol config so I could enable the airframe mode.
Then I added the relevant code to the airsim plugin.
I added to "AirSim\Source\AirLib\include\vehicles\multirotor\firmwares\mavlink\Px4MultiRotorParams.hpp":
```c++
	void setupFrameGenericOct(Params& params)
	{
		params.rotor_count = 8;
		params.arm_count = 4;
		real_T arm_length = 1.1f;

		params.mass = 110.0f;
		real_T motor_assembly_weight = 3.0f;
		real_T box_mass = params.mass - params.rotor_count * motor_assembly_weight;

		params.rotor_params.calculateMaxThrust();

		//set up dimensions of core body box or abdomen (not including arms).
		params.body_box.x() = 0.375f; params.body_box.y() = 0.375f; params.body_box.z() = 1.1f;
		real_T rotor_z = 0.4f;

		//computer rotor poses
		initializeRotorOctoCoax(params.rotor_poses, params.rotor_count, params.arm_count, arm_length, rotor_z);

		//compute inertia matrix
		computeInertiaMatrix(params.inertia, params.body_box, params.rotor_poses, box_mass, motor_assembly_weight);
	}
```

and to "AirSim\Source\AirLib\include\vehicles\multirotor\MultiRotorParams.hpp":
```c++
static void initializeRotorOctoCoax(vector<RotorPose>& rotor_poses /* the result we are building */,
		uint rotor_count /* must be 8 */,
		uint arm_count /* must be 4 */,
		real_T arm_length,
		real_T rotor_z /* z relative to center of gravity */)
	{
		Vector3r unit_z(0, 0, -1);  //NED frame
		if (rotor_count == 8 && arm_count == 4) {
			rotor_poses.clear();

			// vectors below are rotated according to NED left hand rule (so the vectors are rotated counter clockwise).
			Quaternionr octo_rot(AngleAxisr(M_PIf / 4, unit_z));
			real_T arm_off = std::sqrt(std::pow(arm_length, 2)/2);
			rotor_poses.emplace_back(VectorMath::rotateVector(Vector3r(arm_off, arm_off, rotor_z), octo_rot, true),
				unit_z, RotorTurningDirection::RotorTurningDirectionCCW);
			rotor_poses.emplace_back(VectorMath::rotateVector(Vector3r(-arm_off, arm_off, rotor_z), octo_rot, true),
				unit_z, RotorTurningDirection::RotorTurningDirectionCW);
			rotor_poses.emplace_back(VectorMath::rotateVector(Vector3r(-arm_off, -arm_off, rotor_z), octo_rot, true),
				unit_z, RotorTurningDirection::RotorTurningDirectionCCW);
			rotor_poses.emplace_back(VectorMath::rotateVector(Vector3r(arm_off, -arm_off, rotor_z), octo_rot, true),
				unit_z, RotorTurningDirection::RotorTurningDirectionCW);
			rotor_poses.emplace_back(VectorMath::rotateVector(Vector3r(-arm_off, arm_off, rotor_z), octo_rot, true),
				unit_z, RotorTurningDirection::RotorTurningDirectionCCW);
			rotor_poses.emplace_back(VectorMath::rotateVector(Vector3r(arm_off, arm_off, rotor_z), octo_rot, true),
				unit_z, RotorTurningDirection::RotorTurningDirectionCW);
			rotor_poses.emplace_back(VectorMath::rotateVector(Vector3r(arm_off, -arm_off, rotor_z), octo_rot, true),
				unit_z, RotorTurningDirection::RotorTurningDirectionCCW);
			rotor_poses.emplace_back(VectorMath::rotateVector(Vector3r(-arm_off, -arm_off, rotor_z), octo_rot, true),
				unit_z, RotorTurningDirection::RotorTurningDirectionCW);
		}
		else
			throw std::invalid_argument("Rotor count other than 8 is not supported by this method!");
	}
```

I also had to change the default `rotor_count` in "AirSim\Source\Vehicles\Multirotor\FlyingPawn.h" to 8. I also changed the physics parameters "AirSim\Source\AirLib\include\vehicles\multirotor\RotorParams.hpp".

```c++
struct RotorParams {
            /*
            Ref: http://physics.stackexchange.com/a/32013/14061
            force in Newton = C_T * \rho * n^2 * D^4
            torque in N.m = C_P * \rho * n^2 * D^5 / (2*pi)
            where,
            \rho = air density (1.225 kg/m^3)
            n = radians per sec
            D = propeller diameter in meters
            C_T, C_P = dimensionless constants available at
            propeller performance database http://m-selig.ae.illinois.edu/props/propDB.html

            We use values for GWS 9X5 propeller for which,
            C_T = 0.109919, C_P = 0.040164 @ 6396.667 RPM
            */
            real_T C_T = 0.08f; // the thrust co-efficient @ 6396.667 RPM, measured by UIUC.
            real_T C_P = 0.04f; // the torque co-efficient at @ 6396.667 RPM, measured by UIUC.
            real_T air_density = 1.225f; //  kg/m^3
            real_T max_rpm = 3000.0f; // revolutions per minute
            real_T propeller_diameter = 1.016f;   //diameter in meters
            real_T propeller_height = 0.035;   //height of cylindrical area when propeller rotates
            real_T control_signal_filter_tc = 0.005f;    //time constant for low pass filter

            real_T revolutions_per_second;
            real_T max_speed; // in radians per second
            real_T max_speed_square;
            real_T max_thrust = 4.179446268f; //computed from above formula for the given constants
            real_T max_torque = 0.055562f; //computed from above formula

            // call this method to recalculate thrust if you want to use different numbers for C_T, C_P, max_rpm, etc.
            void calculateMaxThrust() {
                revolutions_per_second = max_rpm / 60;
                max_speed = revolutions_per_second * 2 * M_PIf;  // radians / sec
                max_speed_square = pow(max_speed, 2.0f);
                real_T nsquared = revolutions_per_second * revolutions_per_second;
				real_T ncubed = nsquared * revolutions_per_second;
				real_T dquatric = pow(propeller_diameter, 4);
				real_T dquintic = dquatric * propeller_diameter;
				
                max_thrust = C_T * air_density * nsquared * dquatric;
                max_torque = C_P * air_density * nsquared * dquintic / (2 * M_PIf);
            }

        };
```

I've updated my drone blueprint to contain the proper rotations and the "SetupPropRotationMover" function in UE4. Does anyone know if there is anything else I still need to change? Or if there is anything wrong with my current configuration?

rajat2004, 2020-03-11:
Maybe you'll need to create a new asset from scratch? Not sure, https://github.com/microsoft/AirSim/pull/1890 & https://github.com/microsoft/AirSim/pull/1139 might be helpful regarding any additional codebase changes required, they might be overlapping in some places such as the assets

airsimdevdd, 2021-02-23:
i want to import my custom co axial octacopter into airsim please help me out is there  correct and working process to do it

airsimdevdd, 2021-02-26:
@rajat2004 @jonyMarino @M-J-Murray can any one help me out in import my custom coaxial drone .I am a newbie , need a step by step sequence please  need ur support .  thanks in advance

aaqibism, 2021-04-01:
> @rajat2004 @jonyMarino @M-J-Murray can any one help me out in import my custom coaxial drone .I am a newbie , need a step by step sequence please need ur support . thanks in advance

When you say import, are you talking about importing a CAD. If so check out [this tutorial](https://drive.google.com/file/d/1-iOfJ8zO3YA4Mzc0ZkKwTM3L4--4RFtt/view) that demos creating a hexa.

jonyMarino, 2021-04-30:
Hi @airsimdevdd! for importing a mesh take a look at https://youtu.be/Bp86WiLUC80

closed 2021-04-30 by jonyMarino