microsoft/AirSim #2426: Migrating from Gazebo & ROS to AirSim advice needed

caesar84, 2020-02-23:
Hi there.

I am new to Airsim and would appreciate it if you can help me with my questions.
I have tested an autonomous flying MAV algorithm based on deep supervised learning, and it was successful. I had used Matlab, ROS and Gazebo simulator for this purpose and the general scheme is as follow:
1- I read Image from the onboard camera of the MAV in Matlab Via ROS.
2- A DNN running on Matlab will classify the Image.
3- Classification result converts to a Velocity Command By a local motion planner in Matlab.
4-Command  will be sent from Matlab to The MAV via ROS.

the MAV velocity command is in the form of six parameters 3 for linear ( x,y, and z)  
and 3 for angular ( Pitch, Roll and YAW) as of [here ](https://ardrone-autonomy.readthedocs.io/en/latest/commands.html)  and  [here](http://wiki.ros.org/tum_simulator#Run_Simulator), my questions are:
1-Does anyone know how the velocity command looks like in Airsim? is it the same so I wont have problem migrating my work to Airsim?
2- I have skimmed over the documentation and didn't find a concrete example answering the above question, However, it has been mentioned that there is an API ( I'm not that very knowledgable about APIs though) that can facilitate 'controlling the MAV programmatically'
. how easy it to run that in a loop fashion? 
3-If I want to stick to windows rather than going back to Linux, can I safely migrate my work into Airsim from Linux, ROS and Gazebo
assuming that there is a way to establish a link between Matlab and Airsim to read the Images and send back velocity commands via API.

finally, excuse me if I had assumed or misused some of the terms especially API because of my limited knowledge.

rajat2004, 2020-03-10:
About the APIs, you can have a look at the Readme - https://github.com/microsoft/AirSim/blob/master/docs/apis.md, which gives info regarding the supported APIs and also has links to examples which should clear up most of the confusion
It might be the case that some APIs aren't mentioned in the Readme which have been added, directly going through the [client.py](https://github.com/microsoft/AirSim/blob/master/PythonClient/airsim/client.py) file will help in such cases

AirSim works on both Windows and Linux, so it's upto you regarding the platform to use. If you need to use ROS as well, then Linux might be the better option, though you could also do something like run AirSim on Windows while ROS, etc. runs in WSL

caesar84, 2020-03-10:
thank you rajat2004, that was helpful and great explanation

closed 2020-03-10 by caesar84