microsoft/AirSim #2432: update_from_git.bat doesn't generate .sln file

caesar84, 2020-02-26:
I have just installed Airsim and reached the point where I want to do "Setup Blocks Environment for AirSim" according to [https://microsoft.github.io/AirSim/docs/unreal_blocks/](this). I have tried both ways: from command prompt window and directly clicking on the update_from_git.bat however the only output is: `Using AirSimPath = ..\..\..

            New File                 389        E:\Visual studio\AirSim\Unreal\Plugins\AirSim\Source\AirLib\AirSim.props

          *EXTRA File                389        E:\Visual studio\AirSim\Unreal\Environments\Blocks\Plugins\AirSim\Source\AirLib\AirSim.props

E:\Visual studio\AirSim\Unreal\Environments\Blocks>rd /s/q Build
The system cannot find the file specified.

E:\Visual studio\AirSim\Unreal\Environments\Blocks>rd /s/q Binaries
The system cannot find the file specified.

E:\Visual studio\AirSim\Unreal\Environments\Blocks>rd /s/q Intermediate

E:\Visual studio\AirSim\Unreal\Environments\Blocks>rd /s/q Saved

E:\Visual studio\AirSim\Unreal\Environments\Blocks>mkdir Saved\logs

E:\Visual studio\AirSim\Unreal\Environments\Blocks>rd /s/q Plugins\AirSim\Binaries
The system cannot find the file specified.

E:\Visual studio\AirSim\Unreal\Environments\Blocks>rd /s/q Plugins\AirSim\Intermediate
The system cannot find the file specified.

E:\Visual studio\AirSim\Unreal\Environments\Blocks>rd /s/q Plugins\AirSim\Saved
The system cannot find the file specified.

E:\Visual studio\AirSim\Unreal\Environments\Blocks>del *.sln
Could Not Find E:\Visual studio\AirSim\Unreal\Environments\Blocks\*.sln

E:\Visual studio\AirSim\Unreal\Environments\Blocks>setlocal

E:\Visual studio\AirSim\Unreal\Environments\Blocks>del /q gen_temp.txt
Could Not Find E:\Visual studio\AirSim\Unreal\Environments\Blocks\gen_temp.txt

E:\Visual studio\AirSim\Unreal\Environments\Blocks>powershell -command "& { (Get-ItemProperty 'Registry::HKEY_CLASSES_ROOT\Unreal.ProjectFile\shell\rungenproj' -Name 'Icon' ).'Icon' } > gen_temp.tmp"

E:\Visual studio\AirSim\Unreal\Environments\Blocks>type gen_temp.tmp  1>gen_temp.txt

E:\Visual studio\AirSim\Unreal\Environments\Blocks>set /p gen_bin= 0<gen_temp.txt

E:\Visual studio\AirSim\Unreal\Environments\Blocks>del /q gen_temp.tmp

E:\Visual studio\AirSim\Unreal\Environments\Blocks>del /q gen_temp.txt

E:\Visual studio\AirSim\Unreal\Environments\Blocks>for %f in (*.uproject) do (
echo Generating files for %f
 "E:\Epic Games\Epic Games\Launcher\Engine\Binaries\Win64\UnrealVersionSelector.exe" /projectfiles "E:\Visual studio\AirSim\Unreal\Environments\Blocks\%f"
)

E:\Visual studio\AirSim\Unreal\Environments\Blocks>(
echo Generating files for Blocks.uproject
 "E:\Epic Games\Epic Games\Launcher\Engine\Binaries\Win64\UnrealVersionSelector.exe" /projectfiles "E:\Visual studio\AirSim\Unreal\Environments\Blocks\Blocks.uproject"
)
Generating files for Blocks.uproject
Press any key to continue . . .`

would appreciate it if anyone has an idea of how to solve this.
I am runining Unreal 4.18.3, VS 2017 on windows 10

rajat2004, 2020-03-10:
Did you run the `build.cmd` script first from the root of AirSim repo? Any error while running that?

caesar84, 2020-03-10:
rajat2004 , I did clean it and removed a previous version of VS2015 and run it again and everything went well.

thanks

closed 2020-03-10 by caesar84