microsoft/AirSim #2433: It seems that simGetCollisionInfo () sometimes doesn't work

hanlei666-python, 2020-02-27:
Hi.I recently used reinforcement learning to train UAVs to avoid obstacles, but what bothered me was that the simgetcollisioninfo() function didn't always return "true" when UAVs collided in a virtual environment. For example, it will get into the grass and cannot move, and simgetcollisioninfo() will not remind me.
Here is my code：
<def step(self,action,step_count):
                self.client.simPause(False)
                temp = airsim.YawMode()
                temp.is_rate = False
                self.client.moveByVelocityAsync(action[0], action[1], action[2], 3 ,airsim.DrivetrainType.ForwardOnly,temp)
                
                
                start_time = time.time()
                while time.time() - start_time <= 3:
		#pos = state_[1][2]
                #pos_ = v2t(self.client.simGetGroundTruthKinematics().position)                        
                        info = None
                        done = False
                        
                        pos = v2t(self.client.simGetGroundTruthKinematics().position)
                        if self.isDone(pos):
                                done = True
                                reward = 50
                                info = "success"
			
                        elif self.client.simGetCollisionInfo().has_collided:
                                reward = -50
                                done = True
                                info = "collision"
                        elif pos[2] < self.height_limit:
                                done = True
                                info = "too high"
                                reward = -50
                        elif pos[2]>=-1:
                                reward = -50
                                done = True
                                info = "landed"
                        elif step_count>=200:
                                reward = -50
                                done = True
                                info = "freeze"
                        else:
                                self.i=True
                self.client.simPause(True)>

ironclownfish, 2020-08-14:
@hanlei666-python Hi, can you please share the collision settings for the ground in this level?
In particular check whether it's set to "Collision Enabled", "GenerateOverlapEvents=true", and that its set to collide with all collision channels.
The collision settings can be found in the details panel of the actor.

jonyMarino, 2020-09-18:
Closed due to age and incompleteness.
Please feel free to open a new issue (referring this one) if you still have this problem, and include the following (as requested when opening a bug issue)

*Make sure to write all reproduction steps
*Include full error message in text form
*Search issues for error message before filing issue
*Attach screenshot if applicable
*Include code to run if applicable

Thanks!

closed 2020-09-18 by jonyMarino