microsoft/AirSim #2438: DisparityNormalized

XiuXiang, 2020-03-02:
# Read This First

## If you are reporting a bug
* Make sure to write **all reproduction steps**
* Include full error message in text form
* Search issues for error message before filing issue
* Attach screenshot if applicable
* Include code to run if applicable

## If you have question
* Add clear and concise title
* Add OS, AirSim version, Python version, Unreal version if applicable
* Include context on what you are trying to achieve
* Include details of what you already did to find answers

**What's better than filing issue? Filing a pull request :).**

------------------------------------ (Remove above before filing the issue) ------------------------------------

rajat2004, 2020-03-12:
Is there something specific regarding this? Please have a look at https://github.com/Microsoft/AirSim/issues/583 if that clears it up

hanlei666-python, 2020-03-17:
Hello, I am running the program in the airsim environment downloaded from the Internet. This environment map is AirSimNH.
When I want to obtain a normalized disparity image, the pixels in the picture are not within [0, 1].
It's a small number.
`responses=client.simGetImages([airsim.ImageRequest(0,airsim.ImageType.DisparityNormalized ,True,False)])
img1d = np.array(responses[0].image_data_float, dtype=np.float)
img2d = np.reshape(img1d, (responses[0].height, responses[0].width))
print(img2d[2,2])
1.1444091796875e-05`
python version:3.5.2,AirSimNH version:1.2.1