microsoft/AirSim #2440: How can I get extrinsic parameters between camera and other sensors?

Beniko95J, 2020-03-04:
Hi, I am trying to run VINS-Mono on airsim and I have posted an issue about the frame rate of getImuData(). I solved it by using a desktop computer and currently I can achieve a 20Hz image and 200Hz IMU readings.

Now I am wondering how to get (or set) the extrinsic parameters between camera and other sensors like IMU. I find that there are some parameters named "X", "Y", "Z", "Pitch", "Roll", "Yaw" in the setting file while I think the meaning of them is confused. Are they poses from camera to world or inverse, or other things like poses from camera to the center of the vehicle?

Thanks!

Zartris, 2020-05-04:
Hi Beniko95j
I'm sitting in the same situation and wondering if you have solved it :)

Beniko95J, 2020-05-06:
Hi, @Zartris 
Haven't found a good solution, but maybe you can refer to the implementation in #2369.

Best