microsoft/AirSim #2444: Distance Sensor in AirSim

YKembal, 2020-03-05:
I am trying to use the distance sensor (for a Vechile not a Drone) in airsim and it is working properly except that it only measure the distance from the bottom of the car to the ground and i don't know where to change it to make it measure the distance from the front of the car not from it's bottom , PLEASE HELP.

This is an example of the current distance sensor output :

<DistanceSensorData> {   'distance': 0.05738349258899689,
    'max_distance': 40.0,
    'min_distance': 0.20000000298023224,
    'relative_pose': <Pose> {   'orientation': <Quaternionr> {   'w_val': 1.0,
    'x_val': 0.0,
    'y_val': 0.0,
    'z_val': 0.0},
    'position': <Vector3r> {   'x_val': 0.0,
    'y_val': 0.0,
    'z_val': 0.0}},
    'time_stamp': 1583421822500312700}

rajat2004, 2020-03-10:
Distance sensor currently doesn't support modification of orientation or position through the settings, could have a look at using Lidar for this
Another possibility is to add such a transforming code which takes the orientation from settings and applies it, which should be similar to the already existing code for Lidar

closed 2020-03-10 by ironclownfish

YKembal, 2020-03-10:
> Distance sensor currently doesn't support modification of orientation or position through the settings, could have a look at using Lidar for this
> Another possibility is to add such a transforming [...]

Can you give an example on how to taake the orientaaation of a sensor from settings and change it and apply it as you have said ?

rajat2004, 2020-03-11:
The settings for Lidar are parsed here - https://github.com/microsoft/AirSim/blob/master/AirLib/include/common/AirSimSettings.hpp#L1175
This takes in position and orientation of the Lidar, which are then used in [`LidarSimpleParams.hpp`](https://github.com/microsoft/AirSim/blob/master/AirLib/include/sensors/lidar/LidarSimpleParams.hpp#L70)
Some transformation also happens in [`LidarSimple.hpp`](https://github.com/microsoft/AirSim/blob/master/AirLib/include/sensors/lidar/LidarSimple.hpp#L89)

This should be a good starting place for adding such a feature. Files across different sensors have similar names and implementations, to make it easier for developing new features, which is the case across most of the AirSim codebase for different vehicles, etc.

EdwinMeriaux, 2020-05-24:
> Distance sensor currently doesn't support modification of orientation or position through the settings, could have a look at using Lidar for this
> Another possibility is to add such a transforming [...]

does the distance sensor still have the limitation of modification of orientation?

rajat2004, 2020-05-24:
I believe so, don't remember any changes being made to sensors, etc recently