microsoft/AirSim #2446: Vehicle not responding to Hello_Car.py Plugin

jordon96, 2020-03-06:
Hi, i've created a custom Environment and Vehicle in Airsim. I've complied it and the vehicle responded to manual keyboard input to move. However, when i try to run Hello_Car.py. The vehicle does not move, but the throttle and gear change is still reflected at the top left side.

rajat2004, 2020-03-10:
The values shown on the screen are just Log messages with the received values being printed
It could be the case that your custom vehicle isn't responding to the throttle values, though I myself haven't done something like this before

ironclownfish, 2020-03-10:
Please make sure your vehicle derives from AirSim's CarPawn class (Set the parent class via the settings menu in the blueprint). You can check out AirSim/VehicleAdv/SuvCarPawn as an example. You'll see that it contains a VehicleMovement component which has various settings on it you might need to address too.

closed 2020-03-10 by ironclownfish