microsoft/AirSim #2448: is Airsim still maintained?

marcociara379, 2020-03-08:
Hello,
it looks lie AirSim is not maintained anymore. Can we have some clarification?

Thanks

theold190, 2020-04-03:
Checking https://github.com/microsoft/AirSim/commits/master I can see number of recent commits in April 2020. There are bug fixes, improvements, code refactoring that comes from at least 4-5 developers. So, it looks like project is still worked on.

It seems that there were few weeks without commits in February and March 2020. Checking https://github.com/microsoft/AirSim/graphs/contributors (the contribution graph during a year) you can see that there are ups and downs in repository activity. So, I guess it is a normal situation.

@marcociara379, does that answer your question? Or did you mean something else?

marcociara379, 2020-04-04:
@theold190 yes, last month was much more active than the beginning of the current year. 
Thanks!

closed 2020-04-04 by marcociara379