microsoft/AirSim #2449: Running two docker-image on one computer  Causes a problem

DN-xiajp, 2020-03-09:
Hello，
     Has anyone tried to run  two or more airsim-docker images  on one  computer?
     I fllowed the steps to build  the airsim-docker environment.And it was  ok to run the block project! 
     There was a   idea to run two airsim-docker images on one computer( Maybe it was crazy .),I met some errors when i run the second airsim image. and I couldn't find the reason. As a beginner using docker, it was as intersting but difficult problem.
      Anyway ,3Q ！

```
- 2020.03.04-09.26.56:783][  0]LogUMG: Display: Widget Class BP_SimHUDWidget_C - Loaded Fast Template.
- terminating with uncaught exception of type std::__1::system_error: bind: Address already in use
- Signal 6 caught.
- Malloc Size=131076 LargeMemoryPoolOffset=131092 
- CommonLinuxCrashHandler: Signal=6
- Malloc Size=65535 LargeMemoryPoolOffset=196655 
- [2020.03.04-09.26.58:470][  0]LogLinux: === Critical error: ===
- Unhandled Exception: SIGABRT: abort() called
- 
- [2020.03.04-09.26.58:470][  0]LogLinux: Fatal error!
```

rajat2004, 2020-03-10:
`terminating with uncaught exception of type std::__1::system_error: bind: Address already in use`

The error is due to binding to a port already being used. I think this is happening due to the `--net=host` option being used in [run_airsim_image_source.sh](https://github.com/microsoft/AirSim/blob/master/docker/run_airsim_image_source.sh#L43)
Details about what this does are discussed on this page - https://stackoverflow.com/questions/43316376/what-does-net-host-option-in-docker-command-really-do
Basic seems to be that the network of the host and docker container is the same, so binding to a port by one docker container is preventing anyone else from using it

Probably removing that and doing some careful port forwarding could work

closed 2020-03-10 by ironclownfish

DN-xiajp, 2020-03-11:
> `terminating with uncaught exception of type std::__1::system_error: bind: Address already in use`
>
> The error is due to binding to a port already being used. I think this is happening due to the [...]

Thanks a lot！