microsoft/AirSim #2452: Multiple Static Camera in Environment

jordon96, 2020-03-09:
Hi, i have created a custom environment and vehicle. I have a script that allows the vehicle to move. Now, i want to be able to have multiple static camera (kind of like traffic camera) and see through their POV and ideally capture images from those cameras while the vehicle is running on the script.

Can anyhow teach me how to do that ?

ironclownfish, 2020-03-10:
This is not a standard feature of AirSim, but you can achieve static cameras with blueprints.

Make a blueprint that derives from camera, and use the "Set View Target With Blend" node (from within the level blueprint) to switch between cameras. Capturing images from non-vehicle cameras won't be supported through the AirSim API though. I recommend doing some research on capturing view targets from within blueprints and writing them directly to disc. I'm sure someone out there has done it.

closed 2020-03-10 by ironclownfish

jordon96, 2020-03-11:
Hi, i am able to place my static cameras and switch between them, but however i can't get it to work with AirSimGameMode. When i set it to GameMode, it will override my cameras and revert back to the original camera views. i need to maintain the vehicle, letting it run on Hello_car.py script while viewing it from the static cameras that i have blended with.