microsoft/AirSim #2454: PX4 Adding sensors always crashes on play

depauwjimmy, 2020-03-12:
Hello,

I wish to use the sim with a PX4 in SITL.
I managed to make it connects and it somewhat usable with my pad.
Problem is, i need a Lidar but if i just add the configuration required Airsim always crashes instantly when i press play to start simulation.

It works fine using simpleflight so i am confused.

Airsim is last master built on Ubuntu 18.04 LTS
Unreal is 4.18.3 also built from source, project is Blocks.
PX4 is 1.10.1 SITL

I join my settings file and crash folder

[crashinfo-Blocks-pid-30785-390E881E0C03110C0006010EC8B8588F.zip](https://github.com/microsoft/AirSim/files/4325241/crashinfo-Blocks-pid-30785-390E881E0C03110C0006010EC8B8588F.zip)

[settings.zip](https://github.com/microsoft/AirSim/files/4325251/settings.zip)

rajat2004, 2020-03-12:
Hi,
In your `settings.json` file, you have specified the sensors - 
```
"Sensors": {
          "Gps": {
              "SensorType": 3,
              "Enabled" : true
          },
	  "LidarSensor1": { 
              "SensorType": 6,
              "Enabled" : true,
              "NumberOfChannels": 8,
              "RotationsPerSecond": 10,
              "PointsPerSecond": 15000,
              "X": 0, "Y": 0, "Z": 0,
              "Roll": -90, "Pitch": 0, "Yaw" : 0,
              "VerticalFOVUpper": 24,
              "VerticalFOVLower": -24,
              "HorizontalFOVStart": -1,
              "HorizontalFOVEnd": 1,
              "DrawDebugPoints": true,
              "Range": 1000
          }
      },
```

But IMU, Magnetometer and Barometer are missing from this, which causes problems since those sensors are always required in PX4
See https://github.com/microsoft/AirSim/blob/master/docs/sensors.md#configuring-vehicle-specific-sensor-list

BTW, I was thinking of adding such sensors by default so that this doesn't happen
Also, just mentioning, AirSim currently doesn't use the Lidar data with PX4, as in, doesn't send the Lidar data to the PX4 SITL. You'll have to use your own code to access the Lidar data with API and utilize it

depauwjimmy, 2020-03-19:
Ok i understand now. It's a bit counter intuitive that the defaults do not apply when adding another sensor so you suggestion is good.
I already have code to fetch lidar data but i encounter another weird problem.
With simpleflight i can instruct drone movement by code and lidar data reflect the drone change of position.

This does not appear to be the case when using PX4.
If i instruct drone movement by code to PX4, i can see the drone moving accordingly on the simulator but lidar data do not change at all.
It's like it is not made aware of the change of position because i am giving fly commands in code. 

I should probably open a new issue for this?

rajat2004, 2020-03-19:
I'm not sure exactly what problem is occurring, there was a similar problem with visualization of lidar data which has been fixed in master. Since you're also using master, that shouldn't the one.

Sensors are independent of the firmware being used, the same data is generated regardless of SimpleFlight or PX4. I have used Lidar with Ardupilot, so I can say that it's working correctly. 

Though ArduPilot doesn't use the AirSim movement APIs, so there might be something in them causing the problem. What functions are you using for controlling the movement? Also, is the visualization of the Lidar data working correctly, as in, is it moving with the drone?
Another thing could be that the code you're using is expecting the data to be relative to the drone. Default in Airsim currently is data in global coordinates. You can set `DataFrameType` to `SensorLocalFrame` for this (https://github.com/microsoft/AirSim/blob/master/docs/lidar.md)

depauwjimmy, 2020-04-01:
Settings DataFrameType fixed the issue. Thx!

closed 2020-04-01 by depauwjimmy