microsoft/AirSim #2456: make command not working in bulding unreal engine

vishal-raj15, 2020-03-14:
# 'make' command not working

i'm using ubuntu 18.0.4 and stuck on an error in building unreal engine after cloning the repo and setup ./seup.sh and ./generate.....  .sh , make command shows  : ERROR: Cannot use an old toolchain (missing ToolchainVersion.txt file, assuming version earlier than v11)

rajat2004, 2020-03-14:
Which version of Unreal Engine are you using? Check the output of `setup.sh` and `generate.sh` for any errors or failure to downloads

vishal-raj15, 2020-03-15:
unreal 4.18 as in documentation done ./setup and ./generate.... .sh files
and then make command but make shows error in ubuntu 18.0.4

rajat2004, 2020-03-16:
I have been using UE 4.18 on Ubuntu 18.04 & 16.04 both, haven't got this error, plus it's not there in the AirSim issues and google is also not giving much info.
Best thing to try would be to delete the UE folder, and start from scratch. Save the outputs from `Setup.sh` and `GenerateProjectFiles.sh` also

vishal-raj15, 2020-03-16:
ok thanks

vishal-raj15, 2020-03-17:
vishal@vishal-pc  ~/git  git clone -b 4.18
https://github.com/EpicGames/UnrealEngine.git

Cloning into 'UnrealEngine'...
Username for 'https://github.com': vishal-raj15
Password for 'https://vishal-raj15@github.com':
remote: Enumerating objects: 156, done.
remote: Counting objects: 100% (156/156), done.
remote: Compressing objects: 100% (101/101), done.
remote: Total 2345790 (delta 94), reused 102 (delta 54), pack-reused 2345634
Receiving objects: 100% (2345790/2345790), 2.74 GiB | 654.00 KiB/s, done.
Resolving deltas: 100% (1616620/1616620), done.
Checking out files: 100% (82677/82677), done.

 vishal@vishal-pc  ~/git  cd UnrealEngine

vishal@vishal-pc  ~/git  git clone -b 4.18
https://github.com/EpicGames/UnrealEngine.git

Cloning into 'UnrealEngine'...
Username for 'https://github.com': vishal-raj15
Password for 'https://vishal-raj15@github.com':
remote: Enumerating objects: 156, done.
remote: Counting objects: 100% (156/156), done.
remote: Compressing objects: 100% (101/101), done.
remote: Total 2345790 (delta 94), reused 102 (delta 54), pack-reused 2345634
Receiving objects: 100% (2345790/2345790), 2.74 GiB | 654.00 KiB/s, done.
Resolving deltas: 100% (1616620/1616620), done.
Checking out files: 100% (82677/82677), done.

 vishal@vishal-pc  ~/git  cd UnrealEngine

 vishal@vishal-pc  ~/git/UnrealEngine   4.18  ./Setup.sh

Registering git hooks... (this will override existing ones!)

Github build
Checking / downloading the latest archives
Checking dependencies (excluding Win32, Win64, Mac, Android)...
Updating dependencies:   0% (0/28391)...
Failed to download '
http://cdn.unrealengine.com/dependencies/3058682-824eeab7657e4d86a6770c1e54acda73/0dd631c1325b4b592260c6a0a8f8c010b6fc667c':
Proxy scheme not supported. (NotSupportedException)
Result: 1

but when i run sudo ./setup.sh then it is working , is it a wrong way ?

rajat2004, 2020-03-18:
Using `sudo` is not at all recommended, see https://wiki.unrealengine.com/Building_On_Linux#Foreword_before_building
You'll need to figure out your proxy issues, probably that's what causing the errors while downloading the dependencies and due to which it's saying Toolchain is missing

Tcloser, 2020-03-29:
Hi
Unfortunately I cant find this project .. what happend ?
https://github.com/EpicGames/UnrealEngine

rajat2004, 2020-03-29:
You'll have to link your Github account to the Epic Games account, see [this page](https://www.unrealengine.com/en-US/blog/updated-authentication-process-for-connecting-epic-github-accounts)

vishal-raj15, 2020-03-29:
ok thanks

vishal-raj15, 2020-03-29:
@Tcloser make command was not working for the setup of unreal engine in ubuntu because of proxy setting  , incase if you are asking to me !

closed 2020-03-29 by vishal-raj15