microsoft/AirSim #2461: No pedestrian and vehicle segmentation in CityEnviron?

Kapil-23, 2020-03-17:
Hi all,

I have downloaded CityEnviron from AirSim releases v1.2.2 - Windows.
On running script **capture_ir_segmentation.py** from AirSim >> PythonClient >> computer_vision.
**Getting below result (No Segmentation for Vehicle and Pedestrians)**

[image]
[image]
[image]
[image]

Kindly respond if any one facing same issue.

rajat2004, 2020-03-18:
Yes, I've noticed the same issue. This will probably be fixed only with a new AirSim release of which there's no timeline as of now.

jonyMarino, 2020-09-04:
Hi @rajat2004! do you know if it fixed this?

rajat2004, 2020-09-04:
Not sure, I haven't tested City since then, Will try to test the latest binaries and post, however I think it might be a problem with the environment itself.