microsoft/AirSim #2469: How to Assign Every Object a Random Segmentation ID?

flolu, 2020-03-23:
I want to give every tree in the Neighborhood environment a random segmentation id. But I have no access to the Unreal Engine Launcher. Thus I don't know the names of the objects.
[image]
So I was wondering if it simply possible to give every object a random segmentation id?

I've tried the following:
```
for i in range(100):
    success = client.simSetSegmentationObjectID(
        f'Tree_{i}', randrange(255), True)
    print(i, 'success', success)
```
... but this works only for a few trees, not all of them.

ironclownfish, 2020-03-23:
Hi @flolu 
You can see the names of all the actors in the scene by using simListSceneObjects()
I'm not sure if every single tree is it's own actor; maybe some are grouped together. But to the extent they're separate, you'll see all their names with that API :)

flolu, 2020-03-24:
That sounds really good! I've just tried it with this code:
```
import airsim

client = airsim.VehicleClient()
client.confirmConnection()
client.simListSceneObjects()
```

and I get this error:
```
Connected!
Client Ver:1 (Min Req: 1), Server Ver:1 (Min Req: 1)

Traceback (most recent call last):
  File "tree-segmentation.py", line 75, in <module>
    client.simListSceneObjects()
  File "/venv/lib/python3.7/site-packages/airsim/client.py", line 121, in simListSceneObjects
    return self.client.call('simListSceneObjects', name_regex)
  File "/venv/lib/python3.7/site-packages/msgpackrpc/session.py", line 41, in call
    return self.send_request(method, args).get()
  File "/venv/lib/python3.7/site-packages/msgpackrpc/future.py", line 45, in get
    raise error.RPCError(self._error)
msgpackrpc.error.RPCError: rpclib: server could not find function 'simListSceneObjects' with argument count 1.
```

ironclownfish, 2020-03-24:
That release contains a copy of the AirSim plugin that predates the simListSceneObjects() API. I recommend building a fresh plugin from this repo and installing it in your environment.

flolu, 2020-03-24:
I am using [Docker to run AirSim](https://github.com/microsoft/AirSim/blob/67f438629c4ca4bcf25436d5e21e9839487a6491/docs/docker_ubuntu.md#airsim-on-docker-in-linux), because I am on Ubuntu and nothing else worked for me.
Would be awesome if you could quickly show me how to do this (I am not sure what to do)
:)

flolu, 2020-03-26:
@ironclownfish I've now gone through the hassle of setting up Ubuntu 16 and building both, Unreal Engine and AirSim from source.
But I still get this error:`
```
Client Ver:1 (Min Req: 1), Server Ver:1 (Min Req: 1)

Traceback (most recent call last):
  File "objs.py", line 6, in <module>
    objects = client.simListSceneObjects()
  File "/Desktop/AirSim/PythonClient/airsim/client.py", line 124, in simListSceneObjects
    return self.client.call('simListSceneObjects', name_regex)
  File "/.local/lib/python3.5/site-packages/msgpackrpc/session.py", line 41, in call
    return self.send_request(method, args).get()
  File "/.local/lib/python3.5/site-packages/msgpackrpc/future.py", line 45, in get
    raise error.RPCError(self._error)
msgpackrpc.error.RPCError: rpclib: server could not find function 'simListSceneObjects' with argument count 1.
```

when I run this simple script:
```
import setup_path 
import airsim

client = airsim.VehicleClient()
client.confirmConnection()
objects = client.simListSceneObjects()
```

rajat2004, 2020-03-26:
@flolu I just tried the `simListSceneObjects()` and it's working correctly for me.
Which version of Unreal Engine are you using? And you're using the current AirSim master built from source and not any binaries? Please test with the default Blocks environment also if it is working

flolu, 2020-03-26:
I used [those](https://github.com/microsoft/AirSim/blob/master/docs/build_linux.md#build-unreal-engine-and-airsim) setup instructions and I think I have the same issue as @cpow-89 ([his comment](https://github.com/microsoft/AirSim/issues/2283#issuecomment-554547283)), that the environment binaries are out of date

rajat2004, 2020-03-26:
Yes, the binaries would be out-of-date l, therefore to use this API (and many other features also), you'll need to build AirSim from source

flolu, 2020-03-26:
@rajat2004 But the environment I start AirSim with (In my case via `Neighborhood.sh`) is out of date. Which means building AirSim from source will not help. Or am I not understanding something correctly?

rajat2004, 2020-03-26:
Yes, when building AirSim from source, you'll have to use the Blocks environment or others with the entire source code of env. These can be downloaded from the Unreal Marketplace, one such example is the free Landscape Mountains env. Neighborhood env is also present, though that one is not free

closed 2020-03-26 by flolu