microsoft/AirSim #2470: Can't enable fog or dust

ivannson, 2020-03-23:
I'm trying to enable different weather using the API, but I don't seem to be able to use either fog or dust. 

This is the command that I'm running:
```python
client.simSetWeatherParameter(airsim.WeatherParameter.Fog, 0.50)
```

I tried using `weather.py` script, and they don't work there either. 

I can't use weather in editor since I'm using a remote desktop to access Unreal Editor and my F10 key doesn't work there.

ironclownfish, 2020-03-23:
Weather is implemented on a per-level basis, so these affects won't always be applicable.
And don't forget you'll want to call simEnableWeather(True) before anything else.

ivannson, 2020-03-27:
Thanks @ironclownfish , I checked, and the `simEnableWeather(True)` is enabled, and I can control rain/snow/leaves. What I find weird is that a) fog/dust works on Windows b) on Linux fog and dust don't work even in the Blocks environment (other weather effects still do). 

For reference, I'm running Ubuntu 18.04, Unreal 4.22

kochsebastian, 2020-09-21:
Same problem for me. I used fog just the other day but now it's not working anymore and I didn't change anything that should affect fog

lucafei, 2020-11-12:
> Same problem for me. I used fog just the other day but now it's not working anymore and I didn't change anything that should affect fog

maybe you can try the weather window in simulation, check if it works?

kochsebastian, 2020-11-12:
After some experimenting I know what the problem was: I used to render the Engine with OpenGL, because it consumes less processing power. But only when you use Vulcan you can model fog or dust.
Vulcan is the default of course.

ivannson, 2020-11-12:
@kochsebastian Yes, that must be it! In my case it didn't work in Linux where OpenGL is the default.

closed 2020-11-12 by ivannson

fu1199, 2021-06-17:
@ivannson I am beginning to develop maps with airsim plugin in UE. Can you give me an advice how to enable vulkan for UE 4.25 under Linux?

Thanks a lot!