microsoft/AirSim #2474: How can I get position and velocity data of drone when I do HITL?

home26, 2020-03-24:
Hello

I'm going to plot the trajectory of drone when I do HITL. So when I use RC to control drone, how can I get the position and velocity data of drone? It seems that relative API is unavailable when HITL is running. 

Thank you in advance.

rajat2004, 2020-03-26:
Are you sure that the APIs are not working when using HITL? Because that shouldn't be the case, atleast from the code
Any error which you're getting while using the APIs

home26, 2020-03-26:
@rajat2004 That is the point, I do not think the APIs are still working when I use HITL. So I want to see some opinions from other poeple to support mine. I will keep this issue open to see if there are any other different points.
Thank you so much. : )

MSBGit, 2020-04-14:
I am also running HITL with Pixhawk 4/PX4/Unreal 4.24/AirSim 1.2.0 Beta and I use MAVSDK running on an NVIDIA Jetson nano to talk to the Pixhawk via a serial port to get telemetry data. In my case, I plan on moving my code onto a real drone once the simulation work is stable enough that I feel confident to fly.

MAVSDK exposes a Telemetry class (I am using MAVSDK-Python) that gives me the data that I need. Since my code is ultimately targeted to run on a real vehicle and not with AirSim, this lets me test out more of the data path in my system. 

[MAVSDK repo](https://github.com/mavlink/MAVSDK)
[MAVSDK-Python repo](https://github.com/mavlink/MAVSDK-Python)

lovettchris, 2021-06-04:
The AirSim API works fine with HITL mode PX4.  But if you want to insert a mavlink proxy you can get the information you want from mavlink messages, like GPS_RAW_INT which includes a ground speed field as well as lat/lon.  There's also LOCAL_POSITION_NED which might be updated more frequently, and you can compute speed from that yourself.