microsoft/AirSim #2475: How to record segmentation images?

thejeshk, 2020-03-24:
Hi Guys, 

I want to record segmentation images but when I tried to use the record function or press "R" button, I could save the RGB image with ImageType = 0 in setting.json with the resolution of 500x300, but when I tried to save the image in segmentation mode (ImageType = 5), the simulator did not save the images and the out put were 256 x 144 RGB images.

Could you please help me with it?
Thank you.

thejeshk, 2020-03-24:
here is my settings .json file 
{
  "SeeDocsAt": "https://github.com/Microsoft/AirSim/blob/master/docs/settings.md",
  "SettingsVersion": 1.2,
  "SimMode": "Car",
  "EnableCollisionPassthrogh": false,
  "EnableCollisions": true,
  "OriginGeopoint": {
      "Latitude": 49.456417,
      "Longitude": 7.795779
    },
  "PawnPaths": {
    "DefaultCar": {"PawnBP": "Class'/AirSim/VehicleAdv/Mower/Mower_BP.Mower_BP_C'"}
  },
  "CameraDefaults": {
		"CaptureSettings": [{
			"ImageType": 0,
			"Width": 1024,
			"Height": 576
		},
		{
			"ImageType": 3,
			"Width": 1024,
			"Height": 576
		},
		{
			"ImageType": 5,
			"Width": 1024,
			"Height": 576
		},
		{
			"ImageType": 1,
			"Width": 1024,
			"Height": 576
		}]
	},
	"SegmentationSettings": {
		"InitMethod": "CommonObjectsRandomIDs",
		"MeshNamingMethod": "OwnerName",
		"OverrideExisting": false
},
  "DefaultSensors": {
    "Barometer": {
         "SensorType": 1,
         "Enabled" : true
    },
    "Imu": {
         "SensorType": 2,
         "Enabled" : true
    },
    "Gps": {
         "SensorType": 3,
         "Enabled" : true
    },
    "Magnetometer": {
         "SensorType": 4,
         "Enabled" : true
    },
    "Distance": {
         "SensorType": 5,
         "Enabled" : true
    },
    "Lidar2": { 
         "SensorType": 6,
         "Enabled" : true,
         "NumberOfChannels": 4,
		 "DrawDebugPoints": false,
         "PointsPerSecond": 100000
    }
	
}
}

rajat2004, 2020-03-24:
Please have a look at the Recording settings- https://github.com/microsoft/AirSim/blob/master/docs/settings.md#recording

Example of using this is there in the above default settings file
If modifying this doesn't work, then there might be a deeper issue here

jonyMarino, 2020-08-10:
@thejeshk did you try rajat approach?

closed 2020-11-23 by jonyMarino