microsoft/AirSim #2476: Dynamically Change Physical Properties through API

Travestie, 2020-03-24:
Would it be possible to add API calls to change the parameters of the vehicle?
I am looking to dynamically change Drag, Mass, Max Thrust etc. for some optimization development. 
I have tried following the trail on their calls, but I have not been able to find an end.

rajat2004, 2020-03-26:
I'm not sure, but I don't think it's possible to dynamically change the physical parameters of the vehicle. Could be wrong, in that case would love to know more about it
You could have a look at the new APIs for setting the PID gains, etc. of the vehicle, which might be of use here

Travestie, 2020-03-26:
Saw those, yes they are interesting and very helpful for tuning. I am more interested in the design of a quad copter. Restarting and tweaking code instead of an optimization  function and a for loop would take a significant amount of time for the flight envelope to appear. Even with the optimization function, it would still probably take a weekend or longer.

I did manage to compile a significant amount of code revealing the physical properties to the API. Untested right now, but will get back with an answer.

jonyMarino, 2020-08-10:
@Travestie How did you go?

geniusyko, 2020-09-09:
@Travestie How is the progress so far?

closed 2022-03-31 by jonyMarino