microsoft/AirSim #2477: Plan for a new Linux release?

Jaeyoung-Lim, 2020-03-25:
The Airsim releases are already a year old.

Are there any plans to release a new Airsim release for Linux?

madratman, 2020-03-26:
Hey @Jaeyoung-Lim. Yes, we're planning for a new release very soon! We're looking to merge most of the open PRs before that. Stay tuned, and thanks for your patience.

Jaeyoung-Lim, 2020-03-26:
@madratman Thanks, do you have a ETA on when this will happen?

madratman, 2020-03-26:
We're aiming < 1 week.

Jaeyoung-Lim, 2020-03-26:
@madratman Awesome! Thanks

Tcloser, 2020-03-29:
> We're aiming < 1 week.
Hi 
https://github.com/EpicGames/UnrealEngine
Unfortunately I cant find this project .. what happend ? And whether we need UnrealEngine in new release?

Jaeyoung-Lim, 2020-03-29:
@Tcloser Please don't hijack the topic of this issue

madratman, 2020-03-31:
@Jaeyoung-Lim, we just made https://github.com/microsoft/AirSim/releases/tag/v1.3.0-linux with only Blocks for now. I noticed in your PX4 PR, Blocks would be enough to unblock. 
We're adding other environment binaries incrementally. 
Thanks for your patience!

Jaeyoung-Lim, 2020-03-31:
@madratman Thanks! However, I am unable to run the environment locally (I can run 1.2)

The log is as the following
```
$ ./Blocks.sh 
Increasing per-process limit of core file size to infinity.
LogConsoleResponse: Display: Failed to find resolution value strings in scalability ini. Falling back to default.
LogPakFile: Display: Found Pak file ../../../Blocks/Content/Paks/Blocks-LinuxNoEditor.pak attempting to mount.
LogPakFile: Display: Mounting pak file ../../../Blocks/Content/Paks/Blocks-LinuxNoEditor.pak.
LogPlatformFile: Using cached read wrapper
LogTaskGraph: Started task graph with 5 named threads and 14 total threads with 3 sets of task threads.
LogStats: Stats thread started at 0.038449
LogICUInternationalization: ICU TimeZone Detection - Raw Offset: +1:00, Platform Override: ''
LogPluginManager: Mounting plugin AirSim
LogPluginManager: Mounting plugin PhysXVehicles
LogPluginManager: Mounting plugin Paper2D
LogPluginManager: Mounting plugin AISupport
LogPluginManager: Mounting plugin LightPropagationVolume
LogPluginManager: Mounting plugin AnimationSharing
LogPluginManager: Mounting plugin SignificanceManager
LogPluginManager: Mounting plugin UObjectPlugin
LogPluginManager: Mounting plugin AssetManagerEditor
LogPluginManager: Mounting plugin FacialAnimation
LogPluginManager: Mounting plugin DatasmithContent
LogPluginManager: Mounting plugin VariantManagerContent
LogPluginManager: Mounting plugin AlembicImporter
LogPluginManager: Mounting plugin GeometryCache
LogPluginManager: Mounting plugin AutomationUtils
LogPluginManager: Mounting plugin ScreenshotTools
LogPluginManager: Mounting plugin BackChannel
LogPluginManager: Mounting plugin CharacterAI
LogPluginManager: Mounting plugin PlatformCrypto
LogPluginManager: Mounting plugin AvfMedia
LogPluginManager: Mounting plugin ImgMedia
LogPluginManager: Mounting plugin MediaCompositing
LogPluginManager: Mounting plugin WmfMedia
LogPluginManager: Mounting plugin TcpMessaging
LogPluginManager: Mounting plugin UdpMessaging
LogPluginManager: Mounting plugin ActorSequence
LogPluginManager: Mounting plugin NetcodeUnitTest
LogPluginManager: Mounting plugin OnlineSubsystem
LogPluginManager: Mounting plugin OnlineSubsystemNull
LogPluginManager: Mounting plugin OnlineSubsystemUtils
LogPluginManager: Mounting plugin LauncherChunkInstaller
LogPluginManager: Mounting plugin AndroidPermission
LogPluginManager: Mounting plugin AppleImageUtils
LogPluginManager: Mounting plugin ArchVisCharacter
LogPluginManager: Mounting plugin AssetTags
LogPluginManager: Mounting plugin AudioCapture
LogPluginManager: Mounting plugin CableComponent
LogPluginManager: Mounting plugin CustomMeshComponent
LogPluginManager: Mounting plugin EditableMesh
LogPluginManager: Mounting plugin ExampleDeviceProfileSelector
LogPluginManager: Mounting plugin LinuxDeviceProfileSelector
LogPluginManager: Mounting plugin LocationServicesBPLibrary
LogPluginManager: Mounting plugin MobilePatchingUtils
LogPluginManager: Mounting plugin ProceduralMeshComponent
LogPluginManager: Mounting plugin RuntimePhysXCooking
LogPluginManager: Mounting plugin SteamVR
LogPluginManager: Mounting plugin WebMMoviePlayer
LogPluginManager: Mounting plugin WebMMedia
LogInit: Using libcurl 7.65.3-DEV
LogInit:  - built for x86_64-unknown-linux-gnu
LogInit:  - supports SSL with OpenSSL/1.1.1c
LogInit:  - supports HTTP deflate (compression) using libz 1.2.8
LogInit:  - other features:
LogInit:      CURL_VERSION_SSL
LogInit:      CURL_VERSION_LIBZ
LogInit:      CURL_VERSION_IPV6
LogInit:      CURL_VERSION_ASYNCHDNS
[... 506 lines cut ...]
0x0000000007081482 Blocks!UnknownFunction(0x6e81481)
0x00000000043b5266 Blocks!UnknownFunction(0x41b5265)
0x00000000043bfb73 Blocks!UnknownFunction(0x41bfb72)
0x00000000043c11ba Blocks!UnknownFunction(0x41c11b9)
0x00000000043aca8a Blocks!UnknownFunction(0x41aca89)
0x00000000044254d8 Blocks!UnknownFunction(0x42254d7)
0x00000000043f0c92 Blocks!UnknownFunction(0x41f0c91)
0x00000000043f0f3c Blocks!UnknownFunction(0x41f0f3b)
0x00000000043b43f1 Blocks!UnknownFunction(0x41b43f0)
0x00000000043f39a6 Blocks!UnknownFunction(0x41f39a5)
0x00000000043f1617 Blocks!UnknownFunction(0x41f1616)
0x00000000043f126b Blocks!UnknownFunction(0x41f126a)
0x00000000044ec8c5 Blocks!UnknownFunction(0x42ec8c4)
0x00000000026eb0fb Blocks!UnknownFunction(0x24eb0fa)
0x00000000026e4f5c Blocks!UnknownFunction(0x24e4f5b)
0x0000000005f8f08d Blocks!UnknownFunction(0x5d8f08c)
0x00007f4b8e0feb97 libc.so.6!__libc_start_main(+0xe6)
0x00000000026d8029 Blocks!UnknownFunction(0x24d8028)

[2020.03.31-09.16.15:930][  0]LogExit: Executing StaticShutdownAfterError
Malloc Size=52148 LargeMemoryPoolOffset=283268 
Engine crash handling finished; re-raising signal 11 for the default handler. Good bye.
Segmentation fault (core dumped)

```

Jaeyoung-Lim, 2020-03-31:
@madratman Thanks! However, the Blocks environment crashes when I try to run `Blocks.sh` Any idea why this is happening?

The log is as the following: 
```
$ ./Blocks.sh 
Increasing per-process limit of core file size to infinity.
LogConsoleResponse: Display: Failed to find resolution value strings in scalability ini. Falling back to default.
LogPakFile: Display: Found Pak file ../../../Blocks/Content/Paks/Blocks-LinuxNoEditor.pak attempting to mount.
LogPakFile: Display: Mounting pak file ../../../Blocks/Content/Paks/Blocks-LinuxNoEditor.pak.
LogPlatformFile: Using cached read wrapper
LogTaskGraph: Started task graph with 5 named threads and 14 total threads with 3 sets of task threads.
LogStats: Stats thread started at 0.027209
LogICUInternationalization: ICU TimeZone Detection - Raw Offset: +1:00, Platform Override: ''
LogPluginManager: Mounting plugin AirSim
LogPluginManager: Mounting plugin PhysXVehicles
LogPluginManager: Mounting plugin Paper2D
LogPluginManager: Mounting plugin AISupport
LogPluginManager: Mounting plugin LightPropagationVolume
LogPluginManager: Mounting plugin AnimationSharing
LogPluginManager: Mounting plugin SignificanceManager
LogPluginManager: Mounting plugin UObjectPlugin
LogPluginManager: Mounting plugin AssetManagerEditor
LogPluginManager: Mounting plugin FacialAnimation
LogPluginManager: Mounting plugin DatasmithContent
LogPluginManager: Mounting plugin VariantManagerContent
LogPluginManager: Mounting plugin AlembicImporter
LogPluginManager: Mounting plugin GeometryCache
LogPluginManager: Mounting plugin AutomationUtils
LogPluginManager: Mounting plugin ScreenshotTools
LogPluginManager: Mounting plugin BackChannel
LogPluginManager: Mounting plugin CharacterAI
LogPluginManager: Mounting plugin PlatformCrypto
LogPluginManager: Mounting plugin AvfMedia
LogPluginManager: Mounting plugin ImgMedia
LogPluginManager: Mounting plugin MediaCompositing
LogPluginManager: Mounting plugin WmfMedia
LogPluginManager: Mounting plugin TcpMessaging
LogPluginManager: Mounting plugin UdpMessaging
LogPluginManager: Mounting plugin ActorSequence
LogPluginManager: Mounting plugin NetcodeUnitTest
LogPluginManager: Mounting plugin OnlineSubsystem
LogPluginManager: Mounting plugin OnlineSubsystemNull
LogPluginManager: Mounting plugin OnlineSubsystemUtils
LogPluginManager: Mounting plugin LauncherChunkInstaller
LogPluginManager: Mounting plugin AndroidPermission
LogPluginManager: Mounting plugin AppleImageUtils
LogPluginManager: Mounting plugin ArchVisCharacter
LogPluginManager: Mounting plugin AssetTags
LogPluginManager: Mounting plugin AudioCapture
LogPluginManager: Mounting plugin CableComponent
LogPluginManager: Mounting plugin CustomMeshComponent
LogPluginManager: Mounting plugin EditableMesh
LogPluginManager: Mounting plugin ExampleDeviceProfileSelector
LogPluginManager: Mounting plugin LinuxDeviceProfileSelector
LogPluginManager: Mounting plugin LocationServicesBPLibrary
LogPluginManager: Mounting plugin MobilePatchingUtils
LogPluginManager: Mounting plugin ProceduralMeshComponent
LogPluginManager: Mounting plugin RuntimePhysXCooking
LogPluginManager: Mounting plugin SteamVR
LogPluginManager: Mounting plugin WebMMoviePlayer
LogPluginManager: Mounting plugin WebMMedia
LogInit: Using libcurl 7.65.3-DEV
LogInit:  - built for x86_64-unknown-linux-gnu
LogInit:  - supports SSL with OpenSSL/1.1.1c
LogInit:  - supports HTTP deflate (compression) using libz 1.2.8
LogInit:  - other features:
LogInit:      CURL_VERSION_SSL
LogInit:      CURL_VERSION_LIBZ
LogInit:      CURL_VERSION_IPV6
LogInit:      CURL_VERSION_ASYNCHDNS
[... 666 lines cut ...]
0x00000000043ff6f6 Blocks!UnknownFunction(0x41ff6f5)
0x00000000043ff033 Blocks!UnknownFunction(0x41ff032)
0x00000000044030e9 Blocks!UnknownFunction(0x42030e8)
0x00000000044068fd Blocks!UnknownFunction(0x42068fc)
0x0000000002c0d542 Blocks!UnknownFunction(0x2a0d541)
0x000000000456de82 Blocks!UnknownFunction(0x436de81)
0x0000000003d56529 Blocks!UnknownFunction(0x3b56528)
0x0000000003a4fd2a Blocks!UnknownFunction(0x384fd29)
0x0000000003e70e00 Blocks!UnknownFunction(0x3c70dff)
0x0000000003e72657 Blocks!UnknownFunction(0x3c72656)
0x0000000002c63838 Blocks!UnknownFunction(0x2a63837)
0x0000000002c62273 Blocks!UnknownFunction(0x2a62272)
0x000000000457653c Blocks!UnknownFunction(0x437653b)
0x00000000045afa82 Blocks!UnknownFunction(0x43afa81)
0x0000000002cd38e7 Blocks!UnknownFunction(0x2ad38e6)
0x0000000002caec69 Blocks!UnknownFunction(0x2aaec68)
0x00007fcad02ee6db libpthread.so.0!UnknownFunction(0x76da)
0x00007fcaced2688f libc.so.6!clone(+0x3e)

[2020.03.31-09.19.35:910][  2]LogExit: Executing StaticShutdownAfterError
Malloc Size=105624 LargeMemoryPoolOffset=364008 
Engine crash handling finished; re-raising signal 11 for the default handler. Good bye.
Segmentation fault (core dumped)

```

rajat2004, 2020-03-31:
@Jaeyoung-Lim It is working correctly for me on Ubuntu 18.04. There are quite a lot of memory allocation errors in your output, maybe try keeping an eye on `nvidia-smi` while running the binary? You could also try using the `-windowed`  and `-ResX`,`-ResY` options to reduce the rendering and see if it helps
Don't think memory should be an issue, especially since there's a line in between which says it's reducing the Texture memory to 1000MB, but doesn't hurt to check

madratman, 2020-03-31:
@Jaeyoung-Lim does `./Blocks.sh -windowed -opengl4` makes any difference?

Jaeyoung-Lim, 2020-04-01:
@madratman That actually made a difference. What does the command arguments change?

madratman, 2020-04-02:
It changes the rendering API unreal uses from vulkan to opengl. 
With ue 4.21, unreal switched to vulkan in linux by default for better performance. 
However, we have seen vulkan use more memory than opengl at times.  
unreal is still resolving issues wrt the vulkan transition (which might explain higher mem usage) and we suspect it might be complete happen with ue 4.25. 
as a fallback, they let you use opengl backend with that command line arg
in general, here's a complete list - https://docs.unrealengine.com/en-US/Programming/Basics/CommandLineArguments/index.html

saihv, 2020-04-02:
As a side note, the size of the window also determines the resolution UE draws the world at. Smaller the window, lower the memory usage and higher the average FPS. It can either be resized manually or through the flags ResX and ResY: `./Blocks.sh -windowed -ResX *** -ResY ***`

Jaeyoung-Lim, 2020-04-02:
@madratman Thanks for clarifying, my laptop is also at the edge of running Airsim, so I will also try testing on my desktop.

Jaeyoung-Lim, 2020-04-08:
This release currently doesn't work with the PX4 master of today. It fails with the following errors
```
pxh> WARN [commander] Takeoff denied! Please disarm and retry
WARN [PreFlightCheck] Preflight Fail: Compass Sensor #0 missing
WARN [PreFlightCheck] Primary compass not found
WARN [PreFlightCheck] Preflight Fail: Accel Sensor #0 missing
WARN [PreFlightCheck] Primary accelerometer not found
WARN [PreFlightCheck] Preflight Fail: Gyro Sensor #0 missing
WARN [PreFlightCheck] Primary gyro not found
WARN [PreFlightCheck] Preflight Fail: Baro Sensor #0 missing
```

This seems to be introduced by https://github.com/PX4/Firmware/pull/14143

@madratman FYI

rajat2004, 2020-04-08:
> What would be a good way to keep both is some sort of a CI so that this kind of regression can be detected not only when there are releases?

Any ideas for a more thorough CI would be great! Currently, Travis CI for AirSim is limited to building the AirLib code and doesn't even involve UE and the engine-related code. Therefore, many times, problems with that get missed. UE is very heavy, just for Linux takes ~1hr to build on my system, and takes up more than 40-50 GB. Building on Travis just won't be feasible, never mind running which will require GPU

I think manual testing from time-to-time and issues from users seems to be the only way to me right now

Jaeyoung-Lim, 2020-04-08:
@rajat2004 Would it be possible to run Airlib without UE? that would simplify the CI problem greatly

rajat2004, 2020-04-08:
AFAIK don't think that's possible. All the physics, timers, etc are done by UE
But @madratman @saihv @sytelus would be much more knowledgeable about this

madratman, 2020-04-08:
I think it's possible, as AirLib is a standalone cpp lib. 
Looking at the code, I think we need a json file for px4 sitl, setup px4 sitl, and make a standalone c++ script which runs the bare minimum for px4 sitl . 
something on the lines of 

```bash
# prereq
pip3 install --user empy toml;

# clone PX4
mkdir PX4 && cd $_;
git clone https://github.com/PX4/Firmware.git;
cd Firmware;

# run airlib without ue
touch /home/$USER/Documents/AirSim/settings.json
echo "{
    "SettingsVersion": 1.2,
    "SimMode": "Multirotor",
    "Vehicles": {
        "PX4Vehicle": {
            "VehicleType": "PX4Multirotor",
            "UseSerial": false,
            "UseTcp": true,
            "TcpPort": 4560,
            "ControlPort": 14580,
            "params": {
                "NAV_RCL_ACT": 0,
                "NAV_DLL_ACT": 0,
                "LPE_LAT": 47.641468,
                "LPE_LON": -122.140165,
                "COM_OBL_ACT": 1
            }
        }
    }
}">> /home/$USER/Documents/AirSim/settings.json;

# Run PX4 SITL
make px4_sitl_default none_iris

# Run AirLib

## todo 
# make a standalone cpp file / gtest which will make px4 params and create px4 api

# auto pixhawk_params = MultiRotorParamsFactory::createConfig(AirSimSettings::singleton().getVehicleSetting("PX4Vehicle"), 
#     std::make_shared<SensorFactory>());
# auto multirotor_api = pixhawk->createMultirotorApi();

# do things with multirotor_api
# catch errors 
```

Jaeyoung-Lim, 2020-04-08:
@madratman In the meantime, would it be possible to do a new release including https://github.com/microsoft/AirSim/pull/2549 ?

bys1123, 2020-04-08:
@Jaeyoung-Lim Did you tested it, I‘m trying to test it, but not finished yet.

madratman, 2020-04-08:
@Jaeyoung-Lim can you clarify if it's breaking both SITL and HITL? 
Yes, we can make a new release post merging it

TSC21, 2020-04-08:
> @Jaeyoung-Lim can you clarify if it's breaking both SITL and HITL?
> Yes, we can make a new release post merging it

It should since the pipeline for the sensors is the same.

TSC21, 2020-04-08:
> I think it's possible, as AirLib is a standalone cpp lib.
> Looking at the code, I think we need a json file for px4 sitl, setup px4 sitl, and make a standalone c++ script which runs the bare minimum for px4 sitl .
> something on the lines of
> 
> ```shell
> # prereq
> pip3 install --user empy toml;
> 
> # clone PX4
> mkdir PX4 && cd $_;
> git clone https://github.com/PX4/Firmware.git;
> cd Firmware;
> 
> # run airlib without ue
> touch /home/$USER/Documents/AirSim/settings.json
> echo "{
>     "SettingsVersion": 1.2,
>     "SimMode": "Multirotor",
>     "Vehicles": {
>         "PX4Vehicle": {
>             "VehicleType": "PX4Multirotor",
>             "UseSerial": false,
>             "UseTcp": true,
>             "TcpPort": 4560,
>             "ControlPort": 14580,
>             "params": {
>                 "NAV_RCL_ACT": 0,
>                 "NAV_DLL_ACT": 0,
>                 "LPE_LAT": 47.641468,
>                 "LPE_LON": -122.140165,
>                 "COM_OBL_ACT": 1
>             }
>         }
>     }
> }">> /home/$USER/Documents/AirSim/settings.json;
> 
> 
> # Run PX4 SITL
> make px4_sitl_default none_iris
> 
> # Run AirLib
> 
> ## todo 
> # make a standalone cpp file / gtest which will make px4 params and create px4 api
> 
> # auto px4_params = MultiRotorParamsFactory::createConfig(AirSimSettings::singleton().getVehicleSetting("PX4Vehicle"), std::make_shared<SensorFactory>());
> # auto multirotor_api = px4_params->createMultirotorApi();
> 
> # do things with multirotor_api
> # catch errors 
> ```

Why not use a Docker container with UE already in place: https://hub.docker.com/r/connorlanigan/unrealengine-docker/? Or even create a Dockerfile which extends from this one with the required tools and use it in CI?

madratman, 2020-04-08:
@TSC21, we have something on that lines here https://microsoft.github.io/AirSim/docker_ubuntu/ and here https://github.com/microsoft/AirSim/tree/master/docker. I believe [ue4-docker](https://adamrehn.com/docs/ue4-docker/configuration/configuring-linux) is more mature than the one you linked. 
However, the issue is we can't compile UE from source in something like travis as it takes a fairly long time to do so. 
Instead, what is better is to have a dedicate gpu based vm, and make a [installed build of Unreal](https://docs.unrealengine.com/en-US/Programming/Deployment/UsinganInstalledBuild/index.html) on that machine, so time isn't wasted to compile UE on each CI build.

madratman, 2020-04-08:
> @Jaeyoung-Lim Did you tested it, I‘m trying to test it, but not finished yet.

It works well for me on linux.

madratman, 2020-04-08:
https://github.com/microsoft/AirSim/releases/tag/v1.3.1-linux is here

bys1123, 2020-04-09:
@madratman Thanks, tried windows version, is that only can run in full screen now?

bys1123, 2020-04-09:
```
ERROR [px4_work_queue] setting sched params for wq:lp_default failed (134)
INFO  [mavlink] mode: Normal, data rate: 4000000 B/s on udp port 18570 remote port 14550
ERROR [sensors] Accel #0 fail:  TIMEOUT!
ERROR [sensors] Sensor Accel #0 failed. Reconfiguring sensor priorities.
WARN  [sensors] Remaining sensors after failover event 0: Accel #0 priority: 1
WARN  [sensors] Remaining sensors after failover event 0: Accel #1 priority: 0
WARN  [sensors] Remaining sensors after failover event 0: Accel #2 priority: 0
ERROR [sensors] Accel #0 fail:  TIMEOUT!
ERROR [sensors] Sensor Accel #0 failed. Reconfiguring sensor priorities.
WARN  [sensors] Remaining sensors after failover event 0: Accel #0 priority: 1
WARN  [sensors] Remaining sensors after failover event 0: Accel #1 priority: 0
WARN  [sensors] Remaining sensors after failover event 0: Accel #2 priority: 0
ERROR [sensors] Accel #0 fail:  TIMEOUT!
```

Seems windows version still not working.

bys1123, 2020-04-09:
[image]

madratman, 2020-04-09:
I think the doc says to start px4 sitl first.
Once you see - waiting for simulator to connect in the terminal, then hit run on unreal engine.

madratman, 2020-04-09:
You can pass windowed, resx and resy params as mentioned earlier in the thread for non full screen.

bys1123, 2020-04-09:
@madratman Thanks, `start Blocks -windowed -ResX=640 -ResY=480` Works, need to use these params together, different with 1.3.0.

bys1123, 2020-04-09:
@madratman I tried different timing open px4 sitl and airsim, all get this error.

Jaeyoung-Lim, 2020-04-09:
@madratman When running with px4 sitl, airsim crashes on my machine. Were you able to get something like QGroundcontrol to control the vehicle?

```
[2020.04.09-08.43.40:895][  0]LogActor: Warning: BP_CameraDirector_C /Game/FlyingCPP/Maps/FlyingExampleMap.FlyingExampleMap:PersistentLevel.CameraDirector has natively added scene component(s), but none of them were set as the actor's RootComponent - picking one arbitrarily
[2020.04.09-08.43.40:921][  0]LogTemp: Disconnecting mavlink vehicle
Disconnecting mavlink vehicle
Waiting for mavlink vehicle...
Disconnecting mavlink vehicle
Waiting for TCP connection on port 4560, local IP 127.0.0.1
[2020.04.09-08.43.40:921][  0]LogTemp: Waiting for mavlink vehicle...
[2020.04.09-08.43.40:921][  0]LogTemp: Disconnecting mavlink vehicle
[2020.04.09-08.43.40:921][  0]LogTemp: Waiting for TCP connection on port 4560, local IP 127.0.0.1
[2020.04.09-08.43.40:921][  0]LogStreaming: Error: Couldn't find file for package /AirSim/Weather/WeatherFX/WeatherActor requested by async loading code. NameToLoad: /AirSim/Weather/WeatherFX/WeatherActor
Connected to SITL over TCP.
[2020.04.09-08.43.40:921][  0]LogTemp: Connected to SITL over TCP.
Connecting to PX4 Control UDP port 14581, local IP 127.0.0.1, remote IP...
[2020.04.09-08.43.40:921][  0]LogStreaming: Error: Found 0 dependent packages...
[2020.04.09-08.43.40:921][  0]LogTemp: Connecting to PX4 Control UDP port 14581, local IP 127.0.0.1, remote IP...
[2020.04.09-08.43.40:921][  0]LogTemp: Warning: Warning, WeatherAPI got invalid weather actor class!
[2020.04.09-08.43.40:922][  0]LogStreaming: Error: Couldn't find file for package /AirSim/Weather/UI/MenuActor requested by async loading code. NameToLoad: /AirSim/Weather/UI/MenuActor
[2020.04.09-08.43.40:922][  0]LogStreaming: Error: Found 0 dependent packages...
[2020.04.09-08.43.40:922][  0]LogTemp: Warning: Warning, WeatherAPI got invalid menu actor class!
[2020.04.09-08.43.40:922][  0]LogTemp: PX4
PX4
[2020.04.09-08.43.40:922][  0]LogUMG: Display: Widget Class BP_SimHUDWidget_C - Loaded Fast Template.
received first heartbeat
[2020.04.09-08.43.40:923][  0]LogTemp: received first heartbeat
[2020.04.09-08.43.40:923][  0]LogLoad: Took 0.253329 seconds to LoadMap(/Game/FlyingCPP/Maps/FlyingExampleMap)
[2020.04.09-08.43.40:925][  0]LogSlate: Took 0.002072 seconds to synchronously load lazily loaded font '../../../Engine/Content/EngineFonts/Faces/RobotoBold.ufont' (160K)
[2020.04.09-08.43.40:951][  0]LogSlate: Took 0.002341 seconds to synchronously load lazily loaded font '../../../Engine/Content/Slate/Fonts/Roboto-Regular.ttf' (155K)
[2020.04.09-08.43.40:955][  0]LogRHI: Display: ShaderPipelineCache: Paused Batching. 1
[2020.04.09-08.43.40:955][  0]LogRHI: Display: ShaderPipelineCache: Resumed Batching. 0
[2020.04.09-08.43.40:955][  0]LogRHI: Display: ShaderPipelineCache: Batching Resumed.
[2020.04.09-08.43.40:955][  0]LogLoad: (Engine Initialization) Total time: 4.59 seconds
[2020.04.09-08.43.40:967][  0]LogRenderer: Reallocating scene render targets to support 1920x1080 Format 10 NumSamples 1 (Frame:1).
[2020.04.09-08.43.40:970][  0]LogRHI: Using OpenGL program LRU cache: 0
[2020.04.09-08.43.41:069][  0]LogContentStreaming: Texture pool size now 1000 MB
[2020.04.09-08.43.41:180][  1]LogRenderer: Reallocating scene render targets to support 128x128 Format 10 NumSamples 1 (Frame:1).
[2020.04.09-08.43.41:267][  2]LogRenderer: Reallocating scene render targets to support 1920x1080 Format 10 NumSamples 1 (Frame:2).
Ground control connected over UDP.
[2020.04.09-08.43.42:922][ 40]LogTemp: Ground control connected over UDP.
Enabling lockstep mode
[2020.04.09-08.43.43:031][ 42]LogTemp: Enabling lockstep mode
timeout on HilActuatorControlsMessage, resetting lock step mode
[2020.04.09-08.43.43:134][ 44]LogTemp: timeout on HilActuatorControlsMessage, resetting lock step mode
Enabling lockstep mode
[2020.04.09-08.43.43:139][ 44]LogTemp: Enabling lockstep mode
timeout on HilActuatorControlsMessage, resetting lock step mode
[2020.04.09-08.43.43:244][ 46]LogTemp: timeout on HilActuatorControlsMessage, resetting lock step mode
Enabling lockstep mode
[2020.04.09-08.43.43:246][ 46]LogTemp: Enabling lockstep mode
timeout on HilActuatorControlsMessage, resetting lock step mode
[2020.04.09-08.43.43:351][ 49]LogTemp: timeout on HilActuatorControlsMessage, resetting lock step mode
Enabling lockstep mode
[2020.04.09-08.43.43:354][ 49]LogTemp: Enabling lockstep mode
timeout on HilActuatorControlsMessage, resetting lock step mode
[2020.04.09-08.43.43:459][ 51]LogTemp: timeout on HilActuatorControlsMessage, resetting lock step mode
Enabling lockstep mode
[2020.04.09-08.43.43:463][ 51]LogTemp: Enabling lockstep mode
timeout on HilActuatorControlsMessage, resetting lock step mode
[2020.04.09-08.43.43:567][ 54]LogTemp: timeout on HilActuatorControlsMessage, resetting lock step mode
Enabling lockstep mode
[2020.04.09-08.43.43:570][ 54]LogTemp: Enabling lockstep mode
timeout on HilActuatorControlsMessage, resetting lock step mode
[2020.04.09-08.43.43:674][ 56]LogTemp: timeout on HilActuatorControlsMessage, resetting lock step mode
Enabling lockstep mode
[2020.04.09-08.43.43:679][ 56]LogTemp: Enabling lockstep mode
timeout on HilActuatorControlsMessage, resetting lock step mode
[2020.04.09-08.43.43:784][ 58]LogTemp: timeout on HilActuatorControlsMessage, resetting lock step mode
Enabling lockstep mode
[2020.04.09-08.43.43:786][ 58]LogTemp: Enabling lockstep mode
timeout on HilActuatorControlsMessage, resetting lock step mode
[2020.04.09-08.43.43:891][ 61]LogTemp: timeout on HilActuatorControlsMessage, resetting lock step mode
Enabling lockstep mode
[2020.04.09-08.43.43:896][ 61]LogTemp: Enabling lockstep mode
terminating with uncaught exception of type std::runtime_error: UdpClientPort socket send failed with error: 9
```

bys1123, 2020-04-09:
@Jaeyoung-Lim Do you have the error information on PX4 side?

Jaeyoung-Lim, 2020-04-09:
@bys1123 there is no error on the px4 side

madratman, 2020-04-09:
I am seeing both the error logs now. I am not sure how come they didn't show up earlier on my machine. 
Looking into it. 

[image]

Jaeyoung-Lim, 2020-04-09:
Seems like Airsim process exits before that occurs in my case...

madratman, 2020-04-09:
Can you try running PX4 before AirSim. Not directly related: but I am definitely noticing the freeze on exiting unreal issue with PX4 - https://github.com/microsoft/AirSim/issues/2112. Maybe @lovettchris can advice w.r.t. that. 

Debugging a bit, I think this is what's happening (cc @Jaeyoung-Lim @TSC21 @lovettchris)
- The first few calls of `sendHilSensor` do run fine. (It seems to me when I was testing #2549, I exited the process way to quickly before the error messages starting popping up.)
- Now, after the first few successful, calls, the [`if (!received_actuator_controls_)`](https://github.com/microsoft/AirSim/blob/master/AirLib/include/vehicles/multirotor/firmwares/mavlink/MavLinkMultirotorApi.hpp#L1317) becomes true (as well, at least, I am not sending any control commands), and the function simply returns on L1319. 

Now, it does that a few times and keeps on returning, without updating the `last_hil_sensor_time_` (which you can see is happening in [L1326](https://github.com/microsoft/AirSim/blob/master/AirLib/include/vehicles/multirotor/firmwares/mavlink/MavLinkMultirotorApi.hpp#L1326). 

By the time this has happened for >100 milliseconds, we can see why it's failing - the condition on L1311 is - [`if (last_hil_sensor_time_ + 100000 < now)`](https://github.com/microsoft/AirSim/blob/master/AirLib/include/vehicles/multirotor/firmwares/mavlink/MavLinkMultirotorApi.hpp#L1311) is now true, aka last timestamp of hil msg is too old, and hence the error log on the terminal. 

Interestingly enough, we actually still return on the function as (at least in my case), `!received_actuator_controls_` is still true - and therefore no message is sent. 

I am actually surprised how this wasn't failing earlier? Or perhaps people who use PX4 always have `received_actuator_controls_` = true?

Jaeyoung-Lim, 2020-04-09:
@madratman Awesome, thanks for the investigation.

I have also tried starting px4 before but still results in the same. I believe no one really noticed since most people using Airsim used v1.8.2(Nov 21, 2018) of PX4 as written in the documentation. 

I will try on a more powerful desktop to run it with the UE4 editor

madratman, 2020-04-09:
Alright, with the PR, I see ~38 messages being sent successfully. 

However, post that we start returning from the function when `!received_actuator_controls_` condition becomes true. 
The comment in the code says `// drop this one since we are in LOCKSTEP mode and we have not yet received the HilActuatorControlsMessage.`. 
I am assuming this the desired behaviour?
So, after the first 40 or so messages, it doesn't send out anything, until a `MavLinkHilControls` cmd is received.

madratman, 2020-04-09:
Added a minimal prototype for the AirLib minus UE + PX4 SITL idea from https://github.com/microsoft/AirSim/issues/2477#issuecomment-611082708 in  https://github.com/microsoft/AirSim/pull/2558

madratman, 2020-04-10:
> terminating with uncaught exception of type std::runtime_error: UdpClientPort socket send failed with error: 9

I haven't been able to fix the throw, but it's happening in `sendHeartbeat`. 
I am not sure what error code 9 means?   https://community.particle.io/t/udp-error-codes-listed-anywhere/18775/3 says `NOTUP -9 // Interface is not currently Up`

[image]

Jaeyoung-Lim, 2020-04-14:
@madratman Could we reopen this issue so that it gets better visibility?

madratman, 2020-04-14:
Sure. 
I've updated the docs to reflect that latest release of airsim binaries works with latest release tag of px4 (1.10.1)

clintlombard, 2020-09-08:
The latest PX4 release ([v1.11.0](https://github.com/PX4/Firmware/releases/tag/v1.11.0)) no longer works with AirSim in SITL, although v1.10.1 and v1.10.2 did.

#3010 might also be related to this issue in the broader sense of PX4 compatibility.

Jaeyoung-Lim, 2020-09-08:
@madratman From a previous discussion, I briefly recall that we were discussing on detecting PX4 related regressions using a CI. Has there been any progress on this front?

krisklau, 2020-10-21:
Any updates on px4 v1.11 compatibility?

Jaeyoung-Lim, 2020-10-22:
@jonyMarino  Are there any activity to resolve this and preventing it from happening again?

jonyMarino, 2020-10-22:
Hi @Jaeyoung-Lim! Compatibility with PX4 is a major concern. We are not working on this right now, but I will update if no one from the community posts a pull request.

Jaeyoung-Lim, 2020-10-22:
@jonyMarino This has been broken for over a month, I think it is safe to assume no one is trying to fix it in the community?

EivindYN, 2020-11-13:
Any progress on the PX4 1.11 compatibility?

jonyMarino, 2020-11-13:
Hi @EivindYN! We will be working on this soon!

EivindYN, 2020-11-13:
> Hi @EivindYN! We will be working on this soon!

Which engine version will you be using, 4.24 or 4.25? So that I can prepare myself for the eventual merge.

jonyMarino, 2020-11-20:
Yesterday, a PR was merged ( #3156 ) to solve the communication error with the latest PX4 version. Good luck!

Jaeyoung-Lim, 2020-11-20:
@jonyMarino In the PR it seems that it was tested against HITL. Is it also functional for SITL?

If this is functional, would it be possible to do a release for Linux?

jonyMarino, 2020-11-20:
Yes. It was the first intention of that PR. But ended solving the SITL issue too.
We have to decide about another thing before releasing a new release. We will announce.

lovettchris, 2020-11-23:
By the way there is an open SITL problem on Windows if you are using "WSL2" to run PX4.

- https://github.com/PX4/PX4-Autopilot/issues/16252

but WS1 will work, or you can edit the simulator.cpp yourself to take the ip address of the host machine to talk to.

[14 more replies]

closed 2021-05-20 by jonyMarino