microsoft/AirSim #2478: Hundreds of warnings while compiling AirSim for Unity on Linux

rajat2004, 2020-03-25:
OS: Ubuntu 18.04
Unity 2019.f2

Reproduction -
```
cd Unity
./build.sh
```
And watch as hundreds of warnings(about 340) scroll by on your screen. Total output size after redirecting it to a file comes out to be 9 MB

I believe this is happening due to compiler warning flags used to compile RPC being applied to the entire source code. This is quite bothersome, especially when trying to setup Travis for testing where the max output size is 4 MB, and thus requires additional workarounds to reduce output

From my poking around and diving into the CMake files, the problem seems to occur due to -

1. Unity/AirlibWrapper has it's own [CMakeLists.txt](https://github.com/microsoft/AirSim/blob/master/Unity/AirLibWrapper/AirsimWrapper/CMakeLists.txt), where [cmake/rpc-setup.cmake](https://github.com/microsoft/AirSim/blob/master/Unity/AirLibWrapper/AirsimWrapper/cmake/rpc-setup.cmake) is included, which is modified from the [original one used in RPC](https://github.com/rpclib/rpclib/blob/master/CMakeLists.txt)

2. CMake `project()` is used [here](https://github.com/microsoft/AirSim/blob/master/Unity/AirLibWrapper/AirsimWrapper/CMakeLists.txt#L23), which is the first and the only instance of this being used in the Unity CMake setup, and hence defines the `PROJECT_NAME` in all the subsequent ones

`rpc-setup.cmake` sets the [compiler flags for the project](https://github.com/microsoft/AirSim/blob/master/Unity/AirLibWrapper/AirsimWrapper/cmake/rpc-setup.cmake#L165), and this causes them to be applied everywhere. Usage of `PROJECT_NAME` can be seen by searching for it in the file

Such a thing doesn't happen in Unreal because it uses a [wrapper](https://github.com/microsoft/AirSim/blob/master/cmake/rpclib_wrapper/CMakeLists.txt) over RPC's CMakeList itself with a separate project which doesn't affect the other files

This is just my current understanding of trying to figure of why this is happening. I'm not experienced in using CMake, and would love to have others's input in trying to figure out this problem

saihv, 2020-04-30:
Well, using `-Weverything` will do that I guess : https://github.com/microsoft/AirSim/blob/master/Unity/AirLibWrapper/AirsimWrapper/cmake/rpc-setup.cmake#L271

I don't think such heavy diagnostics are needed for us, especially when our code doesn't really adhere to any of those standards. I think we should relax the warning level by removing `-Weverything` and use `-Wall` and maybe `-Wextra`.

@madratman

rajat2004, 2020-05-01:
Removing some of the warning flags might reduce the amount, but this doesn't address the core problem here as far as I understand currently

rajat2004, 2020-05-09:
#2672 reduces the warnings, but I feel that the issue (if I'm correct in my understanding) is still applicable. Therefore keeping the issue open for now