microsoft/AirSim #2480: Is it necessary to install a GPS module when I run HITL?

home26, 2020-03-25:
Hi

My computer was not equipped with a GPS module and there is no GPS module in pixhawk too. I cannot control drone to take off both in QGC and Airsim.(see picture below) So is it necessary to install a GPS module in my computer or pixhawk if I want to run HITL?

Thank you in advance!
[image]

rajat2004, 2020-03-26:
Shouldn't be, GPS messages are generated by AirSim and sent to Pixhawk in HITL. If you have specified sensors in your `settings.json`, check if GPS is present in them

home26, 2020-03-26:
Got it, Thank you so much!

rajat2004, 2020-03-26:
@home26 If the problem has been solved, please do close the issue :)

closed 2020-03-26 by home26