microsoft/AirSim #2484: How to Load an Environment with the Python API?

flolu, 2020-03-27:
How is it possible to load a specific environment vai the Python API? I remember seeing a function like `simLoadLeven()` but I can't find anything like this.

madratman, 2020-03-27:
Hi @flolu, you saw that in the Game of Drones competition repo here https://github.com/microsoft/AirSim-Neurips2019-drone-racing#loading-unreal-engine-environments
We're working towards porting `simLoadLevel` to this repo, it's a bit tricky with the number of environments in the main public repo and some environments misbehaving when tried to be `pak`ed. But we're hoping to get that done soon.

flolu, 2020-03-27:
@madratman Ok, that feature would be really handy. Thanks for the fast response :)

closed 2020-08-28 by jonyMarino