microsoft/AirSim #2496: Compilation error in latest release

songxiaocheng, 2020-03-31:
The Blocks project in AirSim v1.3.0 have compilation error with UE4.18:
```
1><path-ro-airsim>\Unreal\Environments\Blocks\Plugins\AirSim\Source\WorldSimApi.cpp(211): error C2660: 'DrawDebugString': function does not take 8 arguments
1>C:\Program Files\Epic Games\UE_4.18\Engine\Source\Runtime\Engine\Public\DrawDebugHelpers.h(47): note: see declaration of 'DrawDebugString'
1><path-ro-airsim>\Unreal\Environments\Blocks\Plugins\AirSim\Source\WorldSimApi.cpp(230): error C2660: 'DrawDebugString': function does not take 8 arguments
1>C:\Program Files\Epic Games\UE_4.18\Engine\Source\Runtime\Engine\Public\DrawDebugHelpers.h(47): note: see declaration of 'DrawDebugString'
```
It seems that `DrawDebugString` in UE4.18 have only declared 7 arguments but it is fixed in later version (see [here](https://issues.unrealengine.com/issue/UE-71998)).

But the documentation still recommends UE4.18.

rajat2004, 2020-03-31:
Yes, the documentation needs to be updated. I believe from this release AirSim will require UE>=4.22 (or maybe even 4.24). Some more PRs in the line (https://github.com/microsoft/AirSim/pull/2482, https://github.com/microsoft/AirSim/pull/2472) will require the newer UE versions

madratman, 2020-03-31:
Updated the docs https://microsoft.github.io/AirSim/build_windows/ 
Current master works on UE>=4.22, but we do recommend 4.24.

closed 2020-03-31 by madratman