microsoft/AirSim #2499: Linux Ubuntu 19.10 failed make command on installing Unreal Engine 4.23.3

KentaTetsu, 2020-03-31:
Hello everyone,

I tried to install Unreal Engine on Ubuntu and I follow the tutorial from the official Linux Unreal Engine
here is the link : https://docs.unrealengine.com/en-US/Platforms/Linux/BeginnerLinuxDeveloper/SettingUpAnUnrealWorkflow/index.html

I have a error the moment when we need to make the command "make"
Look what I have got some error
`bash "/home/tetsu/UnrealEngine/Engine/Build/BatchFiles/Linux/Build.sh" CrashReportClient Linux Shipping  
chmod: changing permissions of 'bin/mcs': Operation not permitted
chmod: changing permissions of 'bin/xbuild': Operation not permitted
Fixing inconsistent case in filenames.
Setting up Mono
/home/tetsu/UnrealEngine/Engine/Binaries/ThirdParty/Mono/Linux/lib/mono/xbuild/14.0/bin/Microsoft.Common.targets: error : Access to the path "/home/tetsu/UnrealEngine/Engine/Source/Programs/DotNETCommon/DotNETUtilities/obj/Development/DotNETUtilities.csproj.FilesWrittenAbsolute.txt" is denied.  at System.IO.FileStream..ctor (System.String path, System.IO.FileMode mode, System.IO.FileAccess access, System.IO.FileShare share, System.Int32 bufferSize, System.Boolean anonymous, System.IO.FileOptions options) [0x0019e] in <98fac219bd4e453693d76fda7bd96ab0>:0 
  at System.IO.FileStream..ctor (System.String path, System.IO.FileMode mode, System.IO.FileAccess access, System.IO.FileShare share, System.Int32 bufferSize, System.IO.FileOptions options, System.String msgPath, System.Boolean bFromProxy, System.Boolean useLongPath, System.Boolean checkHost) [0x00000] in <98fac219bd4e453693d76fda7bd96ab0>:0 
  at (wrapper remoting-invoke-with-check) System.IO.FileStream..ctor(string,System.IO.FileMode,System.IO.FileAccess,System.IO.FileShare,int,System.IO.FileOptions,string,bool,bool,bool)
  at System.IO.StreamWriter.CreateFile (System.String path, System.Boolean append, System.Boolean checkHost) [0x0001c] in <98fac219bd4e453693d76fda7bd96ab0>:0 
  at System.IO.StreamWriter..ctor (System.String path, System.Boolean append, System.Text.Encoding encoding, System.Int32 bufferSize, System.Boolean checkHost) [0x00055] in <98fac219bd4e453693d76fda7bd96ab0>:0 
  at System.IO.StreamWriter..ctor (System.String path, System.Boolean append, System.Text.Encoding encoding, System.Int32 bufferSize) [0x00000] in <98fac219bd4e453693d76fda7bd96ab0>:0 
  at System.IO.StreamWriter..ctor (System.String path, System.Boolean append) [0x00008] in <98fac219bd4e453693d76fda7bd96ab0>:0 
  at (wrapper remoting-invoke-with-check) System.IO.StreamWriter..ctor(string,bool)
  at Microsoft.Build.Tasks.WriteLinesToFile.Execute () [0x0002e] in <1e06a39ffa2a4ecc9361f5e710a0e7cb>:0 
/home/tetsu/UnrealEngine/Engine/Binaries/ThirdParty/Mono/Linux/lib/mono/xbuild/14.0/bin/Microsoft.Common.targets: error : Access to the path "/home/tetsu/UnrealEngine/Engine/Source/Programs/UnrealBuildTool/obj/Development/UnrealBuildTool.csproj.FilesWrittenAbsolute.txt" is denied.  at System.IO.FileStream..ctor (System.String path, System.IO.FileMode mode, System.IO.FileAccess access, System.IO.FileShare share, System.Int32 bufferSize, System.Boolean anonymous, System.IO.FileOptions options) [0x0019e] in <98fac219bd4e453693d76fda7bd96ab0>:0 
  at System.IO.FileStream..ctor (System.String path, System.IO.FileMode mode, System.IO.FileAccess access, System.IO.FileShare share, System.Int32 bufferSize, System.IO.FileOptions options, System.String msgPath, System.Boolean bFromProxy, System.Boolean useLongPath, System.Boolean checkHost) [0x00000] in <98fac219bd4e453693d76fda7bd96ab0>:0 
  at (wrapper remoting-invoke-with-check) System.IO.FileStream..ctor(string,System.IO.FileMode,System.IO.FileAccess,System.IO.FileShare,int,System.IO.FileOptions,string,bool,bool,bool)
  at System.IO.StreamWriter.CreateFile (System.String path, System.Boolean append, System.Boolean checkHost) [0x0001c] in <98fac219bd4e453693d76fda7bd96ab0>:0 
  at System.IO.StreamWriter..ctor (System.String path, System.Boolean append, System.Text.Encoding encoding, System.Int32 bufferSize, System.Boolean checkHost) [0x00055] in <98fac219bd4e453693d76fda7bd96ab0>:0 
  at System.IO.StreamWriter..ctor (System.String path, System.Boolean append, System.Text.Encoding encoding, System.Int32 bufferSize) [0x00000] in <98fac219bd4e453693d76fda7bd96ab0>:0 
  at System.IO.StreamWriter..ctor (System.String path, System.Boolean append) [0x00008] in <98fac219bd4e453693d76fda7bd96ab0>:0 
  at (wrapper remoting-invoke-with-check) System.IO.StreamWriter..ctor(string,bool)
  at Microsoft.Build.Tasks.WriteLinesToFile.Execute () [0x0002e] in <1e06a39ffa2a4ecc9361f5e710a0e7cb>:0 
Failed to build to build tool (UnrealBuildTool)
make: *** [Makefile:302: CrashReportClient-Linux-Shipping] Error 1
`

madratman, 2020-04-01:
Well, it means you don't have write issues to that path `bin` path. Please google `set read/write permissions for current user for directory in linux`. 
You need something like `$ sudo chown -R $(id -u):$(id -g) PATH_TO_DIR` where `PATH_TO_DIR` is the path to directory of which permissions you want to change. 
In general, this is Unreal Engine installation issue. You perhaps have missed a step while following their instructions, or used `sudo` somewhere when you didn't need to.

KentaTetsu, 2020-04-01:
I have found my error was I have do every instructions with root but we need to do without root. Thank You

closed 2020-04-01 by KentaTetsu