microsoft/AirSim #2505: PX4 SITL + Gimbal + Camera

depauwjimmy, 2020-04-01:
I have been wondering if it is possible to have a mavlink virtual camera/gimbal inside Airsim while running PX4 SITL.

It's working just fine for just the drone but now i need a virtual gimbal and a camera that is mavlink controlled.
As i haven't found any trace of this setup being supported, here i am asking.

Adding the following to my current working PX4 settings.json file just make Airsim crash when i hit play.

`      "Cameras": {
        "GimbalCamera": {
          "CaptureSettings": [
            {
              "ImageType": 0,
              "Width": 800,
              "Height": 600
            }
          ],
          "Gimbal": {
            "Stabilization": 1,
            "Pitch": 0, "Roll": 0, "Yaw": 0
          }
        }
      }`

So in short, is it possible to use airsim with px4 sitl and a virtual mavlink gimbal and camera to work?

rajat2004, 2020-04-02:
Airsim currently doesn't support a Mavlink-controlled Gimbal, similar functionality can be got using [`simSetCameraOrientation`](https://github.com/microsoft/AirSim/blob/master/docs/image_apis.md#camera-apis)
See [gimbal.py](https://github.com/microsoft/AirSim/blob/master/PythonClient/multirotor/gimbal.py) for an example Python script

Please post the entire `settings.json` while filing the issue so that it's easier to figure out what's wrong

closed 2020-08-10 by jonyMarino

rajat2004, 2020-08-10:
Just a note, in the current master, `simSetCameraPose` has replaced `simSetCameraOrientation`, this adds the ability to change the camera position as well.
Mavlink controlled gimbal will be a great feature, but needs some thinking on how to integrate in the codebase