microsoft/AirSim #2508: Fixed Wing UAV Simulations

mustafa-gokce, 2020-04-01:
Hi.
Is there a way to simulate fixed wing UAVs in the AirSim.
I could not be able to find anything about it in docs and the environment.
Where should I start?
Thanks...

rajat2004, 2020-04-04:
AirSim currently doesn't support fixed-wing models. If you would like to add it yourself, then some suggestions on how to go about it can be found in https://github.com/Microsoft/AirSim/issues/1679, https://github.com/Microsoft/AirSim/issues/285, and some other related issues

AOS55, 2020-07-16:
Hey, I am going to try and work out a method to add some of this functionality if anyone else is working on this comment here or DM me on GitHub maybe we could collaborate. I'll try and produce some documentation as I go so we can hopefully merge it back into the main repo if I can get it to work 😕. 

There are a couple of approaches I can see that might work out I will try and feedback on how each pans out in orders of increasing complexity:
1. Using Ardupilot SITL with Airsim's API, there is some recommendations on how to do that [on their documentation](https://ardupilot.org/dev/docs/sitl-with-airsim.html). Ardupilot includes a [fixed wing aircraft](https://ardupilot.org/dev/docs/plane-sitlmavproxy-tutorial.html) simulation tool and [several examples](https://github.com/ArduPilot/ardupilot/tree/master/Tools/autotest/default_params). I'm not sure how to link the APIs together but I hope this will become obvious Ill try to feedback if/when I work it out. 
2. As @RVBldr stated user **Mookie** on unreal has made another plugin called Easy Flight Model that is actually really good in its own right its a point mass but you can directly edit a graph to capture non-linear flight mechanics and all sorts of other fun stuff. I don't know how this would link together but it is in C++, I have sent him a question on the unreal product page asking if anyone has done this before.
3. Adding a new vehicle type in the C++ unreal plugin that can interact with the drone client as suggested by @madratman in #1679. It should be possible to use use a build up of forces approach and assign coefficients to each based on a flight model. This would be refactoring the Multirotor for a new class (FW) tilting the blades to act in the vehicles x axis and then applying a negative z-force (lift) would be useful then there are all the x-terms. I think the implementation of frames (body, wind earth etc) is all already there with Euler too. The firmware and client will almost certainly be a challenge too. This is certainly the most time consuming and I am not that familiar with the source code... yet. 

For reference I am trying to do this for my PhD @ the University of Bristol, where I investigating the use of Reinforcement Learning with Computer Vision to allow fixed wing aircraft to land in unknown environments off-field (away from runways).

rajat2004, 2020-07-17:
@AOS55 The [Easy Flight model](https://www.unrealengine.com/marketplace/en-US/product/easy-flight-model) looks pretty good, and it does state that it works with C++ projects, so should be possible to integrate it with AirSim, though am not sure as to the exact way to go about it.

Integrating AirSim vehicle with ArduPilot would be a straightforward task, once the sensor data such as IMU, etc. are sorted out. AirSim would send the sensor data as a JSON packet to AP, like [done in drone](https://github.com/microsoft/AirSim/blob/master/AirLib/include/vehicles/multirotor/firmwares/arducopter/ArduCopterApi.hpp#L336), and AP would send the servo values back, and the fixed-wing model should apply those to generate new values. Using AP would probably make life easier since you wouldn't need to write a new controller for the plane.

Refactoring out Multirotor to implement the physics of a fixed-wing as described in #1679 seems to be the more difficult approach, and hopefully using Easy Flight will reduce that.

Will be glad to help out!

AOS55, 2020-08-15:
Hi @rajat2004 thanks for your help. Ive been working on this a lot over the past couple of weeks. Ive got something to a point where it should compile with a fixedwingpawn but Im having some issues with the pointers in generated.h under the file FixedWingPawn.h. I have created a [forked repositry](https://github.com/AOS55/AirSim-1) details to follow. Have you encountered E1696 before, cannot open source file <myfile>.generated.h? Im guessing it cannot see the file but I don't know where it would be stored.

Details of the implementation with some open thoughts to follow.

AOS55, 2020-08-15:
# Current Fixed Wing State 

This is an overview of what I have been upto so far in an attempt to add fixedwing functionality to airsim. Most of the work follows a similar style to what has been done with the multirotor implementation. 

## [Unreal/Airsim/](https://github.com/AOS55/AirSim-1/tree/FixedWing/Unreal/Plugins/AirSim/Source/Vehicles/FixedWing)
[FixedWingPawn.h](https://github.com/AOS55/AirSim-1/blob/FixedWing/Unreal/Plugins/AirSim/Source/Vehicles/FixedWing/FixedWingPawn.h) added the URotatingMovementComponents for 3 control surfaces and defines main UCLASS `AFixedWingPawn`.

[FixedWingPawn.cpp](https://github.com/AOS55/AirSim-1/blob/FixedWing/Unreal/Plugins/AirSim/Source/Vehicles/FixedWing/FixedWingPawn.cpp) similar to the multirotor FlyingPawn.cpp implementation, specific elevator, aileron & rudder control surfaces are defined. The use of `std::vectors` library here does not quite make as much sense and they should really be floats I have just indexed all values `.at(0)`.

[FixedWingPawnEvents.h](https://github.com/AOS55/AirSim-1/blob/FixedWing/Unreal/Plugins/AirSim/Source/Vehicles/FixedWing/FixedWingPawnEvents.h) defines the control surfaces as 3 structures all with similar parameters. 

[FixedWingPawnEvents.cpp](https://github.com/AOS55/AirSim-1/blob/FixedWing/Unreal/Plugins/AirSim/Source/Vehicles/FixedWing/FixedWingPawnEvents.cpp) returns the struct values for each control surface with the `get<controlsurface>Signal` method.

[FixedWingPawnSimApi.h](https://github.com/AOS55/AirSim-1/blob/FixedWing/Unreal/Plugins/AirSim/Source/Vehicles/FixedWing/FixedWingPawnSimApi.h) is the header file that relates the unreal engine back to the API defined under airlib.

[FixedWingPawnSimApi.cpp](https://github.com/AOS55/AirSim-1/blob/FixedWing/Unreal/Plugins/AirSim/Source/Vehicles/FixedWing/FixedWingPawnSimApi.cpp) Include methods that link AFixedWingPawn to the Airlib Vehicles file. The vector method that I inherited from the multirotor is an apparent issue here it should work as is it is just crude. 

[SimModeWorldFixedWing.h](https://github.com/AOS55/AirSim-1/blob/FixedWing/Unreal/Plugins/AirSim/Source/Vehicles/FixedWing/SimModeWorldFixedWing.h) & [SimModeWorldFixedWing.cpp](https://github.com/AOS55/AirSim-1/blob/FixedWing/Unreal/Plugins/AirSim/Source/Vehicles/FixedWing/SimModeWorldFixedWing.cpp) have almost identical implementations as they are in the multirotor source file. 

## [AirLib/include/vehicles/fixedwing/](https://github.com/AOS55/AirSim-1/tree/FixedWing/AirLib/include/vehicles/fixedwing)
[FixedWingParams.hpp](https://github.com/AOS55/AirSim-1/blob/FixedWing/AirLib/include/vehicles/fixedwing/FixedWingParams.hpp) is the param file for the aircraft itself. The flight physics itself is held in the DynamicForces file, but the key structure `Params` within `FixedWingParams` is defined with mass, inertia and aeroDerivatives. The current implementation uses linear coefficients with a point mass and inertia array, similar to the multirotor. Currently empty there are several frames for aircraft defined as follows:
- `setupFrameGenericAircraft` should be a basic conventional fixed wing UAV;
- `setupFrameCherokee` will be a PA-28 161 warrior;
- `setupFrameDA62` will be the Diamond DA-62;
- `setupFrameGlider` will be modelled based on the Univeristy of Bristol Glider.

[DynamicParams.hpp](https://github.com/AOS55/AirSim-1/blob/FixedWing/AirLib/include/vehicles/fixedwing/DynamicParams.hpp) includes the `LinearAeroDerivatives` structure which implements linear aeroderivatives ignoring several small cross-terms. The `Dimensions` structure is the reference area for each derivative to get the forces.

[DynamicForces.hpp](https://github.com/AOS55/AirSim-1/blob/FixedWing/AirLib/include/vehicles/fixedwing/DynamicForces.hpp) this ties together in particular the `setOutput` method calculates forces based on airspeed, I still need to cross-check some of these paramaterised EoMs, formal documentation can be written when all is working. 

[FixedWingPhysicsBody.hpp](https://github.com/AOS55/AirSim-1/blob/FixedWing/AirLib/include/vehicles/fixedwing/FixedWingPhysicsBody.hpp) ties the forces created in dynamic forces with the params file and calls the relevant methods. This is referenced in the FixedWing Unreal/airsim calls.

[FixedWingParamsFactory](https://github.com/AOS55/AirSim-1/blob/FixedWing/AirLib/include/vehicles/fixedwing/FixedWingParamsFactory.hpp) constructs the relevant APIs from the FixedWingPhysicsBody. Currently have calls for `kVehicleTypePX4` & `kVehicleTypeArduPlane` but no specific information JSON transfer as described above.

Similarly in the [api](https://github.com/AOS55/AirSim-1/tree/FixedWing/AirLib/include/vehicles/fixedwing/api) folder and [firmwares](https://github.com/AOS55/AirSim-1/tree/FixedWing/AirLib/include/vehicles/fixedwing/firmwares) folder Mavlink and ardupilot files have been made but they are effectively just placeholders with no functionality at the moment. The [api Airlib src code](https://github.com/AOS55/AirSim-1/tree/FixedWing/AirLib/src/vehicles/fixedwing/api) will also need some work as this is effectively just a copy and paste from multirotor.  

## To do
The following are my key short term goals to have fixedwing working:
- [ ] Resolve unreal compilation errors described above so I can at least run the program
- [ ] Sort out the communication link between AirSim and ardupilot for fixedwing control
[... cut ...]
- [ ] Implement Undercarriage for fixedWingLandings
- [ ] DeepReinforcementLearning based landings <-- main PhD objective
- [ ] Validate model and control on real UAV and FW aircraft

If you thoughts, recommendations problems etc please let me know, Im not sure at what point a merge back to master would be useful given the size of the commit this may be a while. Im keen to work with anyone to get this remerged to expand AirSim's features I know in my office at least several other researchers will benefit from this work.

rajat2004, 2020-08-15:
@AOS55 Wow, that's quite a lot of things done! Just a few points from the above description & a brief look I had at the files changed. Will post more queries after a good look at the code and maybe test it out.

You're using UE 4.25 right, from the fact that you've created a new Blocks env. Are the assets modified or copies of the original ones?
Related to this, the assets in Plugins don't seem to be modified, so right now you're trying to use the multirotor model as the plane? Also, didn't see any mention of the Easy Flight plugin, so not sure if you're thinking of using that.

For the compilation errors, maybe try running `clean_rebuild.bat` from the root repo, and `clean.bat` from the env repo, that should force it to regenerate any files required.

If you prefer, you could create a new issue, that'll give this more visibility.

AOS55, 2020-08-18:
@rajat2004 Thanks a lot for your help. 

Yes thats right, should be the same as original I think UE4 prompted me to convert in place but the unmodified airsim seemed to run without issue in 4.25, I can try in 4.24 but I don't think that should be a problem.

Thats a good point I wll take a look at the assets I thought I would need to make a simple plane 3D model, we have several CAD files of FW UAVs in our lab I can use and convert for UE4. 

I looked at Easy Flight Plugin in detail it was useful to give me a sense of how you could go about implementing a fixed wing sim, it relies a lot more on the use of blueprints with a simpler C++ function backend than airsim uses with airlib etc. The code is also copyrighted to the author in the body so I didn't want to use anything that could infringe. Ultimately I don't think it will be that useful here directly. 

I tried running `clean_rebuild.bat` from root and then `clean.bat` followed by `update_from_git.bat` then rebuilt in editor with blocks set as startup project, I think the error is in obtaining the generated.h files during the build I don't quite understand where these come from in the UE4 C++ build process. I have attached a screenshot of my UE4 terminal that should hopefully be useful. 

Ok great Ill take a look at that today and work through how the signals communicate with one another. 

Thats a good idea Ill aim to make something like that towards the end of next week/beginning of next week.

rajat2004, 2020-08-18:
AirSim master supports UE 4.25, so should run without problems. Also, currently Azure Pipelines uses 4.25.1 for CI testing. Just that  I haven't used 4.25 much, actually built it on my system just a few days ago (Maybe I've forgotten how long it takes, but seemed much more longer this time)

I haven't yet had a look at Easy Flight but your points seem valid, and license could also be problem, so better to not use it.

No attachment :)
Maybe the problem is happening due to the new `Blocks_4.25` env, some file such as `GenerateProjectFiles`, or `Build.cs` have Blocks hardcoded somewhere which is causing the problem. Once try with the normal Blocks env.

rajat2004, 2020-08-18:
Now that's a big display! 
Not seeing any problem at first glance atleast, you're using the Blocks env from the photo itself rather than the Blocks_4.25 env right? Maybe the problem's happening due to a compilation error, and it's not able to generate the files.
Did you add the new files in the VS project file? I think that's required for any new files created like in the AirLib folder

AOS55, 2020-08-18:
Yes thats correct I left that folder there but it should be redundant all is from the folder Blocks as default. It could well be a compilation error somewhere in my code.

I didn't Im going to add those now and rebuild to see if that helps.

AOS55, 2020-08-18:
Ok, been playing around with adding and removing various files in the [AirLib.vcxproj](https://github.com/AOS55/AirSim-1/blob/FixedWing/AirLib/AirLib.vcxproj) file. I get build errors when I leave the compile commands as is this seems to be principally related to (FixedWingApiBase.hpp)[https://github.com/AOS55/AirSim-1/blob/FixedWing/AirLib/include/vehicles/fixedwing/api/FixedWingApiBase.hpp] and the (associated C++ file)[https://github.com/AOS55/AirSim-1/blob/FixedWing/AirLib/src/vehicles/fixedwing/api/FixedWingApiBase.cpp] which is where the errors initially appear. 

Ill take a look at these and report if I can spot anything, the higher order function that has been copied over here probably is not required in several spots as this is copied from the MultiRotor, for example move by yaw does not make sense. Anything other than symmetric flight requries an offseting roll force, effectively out of balance flight. Im guessing Px4 has a simple yaw damper to minimise sideslip and ensures symettric flight solving this for instance, this can act on ayy (side acceleration).

AOS55, 2020-08-20:
I have moved this issue to #2955 to increase visibility