microsoft/AirSim #2509: Error C1083. Running the LandscapeMountains.snl

Jload, 2020-04-02:
i clicked on DebugGame Editor and Win65 and ran the code using F5.
This is the output.

1>------ Build started: Project: LandscapeMountains, Configuration: DebugGame_Editor x64 ------
1>Building LandscapeMountainsEditor...
1>Using Visual Studio 2017 14.16.27038 toolchain (C:\Program Files (x86)\Microsoft Visual Studio\2017\Community\VC\Tools\MSVC\14.16.27023) and Windows 10.0.18362.0 SDK (C:\Program Files (x86)\Windows Kits\10).
1>Building 6 actions with 8 processes...
1>  [1/6] Module.AirSim.cpp
1>c:\users\rs\documents\unreal projects\landscapemountains\plugins\airsim\source\AirBlueprintLib.h(20): fatal error C1083: Cannot open include file: 'common/AirSimSettings.hpp': No such file or directory
1>  [2/6] Module.AirSim.gen.cpp
1>C:\Users\RS\Documents\Unreal Projects\LandscapeMountains\Plugins\AirSim\Source/AirBlueprintLib.h(20): fatal error C1083: Cannot open include file: 'common/AirSimSettings.hpp': No such file or directory
1>C:\Program Files (x86)\Microsoft Visual Studio\2017\Community\Common7\IDE\VC\VCTargets\Microsoft.MakeFile.Targets(44,5): error MSB3075: The command ""C:\Program Files\Epic Games\UE_4.24\Engine\Build\BatchFiles\Build.bat" LandscapeMountainsEditor Win64 DebugGame -Project="C:\Users\RS\Documents\Unreal Projects\LandscapeMountains\LandscapeMountains.uproject" -WaitMutex -FromMsBuild" exited with code 5. Please verify that you have sufficient rights to run this command.
1>Done building project "LandscapeMountains.vcxproj" -- FAILED.
========== Build: 0 succeeded, 1 failed, 0 up-to-date, 0 skipped ==========

Please Help!!!!!!!!!!!!!

rajat2004, 2020-04-02:
Hi,
The missing files seem to indicate that the `Unreal\Plugins` folder wasn't copied properly to the LandscapeMountains directory. First try running the default Blocks environment and make sure it's working. Then copy the `Plugins` folder to the custom env directory

Jload, 2020-04-02:
so tried running the Blocks and this was the output.

1>------ Build started: Project: Blocks, Configuration: DebugGame_Editor x64 ------
1>Creating makefile for BlocksEditor (no existing makefile)
1>C:\AirSim\Unreal\Environments\Blocks\Plugins\AirSim\Source\AirSim.Build.cs : warning : Referenced directory 'C:\AirSim\Unreal\Environments\Blocks\Plugins\AirSim\Source\AirLib\deps\MavLinkCom\include' does not exist.
1>UnrealBuildTool : warning : Library 'Shell32.lib' was not resolvable to a file when used in Module 'AirSim', assuming it is a filename and will search library paths for it. This is slow and dependency checking will not work for it. Please update reference to be fully qualified alternatively use PublicSystemLibraryPaths if you do intended to use this slow path to suppress this warning.
1>UnrealBuildTool : warning : Library 'dinput8.lib' was not resolvable to a file when used in Module 'AirSim', assuming it is a filename and will search library paths for it. This is slow and dependency checking will not work for it. Please update reference to be fully qualified alternatively use PublicSystemLibraryPaths if you do intended to use this slow path to suppress this warning.
1>UnrealBuildTool : warning : Library 'dxguid.lib' was not resolvable to a file when used in Module 'AirSim', assuming it is a filename and will search library paths for it. This is slow and dependency checking will not work for it. Please update reference to be fully qualified alternatively use PublicSystemLibraryPaths if you do intended to use this slow path to suppress this warning.
1>UnrealBuildTool : warning : Library 'C:\AirSim\Unreal\Environments\Blocks\Plugins\AirSim\Source\AirLib\deps\MavLinkCom\lib\x64\Release\MavLinkCom.lib' was not resolvable to a file when used in Module 'AirSim', assuming it is a filename and will search library paths for it. This is slow and dependency checking will not work for it. Please update reference to be fully qualified alternatively use PublicSystemLibraryPaths if you do intended to use this slow path to suppress this warning.
1>Parsing headers for BlocksEditor
1>  Running UnrealHeaderTool "C:\AirSim\Unreal\Environments\Blocks\Blocks.uproject" "C:\AirSim\Unreal\Environments\Blocks\Intermediate\Build\Win64\BlocksEditor\DebugGame\BlocksEditor.uhtmanifest" -LogCmds="loginit warning, logexit warning, logdatabase error" -Unattended -WarningsAsErrors -installed
1>Reflection code generated for BlocksEditor in 11.788345 seconds
1>Building BlocksEditor...
1>Using Visual Studio 2017 14.16.27038 toolchain (C:\Program Files (x86)\Microsoft Visual Studio\2017\Community\VC\Tools\MSVC\14.16.27023) and Windows 10.0.18362.0 SDK (C:\Program Files (x86)\Windows Kits\10).
1>[Upgrade]
1>[Upgrade] Using backward-compatible build settings. The latest version of UE4 sets the following values by default, which may require code changes:
1>[Upgrade]     bLegacyPublicIncludePaths = false                 => Omits subfolders from public include paths to reduce compiler command line length. (Previously: true).
1>[Upgrade]     ShadowVariableWarningLevel = WarningLevel.Error   => Treats shadowed variable warnings as errors. (Previously: WarningLevel.Warning).
1>[Upgrade]     PCHUsage = PCHUsageMode.UseExplicitOrSharedPCHs   => Set in build.cs files to enables IWYU-style PCH model. See https://docs.unrealengine.com/en-US/Programming/BuildTools/UnrealBuildTool/IWYU/index.html. (Previously: PCHUsageMode.UseSharedPCHs).
1>[Upgrade] Suppress this message by setting 'DefaultBuildSettings = BuildSettingsVersion.V2;' in BlocksEditor.Target.cs, and explicitly overriding settings that differ from the new defaults.
1>[Upgrade]
1>Building 11 actions with 8 processes...
1>  [1/11] Default.rc2
1>  [2/11] Default.rc2
1>  [3/11] SharedPCH.Engine.NonOptimized.cpp
1>  [4/11] Blocks.cpp
1>  [5/11] UE4Editor-Blocks-Win64-DebugGame.lib
1>     Creating library C:\AirSim\Unreal\Environments\Blocks\Intermediate\Build\Win64\UE4Editor\DebugGame\Blocks\UE4Editor-Blocks-Win64-DebugGame.lib and object C:\AirSim\Unreal\Environments\Blocks\Intermediate\Build\Win64\UE4Editor\DebugGame\Blocks\UE4Editor-Blocks-Win64-DebugGame.exp
1>  [6/11] Module.AirSim.gen.cpp
1>  [7/11] Module.AirSim.cpp
1>C:\AirSim\Unreal\Environments\Blocks\Plugins\AirSim\Source\AirLib\include\vehicles/multirotor/firmwares/mavlink/MavLinkMultirotorApi.hpp(7): fatal error C1083: Cannot open include file: 'MavLinkVehicle.hpp': No such file or directory
1>C:\Program Files (x86)\Microsoft Visual Studio\2017\Community\Common7\IDE\VC\VCTargets\Microsoft.MakeFile.Targets(44,5): error MSB3075: The command ""C:\Program Files\Epic Games\UE_4.24\Engine\Build\BatchFiles\Build.bat" BlocksEditor Win64 DebugGame -Project="C:\AirSim\Unreal\Environments\Blocks\Blocks.uproject" -WaitMutex -FromMsBuild" exited with code 5. Please verify that you have sufficient rights to run this command.
1>Done building project "Blocks.vcxproj" -- FAILED.
========== Build: 0 succeeded, 1 failed, 0 up-t

Do you think I did something wrong? rajat2004

rajat2004, 2020-04-02:
Please have a look at [build_windows.md](https://github.com/microsoft/AirSim/blob/master/docs/build_windows.md)
AirSim has now been updated to work with Visual Studio 2019, so please try with that and see if it works

As for the workflow to follow to build AirSim, ensure that you've run all the following steps-
`build.cmd` from the root of Airsim directory
Then go to the Blocks directory and run `update_from_git.bat`, that should copy all the files required
Then start the `Blocks.sln` file in Visual Studio and build

Check if there were any errors in the previous steps or if anything was missed.

madratman, 2020-04-03:
Clean up your current project by running `git clean -fdx` in root directory of airsim and then please follow each step in readme carefully and re-open if it still doesn't work
The build is working

closed 2020-04-03 by madratman

alokkrsahu, 2021-01-13:
Hi All,
I was facing the same problem. 

Please try the same setup instructions using some other environment in unreal engine. Landscape mountains in not supported with airsim anymore. Landscape mountains requires 4.24 while airsim requires 4.25 unreal engine. 
Please dont waste your time setting up with Landscape mountains. The setup instructions works absolutely fine for other supported environment. 

Regards,
Alok

xxEoD2242, 2021-01-13:
> Hi All,
> I was facing the same problem.
>
> Please try the same setup instructions using some other environment in unreal engine. Landscape mountains in not supported with airsim anymore. Landscape [...]

@alokkrsahu I have successfully setup a Landscape Mountains environment with UE 4.25. The important part is which version you build the environment with. So if you want UE 4.25, you must generate the project with 4.25. I have downloaded and installed AirSim running 4.25 as recently as last week with latest pull from AirSim. I have successfully built on Windows 10 and Ubuntu 20.04 the same project.

Let me know if you need any help setting these up. Other environments have also successfully built.

alokkrsahu, 2021-01-13:
Man

> > Hi All,
> > I was facing the same problem.
> > Please try the same setup instructions using some other environment in unreal engine. Landscape mountains in not supported with airsim anymore. Landscape mountains requires 4.24 while airsim requires 4.25 unreal engine.
> > Please dont waste your time setting up with Landscape mountains. The setup instructions works absolutely fine for other supported environment.
> > Regards,
> > Alok
> 
> @alokkrsahu I have successfully setup a Landscape Mountains environment with UE 4.25. The important part is which version you build the environment with. So if you want UE 4.25, you must generate the project with 4.25. I have downloaded and installed AirSim running 4.25 as recently as last week with latest pull from AirSim. I have successfully built on Windows 10 and Ubuntu 20.04 the same project.
> 
> Reading through these posts, you must use Visual Studio 2019 if you are on Windows. I have compiled from source Unreal Engine on Ubuntu.
> 
> Let me know if you need any help setting these up. Other environments have also successfully built.

Many thanks for your reply. 

I was not aware that previous version of airsim if available. Please let me know if we can install both the airsim version on the same PC (windows)?

xxEoD2242, 2021-01-13:
@alokkrsahu You don't need a previous version of AirSim. You can use the current one. You just need to generate an Unreal Project using the Unreal Engine 4.25 version that you download from the Library of the Epic Games Launcher. From there, generate a new project with LandscapeMountains using that version and install AirSim the way that is presented in the documentation.

tristanthomas3, 2022-03-04:
@xxEoD2242 Hi, so I am having a similar issue. Did you have to do anything aside from what is described here https://microsoft.github.io/AirSim/unreal_custenv/ to generate an Unreal Project with LandscapeMountains that is version 4.25? Because I am still getting an error when I build in Visual Studio.

alokkrsahu, 2022-03-04:
I was able to resolve my error following the steps one by one carefully. That's it.
Just check for version conflict if any between the landscape.snl and IDE.

tristanthomas3, 2022-03-04:
@alokkrsahu So did you build the LandscapeMountains environment using 4.25? If so, how? Because I only have the option of building with 4.24.

Not sure what you mean about version conflict between landscape.sln and IDE.

alokkrsahu, 2022-03-04:
I suggest you to try with something other than landscape mountain. You should be able to run it.

There is something fuzzy about landscape.
You will figure it out.

But trust me I just followed the steps very careful as given in documentation and I was able to run it.