microsoft/AirSim #2513: Unity part of the Build FAILED. error C2259

lingfei1998, 2020-04-02:
# when  running the 'build.cmd' in Unity folder it fails

## I find a previous question exactly like mine
* https://github.com/microsoft/AirSim/issues/1964

## here is the error message
“C:\Users\ling\source\repos\AirSim\Unity\AirLibWrapper\AirsimWrapper.sln”(Clean;Build 目标) (1) ->
“C:\Users\ling\source\repos\AirSim\Unity\AirLibWrapper\AirsimWrapper\AirsimWrapper.vcxproj.metaproj”(默认目标) (2:2) ->
“C:\Users\ling\source\repos\AirSim\Unity\AirLibWrapper\AirsimWrapper\AirsimWrapper.vcxproj”(默认目标) (5:2) ->
(ClCompile 目标) ->
  C:\Users\ling\source\repos\AirSim\Unity\AirLibWrapper\AirsimWrapper\Source\SimMode\SimModeBase.cpp(18): error C2259: “WorldSimApi”: cannot instantiate abstract class[C:\Users\ling\source\repos\AirSim\Unity\AirL
ibWrapper\AirsimWrapper\AirsimWrapper.vcxproj]

#### Looks the same as the 1964 question
I made sure I downloaded the latest version，but I do not know how to solve it

rajat2004, 2020-04-02:
Hi,
Please try out the fixes done in https://github.com/microsoft/AirSim/pull/2514, I have tested it on Linux only, would be great to know whether it works on Windows or not

lingfei1998, 2020-04-03:
yes, it works on Windows now. Thank you.

closed 2020-04-03 by lingfei1998