microsoft/AirSim #2517: g++: error: unrecognized command line option ‘-stdlib=libc++

maerad7, 2020-04-03:
# Read This First

## If you are reporting a bug
* Make sure to write **all reproduction steps**
* Include full error message in text form
* Search issues for error message before filing issue
* Attach screenshot if applicable
* Include code to run if applicable

## If you have question
* Add clear and concise title
* Add OS, AirSim version, Python version, Unreal version if applicable
* Include context on what you are trying to achieve
* Include details of what you already did to find answers

**What's better than filing issue? Filing a pull request :).**

------------------------------------ (Remove above before filing the issue) ------------------------------------

log

./build.sh --gcc
+ MIN_GCC_VERSION=6.0.0
+ gccBuild=false
+ [[ 1 -gt 0 ]]
+ key=--gcc
+ case $key in
+ gccBuild=true
+ shift
+ [[ 0 -gt 0 ]]
+ '[' '!' -d ./external/rpclib/rpclib-2.2.1 ']'
+ '[' -d ./cmake_build ']'
++ which cmake
+ CMAKE=/usr/bin/cmake
+ true
+ build_dir=build_gcc_debug
+ which gcc
/usr/bin/gcc
++ gcc -dumpfullversion
+ gcc_ver=7.5.0
+ version_less_than_equal_to 7.5.0 6.0.0
++ printf '%s\n' 7.5.0 6.0.0
++ sort -V
++ head -n 1
+ test 6.0.0 = 7.5.0
+ export CC=gcc
+ CC=gcc
+ export CXX=g++
+ CXX=g++
+ [[ -d ./AirLib/deps/eigen3/Eigen ]]
+ echo 'putting build in build_gcc_debug folder, to clean, just delete the directory...'
putting build in build_gcc_debug folder, to clean, just delete the directory...
+ [[ -f ./cmake/CMakeCache.txt ]]
+ [[ -d ./cmake/CMakeFiles ]]
+ [[ ! -d build_gcc_debug ]]
+ pushd build_gcc_debug
++ nproc
+ make -j12
[  1%] Building CXX object MavLinkCom/CMakeFiles/MavLinkCom.dir/home/newdrone18/droneutil/AirSim/MavLinkCom/common_utils/ThreadUtils.cpp.o
[  2%] Building CXX object MavLinkCom/CMakeFiles/MavLinkCom.dir/home/newdrone18/droneutil/AirSim/MavLinkCom/common_utils/FileSystem.cpp.o
[  4%] Building CXX object MavLinkCom/CMakeFiles/MavLinkCom.dir/home/newdrone18/droneutil/AirSim/MavLinkCom/src/MavLinkConnection.cpp.o
[  5%] Building CXX object MavLinkCom/CMakeFiles/MavLinkCom.dir/home/newdrone18/droneutil/AirSim/MavLinkCom/src/AdHocConnection.cpp.o
[  7%] Building CXX object MavLinkCom/CMakeFiles/MavLinkCom.dir/home/newdrone18/droneutil/AirSim/MavLinkCom/src/MavLinkMessageBase.cpp.o
[  8%] Building CXX object MavLinkCom/CMakeFiles/MavLinkCom.dir/home/newdrone18/droneutil/AirSim/MavLinkCom/src/MavLinkLog.cpp.o
[ 10%] Building CXX object MavLinkCom/CMakeFiles/MavLinkCom.dir/home/newdrone18/droneutil/AirSim/MavLinkCom/src/MavLinkTcpServer.cpp.o
[ 11%] Building CXX object MavLinkCom/CMakeFiles/MavLinkCom.dir/home/newdrone18/droneutil/AirSim/MavLinkCom/src/MavLinkMessages.cpp.o
[ 13%] Building CXX object MavLinkCom/CMakeFiles/MavLinkCom.dir/home/newdrone18/droneutil/AirSim/MavLinkCom/src/MavLinkFtpClient.cpp.o
[ 14%] Building CXX object MavLinkCom/CMakeFiles/MavLinkCom.dir/home/newdrone18/droneutil/AirSim/MavLinkCom/src/MavLinkNode.cpp.o
g++: error: unrecognized command line option ‘-stdlib=libc++’
[ 16%] Building CXX object MavLinkCom/CMakeFiles/MavLinkCom.dir/home/newdrone18/droneutil/AirSim/MavLinkCom/src/MavLinkVehicle.cpp.o
[... 69 lines cut ...]
make[2]: *** [rpclib_wrapper/rpclib/CMakeFiles/rpc.dir/lib/rpc/detail/server_session.cc.o] Error 1
g++: error: unrecognized command line option ‘-stdlib=libc++’
rpclib_wrapper/rpclib/CMakeFiles/rpc.dir/build.make:278: recipe for target 'rpclib_wrapper/rpclib/CMakeFiles/rpc.dir/lib/rpc/detail/client_error.cc.o' failed
make[2]: *** [rpclib_wrapper/rpclib/CMakeFiles/rpc.dir/lib/rpc/detail/client_error.cc.o] Error 1
g++: error: unrecognized command line option ‘-stdlib=libc++’
rpclib_wrapper/rpclib/CMakeFiles/rpc.dir/build.make:254: recipe for target 'rpclib_wrapper/rpclib/CMakeFiles/rpc.dir/lib/rpc/detail/response.cc.o' failed
make[2]: *** [rpclib_wrapper/rpclib/CMakeFiles/rpc.dir/lib/rpc/detail/response.cc.o] Error 1
g++: error: unrecognized command line option ‘-stdlib=libc++’
g++: error: unrecognized command line option ‘-stdlib=libc++’
g++: error: unrecognized command line option ‘-stdlib=libc++’
rpclib_wrapper/rpclib/CMakeFiles/rpc.dir/build.make:326: recipe for target 'rpclib_wrapper/rpclib/CMakeFiles/rpc.dir/dependencies/src/format.cc.o' failed
make[2]: *** [rpclib_wrapper/rpclib/CMakeFiles/rpc.dir/dependencies/src/format.cc.o] Error 1
rpclib_wrapper/rpclib/CMakeFiles/rpc.dir/build.make:206: recipe for target 'rpclib_wrapper/rpclib/CMakeFiles/rpc.dir/lib/rpc/rpc_error.cc.o' failed
make[2]: *** [rpclib_wrapper/rpclib/CMakeFiles/rpc.dir/lib/rpc/rpc_error.cc.o] Error 1
rpclib_wrapper/rpclib/CMakeFiles/rpc.dir/build.make:302: recipe for target 'rpclib_wrapper/rpclib/CMakeFiles/rpc.dir/lib/rpc/nonstd/optional.cc.o' failed
make[2]: *** [rpclib_wrapper/rpclib/CMakeFiles/rpc.dir/lib/rpc/nonstd/optional.cc.o] Error 1
CMakeFiles/Makefile2:103: recipe for target 'rpclib_wrapper/rpclib/CMakeFiles/rpc.dir/all' failed
make[1]: *** [rpclib_wrapper/rpclib/CMakeFiles/rpc.dir/all] Error 2
Makefile:129: recipe for target 'all' failed
make: *** [all] Error 2

When I run build.sh -gcc I get an error. Please tell me how to fix it. Airsim version = master
ubutnu = 18.04
g ++ gcc = 7.5
clang 9.0

rajat2004, 2020-04-03:
Hi, please have a look at https://github.com/microsoft/AirSim/pull/2519 and check if it fixes the problem

maerad7, 2020-04-03:
thank you for telling me.

closed 2020-04-03 by maerad7

fangice, 2020-06-15:
Hi,I have the same problem as you. How did you solve it?

rajat2004, 2020-06-15:
@fangice GCC build has actually been removed from master now, so not sure as to how you are getting this error

fangice, 2020-06-16:
Thanks, when I run build.sh -gcc I get this error.

> @fangice GCC build has actually been removed from master now, so not sure as to how you are getting this error

rajat2004, 2020-06-16:
Yes, so as mentioned, option for GCC build has been removed from master. I'm not sure about the use-case for GCC build, but you should either update to the latest master (which doesn't have GCC option and use clang), or possibly could checkout a commit earlier than say March before the update to clang-8, etc happened and GCC build was _probably_ working if absolutely required