microsoft/AirSim #2523: Build.cmd Error

bobstone94, 2020-04-04:
Hi, i tried to build.cmd from my AirSim clone. i'm using UE4.24 and VS2017 but i keep getting this error

Found cmake version: 3.10.2
Starting cmake to build rpclib...
CMake Error: Could not create named generator Visual Studio 16 2019

rajat2004, 2020-04-04:
AirSim has been updated to work with Visual Studio 2019, please have a look at https://github.com/microsoft/AirSim/blob/master/docs/build_windows.md and follow all the steps mentioned

jafermk, 2020-04-05:
After installing the latest cmake version (v3.17...), and following the link posted I was able to successfuly build (or at least it seemed to have successfuly built), the I followed the instructions to build the UE4.24 version of Blocks and the landscape example shown in the documents, but the build keeps failing with the following errors: 

[image]

The landscape project build successfully once I delete the Plugins folder from the project file, but the blocks project never successfully builds (What UE4 version is it originally built in?)

Please let me know of any fixes I can try, Thanks

rajat2004, 2020-04-05:
Hi,
AirSim latest master is running on Windows for me, on the Blocks env. Are you getting the same error for the Blocks env as well? Will look into it, and see if any fixes are required. 
BTW, if you delete the Plugins folder, then you're removing Airsim from the project, since it's a plugin :) 
Supported engine versions for LandscapeMountains can be seen on the marketplace, as in mentioned to be compatible with 4.4-4.24

jafermk, 2020-04-05:
Yeah, I'm getting the same error on the blocks environment.
For the landscape environment a fix I had tried was changing the compiler to VS2019, but I am still getting the same error, wasn't able to do that for the blocks demo because it won't open at all.
And yeah that's why I deleted the Plugins folder to make sure that was the problem. I was actually asking about the version the AirSim was built for, I get the message below when trying to open the uproject file:

[image]

This is before getting this message:

[image]

rajat2004, 2020-04-05:
AirSim isn't specifically built for any UE version if that's what you're asking, it's all source code, which has to be compiled. The binaries are built by UE 4.24, and the recommended UE for development is also 4.24
The pop-up can come since in the .uproject, 4.24 is mentioned, but it can be replaced by some specific Engine version, just click Yes when that comes
What I would suggest is to first make sure that things are working correctly in Blocks rather than going for a different env, and once that's okay, then try with LandscapeMountains

I'm not sure why you're getting them as errors, since it's coming as warnings for me, you can try https://github.com/microsoft/AirSim/pull/2526 if it fixes the problems for you

changh95, 2020-04-06:
> AirSim has been updated to work with Visual Studio 2019, please have a look at https://github.com/microsoft/AirSim/blob/master/docs/build_windows.md and follow all the steps mentioned

This is weird. It seems that the build_windows.md file is updated to the latest VS2019 information, but on the [online documentation](https://microsoft.github.io/AirSim/build_windows/) this is not updated. Without knowing this, I followed the online documentation and installed VS2017 only to find out that build.cmd now supports VS2019 thus I should switch.

I'm tagging @madratman , since this issue may cause a lot of confusion to anyone who's new to AirSim.

rajat2004, 2020-04-06:
Currently, docs update requires a manual update after the changes have been committed to the master branch. The HTML files have to be generated locally using `mkdocs`, and then pushed to the `gh-pages` branch to be visible on Github pages

There are ways to automate this, one way could be to use [Travis](https://docs.travis-ci.com/user/deployment/pages), or a [Github action](https://github.com/marketplace/actions/deploy-mkdocs), will need to be explored and discussed to see what might work

madratman, 2020-04-06:
Updated https://microsoft.github.io/AirSim/build_windows/. Thanks for pointing it out, @changh95.

madratman, 2020-04-07:
@rajat2004 w.r.t automating mkdocs, see https://github.com/microsoft/AirSim/pull/2540
mostly there, minus the relative link issue in readme

bobstone94, 2020-04-07:
> Yeah, I'm getting the same error on the blocks environment.
> For the landscape environment a fix I had tried was changing the compiler to VS2019, but I am still getting the same error, wasn't able to do that for the blocks demo because it won't open at all.
> And yeah that's why I deleted the Plugins folder to make sure that AirSim was the problem. I was actually asking about the version the AirSim was built for, I get the message below when trying to open the uproject file:
> 
> [image]
> 
> This is before getting this message:
> 
> [image]

I've worked on Blocks, it works fine. i'm able to run AirSim on Blocks but i'm having the same issues in LandScapeMountains. Anyone has a solution for this ?

rajat2004, 2020-04-07:
Will try later today with Landscape Mountains, but the problem could be https://github.com/microsoft/AirSim/pull/2526
Don't know why it's coming as a compiler error though, since they are warnings in Blocks

changh95, 2020-04-07:
> I've worked on Blocks, it works fine. i'm able to run AirSim on Blocks but i'm having the same issues in LandScapeMountains. Anyone has a solution for this ?

Getting the same error here. Worked on Blocks, but not on Landscape Mountains.

jafermk, 2020-04-07:
> Getting the same error here. Worked on Blocks, but not on Landscape Mountains.

My fix was to copy the Plugins folder directly from Blocks project folder into the LandscapeMountains project folder. It works, if it fails to build after a few times of working, then the fix is to delete the current Plugins folder and re-copy the Plugins folder from Blocks.

bobstone94, 2020-04-08:
> > > > Yeah, I'm getting the same error on the blocks environment.
> > > > For the landscape environment a fix I had tried was changing the compiler to VS2019, but I am still getting the same error, wasn't able to do that for the blocks demo because it won't open at all.
> > > > And yeah that's why I deleted the Plugins folder to make sure that AirSim was the problem. I was actually asking about the version the AirSim was built for, I get the message below when trying to open the uproject file:
> > > > [image]
> > > > This is before getting this message:
> > > > [image]
> > > 
> > > 
> > > I've worked on Blocks, it works fine. i'm able to run AirSim on Blocks but i'm having the same issues in LandScapeMountains. Anyone has a solution for this ?
> > 
> > 
> > Getting the same error here. Worked on Blocks, but not on Landscape Mountains.
> 
> My fix was to copy the Plugins folder directly from Blocks project folder into the LandscapeMountains project folder. It works, if it fails to build after a few times of working, then the fix is to delete the current Plugins folder and re-copy the Plugins folder from Blocks.
> 
> Edit: And make sure to change the compiler in UE4 to VS2019. The steps are outlined at the bottom of this page: https://github.com/microsoft/AirSim/blob/master/docs/build_windows.md

Hi,

i tried using the Plugins from Blocks, and ensured that my compliers is using VS2019( i only have VS2019 on my computer as well). But i'm getting the same errors.

[image]

saihv, 2020-04-08:
@bobstone94 Interesting that we are seeing different behavior on the same
project. I'll look into this asap. Are you using UE 4.24.3?

bobstone94, 2020-04-09:
> @bobstone94 Interesting that we are seeing different behavior on the same project. I'll look into this asap. Are you using UE 4.24.3?
> […](#)

Yes, the only way i can continue using UE4.24 with AirSim is I copy the entire Blocks folder out and create a new level but that's just a temporary fix.

saihv, 2020-04-09:
Please try the latest source code on master, https://github.com/microsoft/AirSim/pull/2526 has been recently merged which should fix this issue.

madratman, 2020-04-14:
Please reopen if you're still facing any errors

closed 2020-04-14 by madratman