microsoft/AirSim #2527: Communication Error when adding Lidar and using PX4

QiuMingZS, 2020-04-05:
I met some problems when adding Lidar, while SITL without Lidar is okay.
The file "Settings.json" is :

```
{
  "SeeDocsAt": "https://github.com/Microsoft/AirSim/blob/master/docs/settings.md",
  "SettingsVersion": 1.2,
  "SimMode": "Multirotor",

  "Vehicles": {
    "PX4": {
      "VehicleType": "PX4Multirotor",
      "UseSerial": false,
      "UseTcp": true,
      "TcpPort": 4560,
      "ControlPort": 14580,
      "params": {
        "NAV_RCL_ACT": 0,
        "NAV_DLL_ACT": 0,
        "LPE_LAT": 47.641468,
        "LPE_LON": -122.140165,
        "COM_OBL_ACT": 1
      },
      "Sensors":{
        "Lidar": {
          "SensorType":6,
          "Enabled":true,
          "NumberOfChannels":1, 
          "PointsPerSecond":10000, 
          "X":0,
          "Y":0,
          "Z":-0.5,
          "Roll": 0, 
          "Pitch": 0, 
          "Yaw": 0,
          "VerticalFOVUpper": 0,
          "VerticalFOVLower": 0,  
          "HorizontalFOVStart": -90, 
          "HorizontalFOVEnd": 90, 
          "DrawDebugPoints": true,
          "DataFrame": "SensorLocalFrame"
        }
      }
    }
  }
}
```

when I run `make px4_sitl_default none_iris`, the terninal get:
```
fire@fire-P7xxTM1:~/PX4/Firmware$ sudo make px4_sitl_default none_iris
[sudo] password for fire: 
[0/1] cd /home/fire/PX4/Firmware/build...re/PX4/Firmware/build/px4_sitl_default
SITL ARGS
sitl_bin: /home/fire/PX4/Firmware/build/px4_sitl_default/bin/px4
debugger: none
program: none
model: iris
src_path: /home/fire/PX4/Firmware
build_path: /home/fire/PX4/Firmware/build/px4_sitl_default
SITL COMMAND: "/home/fire/PX4/Firmware/build/px4_sitl_default/bin/px4" "/home/fire/PX4/Firmware"/ROMFS/px4fmu_common -s etc/init.d-posix/rcS -t "/home/fire/PX4/Firmware"/test_data
INFO  [px4] Creating symlink /home/fire/PX4/Firmware/ROMFS/px4fmu_common -> /home/fire/PX4/Firmware/build/px4_sitl_default/tmp/rootfs/etc

______  __   __    ___ 
| ___ \ \ \ / /   /   |
| |_/ /  \ V /   / /| |
|  __/   /   \  / /_| |
| |     / /^\ \ \___  |
\_|     \/   \/     |_/

px4 starting.

INFO  [px4] Calling startup script: /bin/sh etc/init.d-posix/rcS 0
  SYS_AUTOSTART: curr: 0 -> new: 10016
  BAT_N_CELLS: curr: 0 -> new: 3
  CAL_ACC0_ID: curr: 0 -> new: 1376264
  CAL_ACC1_ID: curr: 0 -> new: 1310728
  CAL_ACC_PRIME: curr: 0 -> new: 1376264
  CAL_GYRO0_ID: curr: 0 -> new: 2293768
  CAL_GYRO_PRIME: curr: 0 -> new: 2293768
  CAL_MAG0_ID: curr: 0 -> new: 196616
  CAL_MAG_PRIME: curr: 0 -> new: 196616
  COM_DISARM_LAND: curr: -1.0000 -> new: 0.1000
  COM_OBL_ACT: curr: 0 -> new: 2
  COM_OF_LOSS_T: curr: 0.0000 -> new: 5.0000
  COM_RC_IN_MODE: curr: 0 -> new: 1
  EKF2_ANGERR_INIT: curr: 0.1000 -> new: 0.0100
  EKF2_GBIAS_INIT: curr: 0.1000 -> new: 0.0100
  MC_PITCH_P: curr: 6.5000 -> new: 6.0000
  MC_PITCHRATE_P: curr: 0.1500 -> new: 0.2000
  MC_ROLL_P: curr: 6.5000 -> new: 6.0000
  MC_ROLLRATE_P: curr: 0.1500 -> new: 0.2000
  MPC_HOLD_MAX_Z: curr: 0.6000 -> new: 2.0000
  MPC_Z_VEL_I: curr: 0.0200 -> new: 0.1500
  MPC_Z_VEL_P: curr: 0.2000 -> new: 0.6000
  MPC_XY_P: curr: 0.9500 -> new: 0.8000
  MPC_XY_VEL_P: curr: 0.0900 -> new: 0.2000
  MPC_XY_VEL_D: curr: 0.0100 -> new: 0.0160
  MPC_JERK_MIN: curr: 8.0000 -> new: 10.0000
  MPC_ACC_HOR_MAX: curr: 10.0000 -> new: 3.0000
  NAV_ACC_RAD: curr: 10.0000 -> new: 2.0000
  NAV_DLL_ACT: curr: 0 -> new: 2
  RTL_DESCEND_ALT: curr: 30.0000 -> new: 5.0000
  RTL_LAND_DELAY: curr: -1.0000 -> new: 5.0000
  RTL_RETURN_ALT: curr: 60.0000 -> new: 30.0000
  SDLOG_MODE: curr: 0 -> new: 1
  SDLOG_PROFILE: curr: 3 -> new: 131
  SDLOG_DIRS_MAX: curr: 0 -> new: 7
  SENS_BOARD_X_OFF: curr: 0.0000 -> new: 0.0000
  SENS_DPRES_OFF: curr: 0.0000 -> new: 0.0010
* RTL_DESCEND_ALT: curr: 5.0000 -> new: 10.0000
* RTL_LAND_DELAY: curr: 5.0000 -> new: 0.0000
  PWM_MAX: curr: 2000 -> new: 1950
  PWM_MIN: curr: 1000 -> new: 1075
INFO  [dataman] Unknown restart, data manager file './dataman' size is 11798680 bytes
INFO  [simulator] Waiting for simulator to connect on TCP port 4560
INFO  [simulator] Simulator connected on TCP port 4560.
INFO  [init] Mixer: etc/mixers/quad_w.main.mix on /dev/pwm_output0
INFO  [mavlink] mode: Normal, data rate: 4000000 B/s on udp port 14570 remote port 14550
INFO  [mavlink] mode: Onboard, data rate: 4000000 B/s on udp port 14580 remote port 14540
INFO  [logger] logger started (mode=all)
INFO  [logger] Start file log (type: full)
INFO  [logger] Opened full log file: ./log/2020-04-05/13_44_20.ulg
INFO  [mavlink] MAVLink only on localhost (set param MAV_BROADCAST = 1 to enable network)
INFO  [px4] Startup script returned successfully
pxh> INFO  [mavlink] partner IP: 127.0.0.1
INFO  [mavlink] partner IP: 127.0.0.1
```
and UE4 shows:

[image]

Thus, I can not arm the drone using QGC.
It need to be mentioned that , when Settings.json has nothing about Lidar, it works well.

rajat2004, 2020-04-05:
Hi, please have a look at https://github.com/microsoft/AirSim/blob/master/docs/sensors.md#configuring-vehicle-specific-sensor-list, specifically
`If a vehicle provides its sensor list, it must provide the whole list. Selective add/remove/update of the default sensor list is NOT supported.`
Therefore you need to add GPS, barometer, IMU to the list as well, like how it's given in the example above that

QiuMingZS, 2020-04-05:
> Hi, please have a look at https://github.com/microsoft/AirSim/blob/master/docs/sensors.md#configuring-vehicle-specific-sensor-list, specifically
> `If a vehicle provides its sensor list, it must [...]

thank you so much!   
it has confused me a long time, aha

QiuMingZS, 2020-04-05:
> Hi, please have a look at https://github.com/microsoft/AirSim/blob/master/docs/sensors.md#configuring-vehicle-specific-sensor-list, specifically
> `If a vehicle provides its sensor list, it must [...]

sorry to bother you again, with your help, now it is ok to use PX4.
There is another problem,  and my drone fly here and there 
[image]

```
ERROR [sensors] Gyro #0 fail:  TIMEOUT!
ERROR [sensors] Sensor Gyro #0 failed. Reconfiguring sensor priorities.
WARN  [sensors] Remaining sensors after failover event 0: Gyro #0 priority: 1
ERROR [sensors] Accel #0 fail:  TIMEOUT!
ERROR [sensors] Sensor Accel #0 failed. Reconfiguring sensor priorities.
```
and my setting.json file is:
```
{
  "SeeDocsAt": "https://github.com/Microsoft/AirSim/blob/master/docs/settings.md",
  "SettingsVersion": 1.2,
  "SimMode": "Multirotor",

  "Vehicles": {
    "PX4": {
      "VehicleType": "PX4Multirotor",
      "UseSerial": false,
      "UseTcp": true,
      "TcpPort": 4560,
      "ControlPort": 14580,
      "params": {
        "NAV_RCL_ACT": 0,
        "NAV_DLL_ACT": 0,
        "LPE_LAT": 47.641468,
        "LPE_LON": -122.140165,
        "COM_OBL_ACT": 1
      },
      "Sensors": {
        "Barometer": {
          "SensorType": 1,
          "Enabled" : true
        },
        "Imu": {
          "SensorType": 2,
          "Enabled" : true
        },
        "Gps": {
          "SensorType": 3,
          "Enabled" : true
        },
        "Magnetometer": {
          "SensorType": 4,
          "Enabled" : true
        },
        "Distance": {
          "SensorType": 5,
          "Enabled" : true
        },
        "Lidar": {
          "SensorType":6,
          "Enabled":true,
          "NumberOfChannels":1, 
          "PointsPerSecond":10000, 
          "X": 0,
          "Y": 0,
          "Z": -0.5,
          "Roll": 0, 
          "Pitch": 0, 
          "Yaw": 0,
          "VerticalFOVUpper": 0,
          "VerticalFOVLower": 0,  
          "HorizontalFOVStart": -90, 
          "HorizontalFOVEnd": 90, 
          "DrawDebugPoints": true,
          "DataFrame": "SensorLocalFrame"
        }
      }
    }
  }
}
```
Do I need to do more with the other sensors(1 - 5)? I have not found related docs
Thx so much.

rajat2004, 2020-04-05:
The sensors look correct, nothing more should be required. This seems to be a connection problem to me, are you running them on different systems or some other network configuration?
Well, if that was the case, then even those messages won't have come.

Googling seems to indicate that this is caused by the computer not being powerful enough to keep up the simulation. Have you disabled the `Use Less CPU while in background` checkbox? If not, then that could be the culprit.
Also, from the images, you're running a different env, possibly the City or something else in the Editor. Running those will definitely require a powerful GPU. Disable the subwindows, etc and see if it helps. You could also try using the latest binaries, they will definitely be lighter on the GPU than running inside the Editor

QiuMingZS, 2020-04-06:
emmm, `Use Less CPU while in background` has been disabled, and the PC has i9-9900KF, RTX2080 and 32GB ram, the problem is some confused.
Thank you for  your help.

bys1123, 2020-04-06:
[image]
[image]
I didn't open any extra sensors, but I got sensor missing errors in windows cygwin environment.

QiuMingZS, 2020-04-06:
> [image]
> [image]
> I didn't open any extra sensors, but I got sensor missing errors in windows cygwin environment.

I have tested this before and it's okay in Ubuntu, but I don't kown about windows environment

rajat2004, 2020-04-08:
https://github.com/microsoft/AirSim/pull/2549 should fix the problems

closed 2020-08-19 by jonyMarino