microsoft/AirSim #2528: getCarControls() throwing error

omarsamyb, 2020-04-05:
Hello everyone, i am trying to record data for a deep learning purpose while driving the car manually. When i try to call getCarControls() to get the steering , throttle values etc.. it throws this error in console
msgpackrpc.error.RPCError: rpclib: server could not find function 'getCarControls' with argument count 1.
although getCarState works just fine.
I'm on Windows 10 using python 3.7 and airsim 1.2.7 package.
I tried searching this issue but found minimal results with no answers to them.
TIA

rajat2004, 2020-04-05:
Hi, 
`getCarControls()` is working correctly for me on Linux with the latest AirSim master as well as the release. Are you using a release binary, or built from source? If it's a release, please try the latest v1.3.0

omarsamyb, 2020-04-06:
> Hi,
> `getCarControls()` is working correctly for me on Linux with the latest AirSim master as well as the release. Are you using a release binary, or built from source? If it's a release, please try the latest v1.3.0

Hi sorry for late reply, i am using binary release from here [https://github.com/microsoft/AirSim/releases/tag/v.1.2.2](url)
using the CityEnviron environment and the latest airsim package 1.2.7 from pip, i noticed that the CityEnviron is under v1.2.2 release , could this be causing the problem like getCarControls() is not compatible in this?

rajat2004, 2020-04-06:
Yes, commit f1bb14c which added the getCarControls API was done after the 1.2.2 release. It's working correctly in master and the latest release

omarsamyb, 2020-04-07:
> Yes, commit [f1bb14c](https://github.com/microsoft/AirSim/commit/f1bb14c2f7d3c86252882a02ec4e65c193c24487) which added the getCarControls API was done after the 1.2.2 release. It's working correctly in master and the latest release

ok thank you, is it possible that any time soon to have the city environment binary posted in the latest release?

closed 2020-04-09 by omarsamyb