microsoft/AirSim #2530: request ApiControl() must be called before using API control

johan606303, 2020-04-05:
Hi. Can someone please wrtite a detailed explanation of how to solve this issue ? I'm new with airsim, and when I tried to follow one of tutorial video, I got this issue when I tried to manualy control a quadcopter.

Thanks.

rajat2004, 2020-04-06:
Don't see what's unclear about the statement, `enableAPIcontrol()` should be called before using any APIs to control the movement
https://github.com/microsoft/AirSim/blob/master/docs/apis.md#common-apis
See the examples in PythonClient/multirotor for exmaples

johan606303, 2020-04-06:
And where am I supposed to write those commands ? When I launched the Blocks, Airsim.sln, DroneShell.exe ?

rajat2004, 2020-04-06:
The video you're following seems to be quite an old one, don't know if Droneshell is being actively maintained or used. I haven't myself used it ever
These are Python or C++ APIs, for Python, can be run from a python terminal with the `airsim` package installed. There are ready-made examples in the PythonClient folder to experiment and use the APIs

johan606303, 2020-04-06:
DroneShell works, just need to write "requestcontrol" first.

johan606303, 2020-04-06:
Then is it possible to pose an another question here, a bit different, not about manual control anymore ?

closed 2020-04-08 by johan606303

majingchunn, 2024-05-05:
> DroneShell 可以工作，只需要先写“requestcontrol”即可。

我不太理解，请问有具体操作嘛？在哪里写requestcontrol？