microsoft/AirSim #2532: Trouble building

kurthill, 2020-04-05:
Hi Everyone, thanks for doing what you do.

I have been following the guide to get AirSim to build in Windows 10, but I am struggling. Here is what I've done:

1) Downloaded and installed Unreal Engine 4.24.3
2) Downloaded Visual Studio 2017 with the following selections:
[image]
3) Cloned the repo using a "x64 Native Tools Command Prompt for VS 2017"
4) Build with "build.cmd

Here I get: 
"Starting cmake to build rpclib...
CMake Error: Could not create named generator Visual Studio 16 2019"

Why is it looking for VS 2019? Searching the internet, I see that this can happen with out of date CMake. The version downloaded with VS is "cmake version 3.12.18081601-MSVC_2". If I update to the latest cmake, I get:

"Found cmake version: 3.17.0
Starting cmake to build rpclib...
CMake Error at CMakeLists.txt:2 (project):
  Generator

    Visual Studio 16 2019

  could not find any instance of Visual Studio."

Alternatively, if I use VS 2019 with the latest SDK, it seems to compile, but there are no .sln files in Unreal/Environments/Blocks directory as the documentations says there should be.

I am not too familiar with developing in windows, so I am probably just doing something stupid. Any ideas?

johan606303, 2020-04-05:
I tried Visual Studio 2019, and I have .sln file at Unreal/Environments/Blocks, just don't forget to run the file "update_from_git.bat". I had this error, to solve it I have just unistalled VS and installed again.

saihv, 2020-04-06:
Please note that we have upgraded this project to Visual Studio 2019. See [build instructions](https://github.com/microsoft/AirSim/blob/master/docs/build_windows.md)

As to the .sln in Blocks, you need to generate it manually by right clicking the uproject and selecting "Generate Visual Studio project files".

kurthill, 2020-04-06:
Thanks for the help! I was following along with the docs here (which I guess is an older version): https://microsoft.github.io/AirSim/build_windows/

I'll give the update_from_git.bat a shot. Thanks again!

closed 2020-04-06 by kurthill