microsoft/AirSim #2533: Bump to UE 4.24 beaks nvidia-docker due to Vulkan

petergerten, 2020-04-06:
While the latest switch of the recommended UE4 version to 4.24 looks good with Vulkan support that seems to improve Linux performance - there seems to be a major issue: no support for nvidia docker.

We use nvidia-docker for all of our UE4/AirSim environments on Linux as they provide a reproducible environment to build and develop. Unfortunately there seems to be no official Vulkan support for nvidia docker. We have tried some suggestions from various threads but could not get it working either.

Is anyone here using nvidia docker as well? Any success with Vulkan and a Dockerfile to share?

petergerten, 2020-04-06:
@saihv  I just notice here https://github.com/adamrehn/ue4-docker/issues/76 that you have tried it as well. Any progress?

saihv, 2020-04-06:
@petergerten Nope, unfortunately Vulkan support is broken from UE's side, and it's likely it might get fixed in UE 4.25. The only way out for now to my knowledge is to run it in OpenGL mode.

`<project-name>.sh -opengl4`

If your environment is not too intensive, openGL seems to work okay.

petergerten, 2020-04-06:
@saihv thanks, will try to force openGL mode

rajat2004, 2020-04-06:
`-opengl4` works with Editor as well, had to uncomment a line - https://www.reddit.com/r/unrealengine/comments/e4i7ny/cannot_start_unreal_engine_4_in_linux/
Incidentally, OpenGL runs smoother for me on Linux than Vulkan

petergerten, 2020-04-06:
@rajat2004 thanks - I was just going to try if the editor also works without Vulkan

adamrehn, 2020-04-18:
@petergerten the ue4-docker issue regarding Vulkan support now has a resolution that outlines the available options: https://github.com/adamrehn/ue4-docker/issues/76#issuecomment-615577529

jonyMarino, 2020-10-19:
Issue resolved. Doc update pending

closed 2020-10-19 by jonyMarino