microsoft/AirSim #2535: unable to locate clang-8

NishanthARao, 2020-04-06:
I am using Ubuntu 18.04. I have built Unreal engine and now I am about to install Airsim as per the instructions provided here: https://microsoft.github.io/AirSim/build_linux/

When I run the command ./setup.sh, there are no errors, in fact at the end it says Airsim Setup completed successfully!, however when i scroll up, i see this:
`
E: Unable to locate package clang-8 
E: Unable to locate package clang++-8
E: Couldn't find any package by regex 'clang++-8'
E: Unable to locate package libc++-8-dev
E: Couldn't find any package by regex 'libc++-8-dev'
E: Unable to locate package libc++abi-8-dev
E: Couldn't find any package by regex 'libc++abi-8-dev'
`

I did not notice this when i tried earlier,and when i ran the next command `./build.sh`, I get the following error:
`Could not find compiler set in environment variable CC: clang-8.
 `
Thankfully, in some answer they had pointed out that the software is missing, and then i noticed the output of the `./setup.sh`, which said that clang-8 is missing.

I also checked the official package website, and it says that clang-8 has been removed: https://www.ubuntuupdates.org/package/ubuntu_mozilla_security/bionic/main/base/clang-8

What do I do here? Can someone please help me out? Thanks

rajat2004, 2020-04-06:
Have you by ay chance disabled the Universe repository in which the `clang` packages are present? You can check this in Software & Updates 
The link you've put is for the `ubuntu_mozilla_security` PPA where it has been removed from, it's there in the Universe repo - https://www.ubuntuupdates.org/package_metas?exact_match=1&q=clang-8

NishanthARao, 2020-04-06:
I am extremely sorry for posting the wrong link. 

Yes. The Universe repository is ticked in the software and updates. However, I am still unable to install it via sudo apt-get install. What could be wrong?

rajat2004, 2020-04-06:
Then a few more things you can try, `sudo apt update` to update the package lists and try again, and check if the package is present in any repo using `apt-cache search clang-8`
If nothing works, then you can add the LLVM repo and then try installing again
```
wget -O - http://apt.llvm.org/llvm-snapshot.gpg.key | sudo apt-key add -
sudo apt-get update
```
This is quite baffling, since installing the packages works correctly on Travis as well

NishanthARao, 2020-04-06:
Thank you. However, the command 
`wget -O - http://apt.llvm.org/llvm-snapshot.gpg.key | sudo apt-key add -` takes forever. I don't see anything after ".... written to stdout". It freezes after that and the cursor keeps blinking.

rajat2004, 2020-04-06:
Maybe there's a problem with permission? Just tried this out in a docker container, and working correctly for me

```
root@47132d555bef:/# wget -O - http://apt.llvm.org/llvm-snapshot.gpg.key | apt-key add -
URL transformed to HTTPS due to an HSTS policy
--2020-04-06 15:12:48--  https://apt.llvm.org/llvm-snapshot.gpg.key
Resolving apt.llvm.org (apt.llvm.org)... 151.101.14.49, 2a04:4e42:3::561
Connecting to apt.llvm.org (apt.llvm.org)|151.101.14.49|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 3145 (3.1K) [application/octet-stream]
Saving to: 'STDOUT'

-                                                  100%[================================================================================================================>]   3.07K  --.-KB/s    in 0s      

2020-04-06 15:12:50 (26.3 MB/s) - written to stdout [3145/3145]

OK
```
Try running the two commands separately, the `wget http://apt.llvm.org/llvm-snapshot.gpg.key` which will save the key to a file, and then `sudo apt-key add <file-name>` and see what happens

NishanthARao, 2020-04-06:
I figured out what the problem was:

I was observing that for quite some time, I wasn't recieving any package upgrades when I executed the command `sudo apt-get upgrade`. Then, I checked the `software and updates` tool, for some insight of what was going on. For some reason, the `Important security updates` and the `Recommended updates` were unticked in the `updates` tab of the `software and updates` were unticked.
[image]
After ticking them, and updating the sources files, (By just clicking close, followed by clicking reload), I was able to upgrade all existing packages worth a ton of GBs.

After this, the command `sudo apt-get install -y clang-8 clang++-8 libc++-8-dev libc++abi-8-dev`  worked perfectly well (alternately, you could run `.\setup.sh` again), and the AirSim package was built successfully. 

Perhaps, if there is some sort of warning that .\setup.sh file could give, when it couldn't find the llvm packages instead of printing `AirSim setup completed successfully` could help save time in debugging these issues. Adding the following lines after `sudo apt-get install -y clang-8 clang++-8 libc++-8-dev libc++abi-8-dev` to check if the packages were installed would give a better picture:
```
if [ "$(dpkg -l | awk '/clang-8/ {print }'|wc -l)" -ge 1 ]; then 
  echo OK 
else 
  echo FAIL 
fi 
```

@rajat2004 , Thank you so much for your precious time!

closed 2020-04-06 by NishanthARao