microsoft/AirSim #2541: Got error when build AirSim for Linux_AArch64 platform:Error: AutomationTool was unable to run successfully

suchb, 2020-04-07:
Hello,
Bellow is the error info:
[2020.04.07-01.39.54:728][ 59]UATHelper: Packaging (Linux (AArch64)): Took 2498.865683s to run mono, ExitCode=5
[2020.04.07-01.39.54:728][ 59]UATHelper: Packaging (Linux (AArch64)): UnrealBuildTool failed. See log for more details. (/home/su/Library/Logs/Unreal Engine/LocalBuildLogs/UBT-Blocks-LinuxAArch64-Development_2.txt)
[2020.04.07-01.39.54:728][ 59]UATHelper: Packaging (Linux (AArch64)): AutomationTool exiting with ExitCode=5 (5)
[2020.04.07-01.39.54:728][ 59]UATHelper: Packaging (Linux (AArch64)): RunUAT ERROR: AutomationTool was unable to run successfully.
[2020.04.07-01.39.54:728][ 59]PackagingResults: Error: AutomationTool was unable to run successfully.
[2020.04.07-01.39.54:728][ 59]PackagingResults: Error: Unknown Error

Does Airsim support LinuxAArch64?

rajat2004, 2020-04-07:
You can try to build for the target platform, the Logs will be helpful to figure out where the compilation error is coming from.
Note that currently packaging a project on Linux on UE 4.24 is broken, see https://github.com/microsoft/AirSim/pull/2522#issuecomment-609554626 due to problem in the Engine code, if it's the same error, then manual application of the commit should work

saihv, 2020-04-07:
Interesting - we might not support AArch64 at all, regardless of broken packaging. AArch64 is for ARM, which platform are you targeting with your packaging?

suchb, 2020-04-08:
> @suchb We might not support AArch64 at all, regardless of broken packaging. AArch64 is for ARM, which platform are you targeting with your packaging?

Hello,saihv,
I want run AirSim on Nvidia Jetson-TX2 platform(ubuntu1804 ARM64),Could you please point me how to make it work on tx2?

Looking forward to your reply. Thanks

suchb, 2020-04-08:
@saihv 
I am still working on it.
When packaging AirSim to LinuxAArch64，I got error：
In file included from F:/AirSim/AirSim/Unreal/Environments/Blocks/Plugins/AirSim/Intermediate/Build/LinuxAArch64/93821A52/UE4/Development/AirSim/Module.AirSim.cpp:3:
F:\AirSim\AirSim\Unreal\Environments\Blocks\Plugins\AirSim\Source\AirLib\src\api\RpcLibClientBase.cpp(30,10): fatal error: 'rpc/client.h' file not found

saihv, 2020-04-08:
@suchb We do not support ARM platforms with AirSim, the computational requirements of AirSim are far beyond what a Jetson TX2 can provide. Compiling Unreal Engine is also going to be quite hard on ARM. 

You can run AirSim on an x64 platform: Ubuntu, Windows or Mac.

suchb, 2020-04-10:
> @suchb We do not really support ARM platforms with AirSim, the graphical requirements of AirSim are far beyond what a Jetson TX2 can provide. Compiling Unreal Engine is also going to be quite hard on ARM.
> 
> You can run AirSim on an x64 platform: Ubuntu, Windows or Mac.

Hi saihv,

Could you please tell me is there a plan to support Linux Arm64 platform?

thanks

madratman, 2020-04-13:
No, I don't see how you expect Unreal Engine to run in linux arm64. 
What is the higher level goal you're trying to accomplish?

suchb, 2020-04-14:
> No, I don't see how you expect Unreal Engine Editor to run in linux arm64. If your goal is to make arm64 binary, unreal engine has documentation for cross-compiling a binary for that platform.
> What is the higher level goal you're trying to accomplish?

Cross-compiling UE4 for linux arm64 was already success in my case, but UE4 for linux arm64 have no EDITOR, So I Compiled UE4-Editor on Ubuntu x64 platform, and Compiled Airsim source code on it, I got error when package Airsim to Linux arm64.  I want to run Airsim on Linux arm64 platform for UAV simulation.

closed 2020-04-22 by suchb

ale99viv, 2021-06-28:
Any solution for this problem?