microsoft/AirSim #2545: Problem compiling landscapeMountains with AirSim

kurthill, 2020-04-07:
Hi Everyone,

I'm following along with the tutorial to setup AirSim with LandscapeMountains (LM) here: https://github.com/microsoft/AirSim/blob/master/docs/unreal_custenv.md

I have compiled AirSim with VS2019 and have the stand-alone LM game working in Unreal editor. I am now trying to compile LM in VS after copying the plugins into the project directory and editing the .uproject file to add the AdditionalDependencies etc. So I have done steps 1-8 without issue.

On step 9 I press F5 to compile and run. I get the following output with errors:

> 1>------ Build started: Project: LandscapeMountains, Configuration: DebugGame_Editor x64 ------
1>Building LandscapeMountainsEditor...
1>Using Visual Studio 2019 14.25.28612 toolchain (C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.25.28610) and Windows 10.0.18362.0 SDK (C:\Program Files (x86)\Windows Kits\10).
1>Building 5 actions with 12 processes...
1>  [1/5] Module.AirSim.cpp
1>C:/Users/Kurt/Documents/Unreal Projects/LandscapeMountains/Plugins/AirSim/Source/AirBlueprintLib.cpp(415): error C4456: declaration of 'IndexBuffer' hides previous local declaration
1>  C:/Users/Kurt/Documents/Unreal Projects/LandscapeMountains/Plugins/AirSim/Source/AirBlueprintLib.cpp(406): note: see declaration of 'IndexBuffer'
1>C:/Users/Kurt/Documents/Unreal Projects/LandscapeMountains/Plugins/AirSim/Source/AirBlueprintLib.cpp(416): error C4456: declaration of 'data' hides previous local declaration
1>  C:/Users/Kurt/Documents/Unreal Projects/LandscapeMountains/Plugins/AirSim/Source/AirBlueprintLib.cpp(396): note: see declaration of 'data'
1>C:/Users/Kurt/Documents/Unreal Projects/LandscapeMountains/Plugins/AirSim/Source/AirBlueprintLib.cpp(439): error C4456: declaration of 'IndexBuffer' hides previous local declaration
1>  C:/Users/Kurt/Documents/Unreal Projects/LandscapeMountains/Plugins/AirSim/Source/AirBlueprintLib.cpp(406): note: see declaration of 'IndexBuffer'
1>C:/Users/Kurt/Documents/Unreal Projects/LandscapeMountains/Plugins/AirSim/Source/AirBlueprintLib.cpp(440): error C4456: declaration of 'data' hides previous local declaration
1>  C:/Users/Kurt/Documents/Unreal Projects/LandscapeMountains/Plugins/AirSim/Source/AirBlueprintLib.cpp(396): note: see declaration of 'data'
1>C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\MSBuild\Microsoft\VC\v160\Microsoft.MakeFile.Targets(44,5): error MSB3075: The command ""C:\Program Files\Epic Games\UE_4.24\Engine\Build\BatchFiles\Build.bat" LandscapeMountainsEditor Win64 DebugGame -Project="C:\Users\Kurt\Documents\Unreal Projects\LandscapeMountains\LandscapeMountains.uproject" -WaitMutex -FromMsBuild" exited with code 5. Please verify that you have sufficient rights to run this command.
1>Done building project "LandscapeMountains.vcxproj" -- FAILED.
========== Build: 0 succeeded, 1 failed, 0 up-to-date, 0 skipped ==========

The C4456 seems like it should be a warning... I'm not sure about MSB3075.

Do you have any ideas?

Thanks,
Kurt

rajat2004, 2020-04-08:
#2526 has been merged which fixes this problem, so I think this can be closed now

kurthill, 2020-04-09:
Thanks, pulling the latest version fixed the issue.

closed 2020-04-09 by kurthill