microsoft/AirSim #2546: try to control the drone by using api

dreamerlang, 2020-04-08:
# Read This First

## If you are reporting a bug
* Make sure to write **all reproduction steps**
* Include full error message in text form
* Search issues for error message before filing issue
* Attach screenshot if applicable
* Include code to run if applicable

## If you have question
* Add clear and concise title
* Add OS, AirSim version, Python version, Unreal version if applicable
* Include context on what you are trying to achieve
* Include details of what you already did to find answers

**What's better than filing issue? Filing a pull request :).**

------------------------------------ (Remove above before filing the issue) ------------------------------------

rajat2004, 2020-04-08:
You seem to have modified the default settings for PX4, along with the port numbers. Try changing it back to the default TCP and port numbers and see if happens, I think the latest PX4 requires TCP for Lockstep scheduling

madratman, 2020-04-14:
```json
{
  "SettingsVersion": 1.2,
  "SimMode": "Multirotor",
  "Vehicles": {
      "PX4": {
          "VehicleType": "PX4Multirotor",
          "UseSerial": false,
          "UseTcp": true,
          "TcpPort": 4560,
          "ControlPort": 14580,
          "params": {
              "NAV_RCL_ACT": 0,
              "NAV_DLL_ACT": 0,
              "LPE_LAT": 47.641468,
              "LPE_LON": -122.140165,
              "COM_OBL_ACT": 1
          }
      }
  }
}
```
px4 v1.10.1 should work
https://microsoft.github.io/AirSim/px4_sitl/

closed 2020-04-14 by madratman