microsoft/AirSim #2548: Problem compiling landscapeMountains with AirSim

wang6895, 2020-04-08:
Hi Everyone,

I have compiled AirSim1.2.2 with VS2017.

I'm following along with the tutorial to setup AirSim with LandscapeMountains (LM) here: 
https://github.com/microsoft/AirSim/blob/master/docs/unreal_custenv.md

On step 9 I press F5 to compile and run. I get the following output with errors:

1>------ Build started: Project: LandscapeMountains, Configuration: DebugGame_Editor x64 ------
1>Creating makefile for LandscapeMountainsEditor (no existing makefile)
1>UnrealBuildTool : warning : Library 'Shell32.lib' was not resolvable to a file when used in Module 'AirSim', assuming it is a filename and will search library paths for it. This is slow and dependency checking will not work for it. Please update reference to be fully qualified alternatively use PublicSystemLibraryPaths if you do intended to use this slow path to suppress this warning.
1>UnrealBuildTool : warning : Library 'dinput8.lib' was not resolvable to a file when used in Module 'AirSim', assuming it is a filename and will search library paths for it. This is slow and dependency checking will not work for it. Please update reference to be fully qualified alternatively use PublicSystemLibraryPaths if you do intended to use this slow path to suppress this warning.
1>UnrealBuildTool : warning : Library 'dxguid.lib' was not resolvable to a file when used in Module 'AirSim', assuming it is a filename and will search library paths for it. This is slow and dependency checking will not work for it. Please update reference to be fully qualified alternatively use PublicSystemLibraryPaths if you do intended to use this slow path to suppress this warning.
1>Parsing headers for LandscapeMountainsEditor
1>  Running UnrealHeaderTool "G:\UE_4.24\project\LandscapeMountains\LandscapeMountains.uproject" "G:\UE_4.24\project\LandscapeMountains\Intermediate\Build\Win64\LandscapeMountainsEditor\DebugGame\LandscapeMountainsEditor.uhtmanifest" -LogCmds="loginit warning, logexit warning, logdatabase error" -Unattended -WarningsAsErrors -installed
1>G:/UE_4.24/project/LandscapeMountains/Plugins/AirSim/Source/CameraDirector.h(24) : LogCompile: Error: 'ECameraDirectorMode' does not have a 0 entry! (This is a problem when the enum is initalized by default)
1>C:\Program Files (x86)\Microsoft Visual Studio\2017\Community\Common7\IDE\VC\VCTargets\Microsoft.MakeFile.Targets(44,5): error MSB3075: The command "G:\UE_4.24\Engine\Build\BatchFiles\Build.bat LandscapeMountainsEditor Win64 DebugGame -Project="G:\UE_4.24\project\LandscapeMountains\LandscapeMountains.uproject" -WaitMutex -FromMsBuild" exited with code 5. Please verify that you have sufficient rights to run this command.
1>Done building project "LandscapeMountains.vcxproj" -- FAILED.
========== Build: 0 succeeded, 1 failed, 0 up-to-date, 0 skipped ==========

Thanks

rajat2004, 2020-04-08:
Hi,
You seem to be building an earlier version of AirSim. This problem has been fixed in master, please try that

lfrusse, 2020-04-08:
I'm also experiencing similar issue: 
[image]

madratman, 2020-04-08:
the terminal says VS 2017. Please use VS 2019

saihv, 2020-04-08:
@lfrusse 

Please pull the latest version of the source code (or use our [newly released binaries](https://github.com/microsoft/AirSim/releases/tag/v1.3.1-windows)), and make sure you upgrade to VS 2019 + UE 4.24 if you're using the source code. You can also try removing all untracked files using ` git clean -fdx` from your workspace.

wang6895, 2020-04-09:
Thanks.
I need to uninstall VS2017

closed 2020-04-10 by wang6895