microsoft/AirSim #2550: Environment generation issue

johan606303, 2020-04-08:
Hi. I successfully created environments for MontainLandscape, for Blocks, by following tutorial. But when I try to create my own environment, for example the basic "blank", when I execute .sln file, I have build errors.
Nevertheless I added plugins to the project and I changed .uproject where I added :
			"AdditionalDependencies": [
				"AirSim"
			]
to the Module field and 
	"Plugins": [
		{
			"Name": "AirSim",
			"Enabled": true
		}
	]
I don't understand what is wrong.
I use engine 4.24.3, VS2019 and I just try to create Blank environment where I would simulate a multirotor.

Thanks for help.

rajat2004, 2020-04-08:
Can you post the compilation errors which you're getting?

johan606303, 2020-04-08:
Well here it is. Sorry for the wait.
[image]

rajat2004, 2020-04-08:
#2526 has been merged which should fix this error

johan606303, 2020-04-08:
Thank you, I will try.

johan606303, 2020-04-08:
OK. Thank you.

rajat2004, 2020-04-14:
The problem has been fixed, issue can be closed now

closed 2020-04-14 by madratman