microsoft/AirSim #2551: Cannot build airsim ros packages

hkgrov, 2020-04-08:
I am trying to build the Airsim ros packages in WSL on windows 10. I followed the guide https://microsoft.github.io/AirSim/airsim_ros_pkgs/. When running catkin build i get an error saying: c++: error: unrecognized command line option ‘-stdlib=libc++’

Tried checking other issues, but could not find a solution. Have tried setting the CXX compiler to Clang. Then the build did not throw any errors, but would not complete build either. 

Full log can be seen below. 
 ```[35m[1mScanning dependencies of target geometry_msgs_generate_messages_lisp[0m
[35m[1mScanning dependencies of target _airsim_ros_pkgs_generate_messages_check_deps_TakeoffGroup[0m
[35m[1mScanning dependencies of target _airsim_ros_pkgs_generate_messages_check_deps_Land[0m
[35m[1mScanning dependencies of target _airsim_ros_pkgs_generate_messages_check_deps_VelCmd[0m
[35m[1mScanning dependencies of target _airsim_ros_pkgs_generate_messages_check_deps_VelCmdGroup[0m
[35m[1mScanning dependencies of target _airsim_ros_pkgs_generate_messages_check_deps_Takeoff[0m
[35m[1mScanning dependencies of target _airsim_ros_pkgs_generate_messages_check_deps_GPSYaw[0m
[35m[1mScanning dependencies of target geometry_msgs_generate_messages_py[0m
[35m[1mScanning dependencies of target _airsim_ros_pkgs_generate_messages_check_deps_SetGPSPosition[0m
[35m[1mScanning dependencies of target std_msgs_generate_messages_py[0m
[35m[1mScanning dependencies of target _airsim_ros_pkgs_generate_messages_check_deps_GimbalAngleEulerCmd[0m
[35m[1mScanning dependencies of target MavLinkCom[0m
[  0%] Built target std_msgs_generate_messages_py
[  0%] Built target geometry_msgs_generate_messages_lisp
[  0%] Built target geometry_msgs_generate_messages_py
[  0%] Built target _airsim_ros_pkgs_generate_messages_check_deps_TakeoffGroup
[35m[1mScanning dependencies of target _airsim_ros_pkgs_generate_messages_check_deps_SetLocalPosition[0m
[  0%] Built target _airsim_ros_pkgs_generate_messages_check_deps_GPSYaw
[  0%] Built target _airsim_ros_pkgs_generate_messages_check_deps_VelCmdGroup
[  0%] Built target _airsim_ros_pkgs_generate_messages_check_deps_VelCmd
[  0%] Built target _airsim_ros_pkgs_generate_messages_check_deps_Takeoff
[35m[1mScanning dependencies of target _airsim_ros_pkgs_generate_messages_check_deps_LandGroup[0m
[  0%] Built target _airsim_ros_pkgs_generate_messages_check_deps_Land
[  0%] Built target _airsim_ros_pkgs_generate_messages_check_deps_GimbalAngleEulerCmd
[35m[1mScanning dependencies of target _airsim_ros_pkgs_generate_messages_check_deps_Reset[0m
[  1%] [32mBuilding CXX object MavLinkCom/CMakeFiles/MavLinkCom.dir/home/ascend/AirSim/MavLinkCom/common_utils/FileSystem.cpp.o[0m
[  1%] Built target _airsim_ros_pkgs_generate_messages_check_deps_SetGPSPosition
[01m[Kc++:[m[K [01;31m[Kerror: [m[Kunrecognized command line option ‘[01m[K-stdlib=libc++[m[K’
[  2%] [32mBuilding CXX object MavLinkCom/CMakeFiles/MavLinkCom.dir/home/ascend/AirSim/MavLinkCom/common_utils/ThreadUtils.cpp.o[0m
MavLinkCom/CMakeFiles/MavLinkCom.dir/build.make:62: recipe for target 'MavLinkCom/CMakeFiles/MavLinkCom.dir/home/ascend/AirSim/MavLinkCom/common_utils/FileSystem.cpp.o' failed
make[2]: *** [MavLinkCom/CMakeFiles/MavLinkCom.dir/home/ascend/AirSim/MavLinkCom/common_utils/FileSystem.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
[  2%] [32mBuilding CXX object MavLinkCom/CMakeFiles/MavLinkCom.dir/home/ascend/AirSim/MavLinkCom/src/AdHocConnection.cpp.o[0m
[  4%] [32mBuilding CXX object MavLinkCom/CMakeFiles/MavLinkCom.dir/home/ascend/AirSim/MavLinkCom/src/MavLinkFtpClient.cpp.o[0m
[  4%] [32mBuilding CXX object MavLinkCom/CMakeFiles/MavLinkCom.dir/home/ascend/AirSim/MavLinkCom/src/MavLinkConnection.cpp.o[0m
[01m[Kc++:[m[K [01;31m[Kerror: [m[Kunrecognized command line option ‘[01m[K-stdlib=libc++[m[K’
MavLinkCom/CMakeFiles/MavLinkCom.dir/build.make:86: recipe for target 'MavLinkCom/CMakeFiles/MavLinkCom.dir/home/ascend/AirSim/MavLinkCom/common_utils/ThreadUtils.cpp.o' failed
make[2]: *** [MavLinkCom/CMakeFiles/MavLinkCom.dir/home/ascend/AirSim/MavLinkCom/common_utils/ThreadUtils.cpp.o] Error 1
[35m[1mScanning dependencies of target std_msgs_generate_messages_lisp[0m
[35m[1mScanning dependencies of target _airsim_ros_pkgs_generate_messages_check_deps_GimbalAngleQuatCmd[0m
[01m[Kc++:[m[K [01;31m[Kerror: [m[Kunrecognized command line option ‘[01m[K-stdlib=libc++[m[K’
MavLinkCom/CMakeFiles/MavLinkCom.dir/build.make:110: recipe for target 'MavLinkCom/CMakeFiles/MavLinkCom.dir/home/ascend/AirSim/MavLinkCom/src/AdHocConnection.cpp.o' failed
make[2]: *** [MavLinkCom/CMakeFiles/MavLinkCom.dir/home/ascend/AirSim/MavLinkCom/src/AdHocConnection.cpp.o] Error 1
[01m[Kc++:[m[K [01;31m[Kerror: [m[Kunrecognized command line option ‘[01m[K-stdlib=libc++[m[K’
[01m[Kc++:[m[K [01;31m[Kerror: [m[Kunrecognized command line option ‘[01m[K-stdlib=libc++[m[K’
MavLinkCom/CMakeFiles/MavLinkCom.dir/build.make:134: recipe for target 'MavLinkCom/CMakeFiles/MavLinkCom.dir/home/ascend/AirSim/MavLinkCom/src/MavLinkConnection.cpp.o' failed
make[2]: *** [MavLinkCom/CMakeFiles/MavLinkCom.dir/home/ascend/AirSim/MavLinkCom/src/MavLinkConnection.cpp.o] Error 1
make[2]: *** [MavLinkCom/CMakeFiles/MavLinkCom.dir/home/ascend/AirSim/MavLinkCom/src/MavLinkFtpClient.cpp.o] Error 1
MavLinkCom/CMakeFiles/MavLinkCom.dir/build.make:158: recipe for target 'MavLinkCom/CMakeFiles/MavLinkCom.dir/home/ascend/AirSim/MavLinkCom/src/MavLinkFtpClient.cpp.o' failed
CMakeFiles/Makefile2:3604: recipe for target 'MavLinkCom/CMakeFiles/MavLinkCom.dir/all' failed
make[1]: *** [MavLinkCom/CMakeFiles/MavLinkCom.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[35m[1mScanning dependencies of target geometry_msgs_generate_messages_eus[0m
[35m[1mScanning dependencies of target std_msgs_generate_messages_eus[0m
[35m[1mScanning dependencies of target geographic_msgs_generate_messages_cpp[0m
[  4%] Built target _airsim_ros_pkgs_generate_messages_check_deps_SetLocalPosition
[35m[1mScanning dependencies of target std_msgs_generate_messages_nodejs[0m
[  4%] Built target std_msgs_generate_messages_lisp
[35m[1mScanning dependencies of target geometry_msgs_generate_messages_nodejs[0m
[  4%] Built target _airsim_ros_pkgs_generate_messages_check_deps_LandGroup
[  4%] Built target std_msgs_generate_messages_eus
[  4%] Built target geometry_msgs_generate_messages_eus
[  4%] Built target _airsim_ros_pkgs_generate_messages_check_deps_Reset
[  4%] Built target geographic_msgs_generate_messages_cpp
[  4%] Built target std_msgs_generate_messages_nodejs
[  4%] Built target geometry_msgs_generate_messages_nodejs
[  4%] Built target _airsim_ros_pkgs_generate_messages_check_deps_GimbalAngleQuatCmd
Makefile:140: recipe for target 'all' failed
make: *** [all] Error 2
 ```

GimpelZhang, 2020-04-09:
I had the same problem. I built the AirSim folder on a Linux ubuntu 16.04 machine. 
After I changed to the /ros folder and typed catkin_make, I got the same error: 

c++: error: unrecognized command line option ‘-stdlib=libc++’

I also tried to change the compilers to clang: 

catkin_make -DCMAKE_C_COMPILER=/usr/bin/clang-8 -DCMAKE_CXX_COMPILER=/usr/bin/clang-8

or

catkin_make -DCMAKE_CXX_COMPILER=/usr/bin/clang-8

The catkin_make process stopped with an error: 
......
clang: error: linker command failed with exit code 1 (use -v to see invocation)

archsoong, 2020-04-09:
I have the same problem, however I was quite successful on my first attempt, but after I delete and clone again, after `setup.sh` and `build.sh`, `catkin_make` pop up this problem:
c++: error: unrecognized command line option ‘-stdlib=libc++

GimpelZhang, 2020-04-09:
@archsoong 

> I have the same problem, however I was quite successful on my first attempt, but after I delete and clone again, after `setup.sh` and `build.sh`, `catkin_make` pop up this problem:
> c++: error: unrecognized command line option ‘-stdlib=libc++

I guess that's because on your first attempt, the AirSim folder was in an earlier version. After you cloned again, you got the latest version which matched the Unreal Engine 4.24 with clang 8. Then the problem showed up.

manthan99, 2020-04-09:
I am also facing the same issue. I tried the following [link](https://stackoverflow.com/questions/7031126/switching-between-gcc-and-clang-llvm-using-cmake) to switch to clang which removed the error `unrecognized command line option ‘-stdlib=libc++`
Still not building and I am getting 
`AirSim/MavLinkCom/common_utils/FileSystem.hpp:7:10: fatal error: 
      'codecvt' file not found
#include <codecvt>
         ^~~~~~~~~
`

AustinDeric, 2020-04-13:
I am also unable to build this. Here are the sequence of errors i ran into:
1) i ran into the `unrecognized command line option ‘-stdlib=libc++` which i passed by `catkin build -DCMAKE_C_COMPILER=/usr/bin/clang-8 -DCMAKE_CXX_COMPILER=/usr/bin/clang-8` as per this post: https://github.com/microsoft/AirSim/issues/2551#issuecomment-611349717
2) linker not finding references: [catkin_build_error.txt](https://github.com/microsoft/AirSim/files/4467532/catkin_build_error.txt)

Here are some related issues: 
* https://github.com/microsoft/AirSim/pull/2519#issuecomment-612126815 <- 3 days ago this describes our very problem. 
* https://github.com/microsoft/AirSim/issues/2517

madratman, 2020-04-13:
Added a fix here https://github.com/microsoft/AirSim/pull/2571
@AustinDeric yes that comment is relevant, however what mentioned in this comment https://github.com/microsoft/AirSim/pull/2519#issuecomment-608269380 can essentially be added in airsim_ros_pkgs' CMakeLists.txt, which is what PR #2571 is doing

madratman, 2020-04-13:
#2571 is merged, and this issue should be resolved now

closed 2020-04-13 by madratman