microsoft/AirSim #2559: Some problem abot the OSX

Gatsby23, 2020-04-10:
[image]
My mac is 10.15.4 OSX. There are some problem when I run the./build.sh. I have see the pollin and pollout function in the class definition, but I don't know why it will be crashed.

rajat2004, 2020-04-10:
Hi,
Could you test https://github.com/microsoft/AirSim/pull/2560 and see if it fixes the problem? Also, which version of XCode are you using?

Gatsby23, 2020-04-10:
> Hi,
> Could you test #2560 and see if it fixes the problem? Also, which version of XCode are you using?

My Xcode is 11.4, should I change my xcode ?

rajat2004, 2020-04-10:
Not sure, just getting more info as to why the error is occurring, since this hasn't popped up before

Gatsby23, 2020-04-10:
> Not sure, just getting more info as to why the error is occurring, since this hasn't popped up before

OK, I'm sorry, but how to use the #2560? Should I git clone and check to this branch ?

rajat2004, 2020-04-10:
The change is just 1 line right now, you can manually change that and try, else

```
cd AirSim
git checkout master; # update if not already: git pull upstream master --rebase
git clean -fdx
git fetch upstream pull/2560/head:PR/2560; git checkout PR/2560;
./clean.sh
./build.sh;
```

saihv, 2020-04-11:
@Gatsby23 Can you provide some more details about your XCode version? Did you run the latest `setup.sh` before `build.sh`? 

master branch builds fine on my macOS - 10.15.4, XCode 11.4 (11E146)

```
➜  airsim git:(master) ✗ clang -v     
Apple clang version 11.0.3 (clang-1103.0.32.29)
Target: x86_64-apple-darwin19.4.0
Thread model: posix
InstalledDir: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin
```

Gatsby23, 2020-04-11:
> @Gatsby23 Can you provide some more details about your XCode version? Did you run the latest `setup.sh` before `build.sh`?
>
> master branch builds fine on my macOS - 10.15.4, XCode 11.4 (11E146)
>
> ``` [...]

clang version 9.0.1
Target: x86_64-apple-darwin19.4.0
Thread model: posix
InstalledDir: /usr/local/opt/llvm/bin

Gatsby23, 2020-04-11:
> @Gatsby23 Can you provide some more details about your XCode version? Did you run the latest `setup.sh` before `build.sh`?
>
> master branch builds fine on my macOS - 10.15.4, XCode 11.4 (11E146)
>
> ``` [...]

Is this right ?

saihv, 2020-04-11:
Looks like you haven't run the `setup.sh` script, because setup should install llvm 8, not 9. I have updated my previous comment to show you what the right version should look like. When you run setup.sh, the script installs clang-8 from homebrew. You might want to remove clang-9 just in case.

Gatsby23, 2020-04-11:
> Looks like you may not have run the `setup.sh` script, because setup should install llvm 8, not 9. I have updated my previous comment to show you what the right version should look like. When you run setup.sh, the script installs clang-8 from homebrew. In case there is a conflict of versions, you might want to remove clang-9.
> 
> I'd recommend you try this:
> 
> 1. Clone the repo again in a fresh folder.
> 2. Remove clang-9.
> 3. Run `./setup.sh`
> 4. Run `./build.sh`

[image]
That's weired, I have alread run the setup.sh and install the llvm-8, but it still show this.

Gatsby23, 2020-04-11:
> Looks like you may not have run the `setup.sh` script, because setup should install llvm 8, not 9. I have updated my previous comment to show you what the right version should look like. When you [...]

and how to remove clang-9 ?

Gatsby23, 2020-04-11:
> The change is just 1 line right now, you can manually change that and try, else
> 
> ```
> cd AirSim
> git checkout master; # update if not already: git pull upstream master --rebase
> git clean -fdx
> git fetch origin pull/2560/head:PR/2560; git checkout PR/2560;
> ./clean.sh
> ./build.sh;
> ```
> 
> If you have your fork and have set that as `origin` and this main repo as `upstream`, replace `origin` with `upstream`

I have used this command, but there are the same problem. The llvm I installed from brew is the llvm 10. Is this the problem ?

Gatsby23, 2020-04-11:
Another interesting thing is that I can't built the UnrealEngine and there some problem with unrealengine  buiding. It can't make which I have already run the GenerateProjectFiles.sh and setup.sh, maybe I have already installed it from EPIC Games ?
[image]
[image]

rajat2004, 2020-04-11:
`build.sh` automatically uses `clang-8` by specifying the path of the binary as seen [here](https://github.com/microsoft/AirSim/blob/master/build.sh#L68), so you _probably_ don't need to remove the other LLVM installations. Run `clean.sh` & `build.sh` and check the CMake output to make sure that it's using Clang 8. If not, then you might have to remove the conflicting one

Since you already have UE installed from the Epic Games launcher, you don't need to build it from source

Gatsby23, 2020-04-12:
> `build.sh` automatically uses `clang-8` by specifying the path of the binary as seen [here](https://github.com/microsoft/AirSim/blob/master/build.sh#L68), so you _probably_ don't need to remove the [...]

I'm sorry to boring you. But I still get the same problem use this branch. I think the main problem is that it can't find the definition of the member function in this class, but I have already find it in the UdpSocketImpl. I can't figure out why it hasn't link in libMavLinkCom.a？

Gatsby23, 2020-04-12:
I have checked the build.sh and CMakeLists.txt, I find there are some problem with the AirlibUnitTest building. The main problem may be it can't find the lib library. May be there are some problem with the cp lib library ? I can't find the lib in AirLib folder

rajat2004, 2020-04-12:
The `lib` folder is created by `build.sh` [here](https://github.com/microsoft/AirSim/blob/master/build.sh#L111)
Maybe try this once as suggested by @saihv

> I'd recommend you try this:
    Clone the repo again in a fresh folder.
    Run ./setup.sh
    Run ./build.sh

Try building AirSim again, and see if the error still occurs. Paste the outputs here from the commands and the error message. 
I'm out of any ideas right now

Gatsby23, 2020-04-12:
> The `lib` folder is created by `build.sh` [here](https://github.com/microsoft/AirSim/blob/master/build.sh#L111)
> Maybe try this once as suggested by @saihv
>
>
> Try building AirSim again, and see if the [...]

Alright, but I have tried a lot, I can only wait for update

aleriboni, 2020-04-16:
I have the same problem of @Gatsby23 using a MacBook Pro (macOS 10.15.4). The setup completes successfully but I have the same error when I run ./build.sh.
I tried to take your advice but it didn't work. I hope someone can solve it.
Thank you.

rajat2004, 2020-04-17:
This seems to be a bug in XCode 11.4 - https://openradar.appspot.com/FB7647406, and not specific to AirSim, such as https://github.com/curl/curl/issues/5210, https://github.com/mono/mono/issues/19393

aleriboni, 2020-04-17:
Thank you for your advice. Do you know where are the pieces of code that I have to comment?

rajat2004, 2020-04-17:
The functions causing the problem are https://github.com/microsoft/AirSim/blob/master/MavLinkCom/src/impl/UdpSocketImpl.cpp#L154-L197 I think
So you can comment out the stuff inside them and just leave the last `return true` statement inside the functions as it is for compilation

aleriboni, 2020-04-17:
Thank you @rajat2004! I was able to fix the problem temporarily.

wss020, 2020-05-22:
Still stuck in this problem. I have exact the same problem with @Gatsby23. Hoping someone to solve it. Thanks!

rajat2004, 2020-05-22:
@wss020 Travis now has support for XCode 11.4 on Catalina, and the build is working - https://travis-ci.org/github/microsoft/AirSim/jobs/689912647
Maybe try upgrading to the XCode 11.4.1

wss020, 2020-05-22:
> @wss020 Travis now has support for XCode 11.4 on Catalina, and the build is working - https://travis-ci.org/github/microsoft/AirSim/jobs/689912647
> Maybe try upgrading to the XCode 11.4.1

Thanks for the reply. My Xcode is 11.5. Do I need to make it 11.4.1?

rajat2004, 2020-05-22:
That I'm not sure about, maybe the fixes in 11.4.1 haven't been applied to 11.5? Travis doesn't provide that, so can't test.
You could comment out the code which is causing the problem if you aren't aiming to use ArduPilot with AirSim right now, or otherwise try to figure out which flags are being applied and modify them to consider this as warning instead of error

wss020, 2020-05-22:
> That I'm not sure about, maybe the fixes in 11.4.1 haven't been applied to 11.5? Travis doesn't provide that, so can't test.
> You could comment out the code which is causing the problem if you aren't [...]

Thanks! Will try it for now.

dataplayer12, 2020-05-27:
I had the same problem with Xocde 11.5 and was able to fix it by commenting out the two problematic functions, as pointed out by @rajat2004 However, this fix will cause problems if I ever use ardupilot with airsim. Is there a better solution?

hellotanuky, 2020-06-07:
Met the same problem on macOS 10.15.5 & Xcode 11.5.1. I tried https://github.com/microsoft/AirSim/pull/2560 but it didn't work. I solved this by copying `__darwin_fd_set()` function from `<sys/_types/_fd_def.h>`, removing call of __darwin_check_fd_set and then replacing `FD_SET()` with `__darwin_fd_set_temp()` in `MavLinkCom/src/impl/UdpSocketImpl.cpp`.

```diff
__header_always_inline void
__darwin_fd_set_temp(int _fd, struct fd_set *const _p)
{
-   if (__darwin_check_fd_set(_fd, (const void *) _p)) {
	    (_p->fds_bits[(unsigned long)_fd / __DARWIN_NFDBITS] |= ((__int32_t)(((unsigned long)1) << ((unsigned long)_fd % __DARWIN_NFDBITS))));
-   }
}
```

But this is dangerous and may cause security or stability problem.

rajat2004, 2020-06-07:
#2560 won't work, that was just an initial guess at what could be wrong before finding out that it's an XCode bug
@hellotanuky The code replacement does seem pretty strange, and also non-portable. Another method is to comment out the lines causing problem, but not desirable if using ArduPilot. Tried to have a look into where `no_weak_imports` flag is being applied in Airsim but no luck there, might try to dig deeper today
Another option is to find an equivalent replacement for FD_SET

akinerkan, 2020-08-24:
> The functions causing the problem are https://github.com/microsoft/AirSim/blob/master/MavLinkCom/src/impl/UdpSocketImpl.cpp#L154-L197 I think
> So you can comment out the stuff inside them and just leave the last `return true` statement inside the functions as it is for compilation

Mac os x catalina 10.15.6 and it worked for me. Thank you very much

closed 2022-03-31 by jonyMarino