microsoft/AirSim #2563: Blocks fails to link using VS2019

seanmcleod, 2020-04-10:
I cloned the AirSim repo today (10 April) and followed the readme instructions for building on Windows, with Unreal 4.24 installed and using VS2019.

No issues building AirSim using VS2019.

First hint of an issue is when opening the generated solution file for Blocks VS2019 prompted me asking whether to upgrade the toolset to v142 (VS2019), which I did.

Then building Blocks results in a number of linker errors, all do to do with 2 symbols.

```sh
1>rpc.lib(server_session.obj) : error LNK2001: unresolved external symbol __CxxFrameHandler4
1>rpc.lib(client_error.obj) : error LNK2001: unresolved external symbol __GSHandlerCheck_EH4
```

Doing a quick online search it appears to possibly be due to a mixing of VS2017 and VS2019 libs.

And looking at the log, sure enough the VS2017 (which I also have installed) toolchain is being used. 

```sh
1>------ Build started: Project: Blocks, Configuration: DebugGame_Editor x64 ------
1>Creating makefile for BlocksEditor (no existing makefile)
.....
1>  Running UnrealHeaderTool "C:\source\AirSim\Unreal\Environments\Blocks\Blocks.uproject" "C:\source\AirSim\Unreal\Environments\Blocks\Intermediate\Build\Win64\BlocksEditor\DebugGame\BlocksEditor.uhtmanifest" -LogCmds="loginit warning, logexit warning, logdatabase error" -Unattended -WarningsAsErrors -installed
1>Reflection code generated for BlocksEditor in 5.6814853 seconds
1>Building BlocksEditor...
1>Using Visual Studio 2017 14.16.27034 toolchain (C:\Program Files (x86)\Microsoft Visual Studio\2017\Professional\VC\Tools\MSVC\14.16.27023) and Windows 10.0.16299.0 SDK (C:\Program Files (x86)\Windows Kits\10).
```

Looking at Blocks.vcxproj it looks like it makes use of a build batch file from Unreal.

```sh
    <NMakeBuildCommandLine>"C:\Program Files\Epic Games\UE_4.24\Engine\Build\BatchFiles\Build.bat" Blocks Win32 DebugGame -Project="$(SolutionDir)$(ProjectName).uproject" -WaitMutex -FromMsBuild</NMakeBuildCommandLine>
```

Which then calls the UnrealBuildTool.exe to perform the build.

```sh
..\..\Engine\Binaries\DotNET\UnrealBuildTool.exe %*
```

So the question is why or what is used by the UnreadlBuildTool.exe to decide which compiler toolchain to use.

bys1123, 2020-04-10:
https://github.com/microsoft/AirSim/pull/2526#issuecomment-611274394

bys1123, 2020-04-10:
This might help: https://github.com/microsoft/AirSim/pull/2526#issuecomment-611274394

seanmcleod, 2020-04-10:
Thanks @bys1123, editing UnrealBuildTool\BuildConfiguration.xml as per the comment did the trick.

closed 2020-04-10 by seanmcleod