microsoft/AirSim #2566: Generation of Landscape Mountains issue

johan606303, 2020-04-12:
I have an issue with the generation of a new project.
To generate landscape mountains, I follow steps descibed in AirSim tutorial.
I generate Landscape Mountain project from Unreal Project, Library, and launch it.
I create a New C++ class, what triggers the launch of VS2019.
Then I copy in the new project, the folder "Plugins" from ...\Airsim\Unreal.
I modify the "LandscapeMountains.uproject".

Before I had : 
{
	"FileVersion": 3,
	"EngineAssociation": "4.24",
	"Category": "Samples",
	"Description": "",
	"Modules": [
		{
			"Name": "LandscapeMountains",
			"Type": "Runtime",
			"LoadingPhase": "Default"
		}
	],
	"TargetPlatforms": [
		"MacNoEditor",
		"WindowsNoEditor"
	],
	"EpicSampleNameHash": "1226740271"
}

After : 
{
    "FileVersion": 3,
    "EngineAssociation": "4.24",
    "Category": "Samples",
    "Description": "",
    "Modules": [
        {
            "Name": "LandscapeMountains",
            "Type": "Runtime",
            "LoadingPhase": "Default",
            "AdditionalDependencies": [
                "AirSim"
            ]
        }
    ],
    "TargetPlatforms": [
        "MacNoEditor",
        "WindowsNoEditor"
    ],
    "Plugins": [
        {
            "Name": "AirSim",
            "Enabled": true
        }
    ]
}

Close the previous UE4 and VS2019 class windows, I right clic on .uproject file and then on option "Generate Visual Studio Project Files".

I open the .sln file, I configure the build to "DebugGame Editor", and "Win64" (this last one is already done), and when I press on F5 I have the following issues : 

[image]

In output window I have : 
[image]

And at last I have this is the content of my project folder

[image]

Thanks for help.

rajat2004, 2020-04-12:
Thanks for the detailed steps! This problem has been fixed in latest master
To get the latest master branch, go to the AirSim directory and run `git checkout master`,  `git fetch origin`, `git rebase origin/master`
If you have your own fork and have configured this repo to be `upstream` or something else, replace `origin` wih `upstream` or the other name

The run `build.cmd`, go to the Unreal\Environments\Blocks folder and run `update_from_git.bat`, and make sure that Blocks env is working correctly. Also, when building, see if the warnings come.
If it works correctly till now, then copy the Plugins folder to Landscape folder and build again

johan606303, 2020-04-12:
Thanks for this answear.
Nevertheless, I still have an issue when I use cmd, with the command "git rebase origin/master".
[image]

when I run the command "git checkout master" I have : 
[image]

rajat2004, 2020-04-12:
As can be seen from the output, your local repo is behind the master by 27 commits.
You can either stash the changes and do the rebase, or just run `git pull` to fetch and merge the latest master branch with your local repo

johan606303, 2020-04-12:
Thank you for the fast answear. I runed the following commands. Looks like it has worked, for the command "git rebase origin". (Sorry if I'm so boring, I'm beginer with programming, student).

[image]

johan606303, 2020-04-12:
Thank you for your help, LandscapeMountains project works.

closed 2020-04-12 by johan606303

madratman, 2020-04-14:
We also just added a windows binary for landscape mountains https://github.com/microsoft/AirSim/releases/tag/v1.3.1-windows