microsoft/AirSim #2567: Build issue

blacknaki, 2020-04-12:
I use Windows 10 and Visual Studio 2017 and did the build steps but in the end I get that message:
<Hello there! We just upgraded AirSim to Unreal Engine 4.24 and Visual Studio 2019.
Here are few easy steps for upgrade so everything is new and shiny:
https://github.com/Microsoft/AirSim/blob/master/docs/unreal_upgrade.md>

Do I need to upgrade and do again all the steps?

rajat2004, 2020-04-13:
Yes, as mentioned, you'll need to install Visual Studio 2019. 
It can be installed in conjunction with VS 2017, see https://github.com/microsoft/AirSim/pull/2526#issuecomment-611274394 and below to see how to modify UE to use 2019 in that case, but it might be better to remove VS 2017 so as to reduce the chance of problems

blacknaki, 2020-04-13:
Even if I want to build for Unity ?

madratman, 2020-04-13:
Yes, if documentation say to use VS 2019, it is for a reason.

closed 2020-04-13 by madratman